pimox/pimox7 #1: Successfull upgrade from Pimox6to7

semool, 2021-08-18:
I set it as issue :) but it is a success Message.
I have edit all source files to Bullseye and Pimox7 and made a dist-upgrade.
All went smooth and working fine now with v7.
Big Thanks for your great work to contribute this Packages to us.

celangoni, 2021-08-28:
I can confirm!
I need to reinstall pve-manager, but all configuration are kept.
Thanks for an amazing job!

Glutamat42, 2021-08-29:
can confirm it too. I didn't had to reinstall pve-manager. After update i had to do a force refresh in the web interface (strg shift r) to make the statistics working. Sadly i cant start one of my containers, but everything else looks fine (just had a quick look on it, also didn't look into why the container isn't starting anymore after the update)

Interestingly i failed installing pimox7 when i first updated to bullseye (raspbian) and then tried to install pimox7. apt failed with a dependency error (dont remember the exact error message). Now i first installed pimox6 und then updated to bullseye + pimox7 in one step and it worked.

Glutamat42, 2021-08-29:
The problem i have with this one container seems to be an upstream bug. 

https://bugzilla.proxmox.com/show_bug.cgi?id=3516

Just changing ostype in /etc/pve/lxc/ID.conf to debian wasnt enough. The next problem seemed to be that inside the container the file /etc/os-release was missing (dont know why, after creating a new debian 11 container from the same image the file existed). But adding this file also didn't solve the problem. Here i gave up (it was just a testing container), since creating a new one worked.

closed 2021-09-23 by aanon4