pimox/pimox7 #2: Running on RPi 3B+ !!!

TuxfeatMac, 2021-08-22:
**First of all thanks for your work! This is absolutely amazing!**
   So far it is perfectly running fine on my RPi4.

I have some "old" modded RPi3B+ witch I think I can't sell (removed display/camera/audio connectors for a heat spreader & mount plate).... I don't have a use case for them anymore, so I gave it a try... =)

I saw on my RPi4 installation that the idle memory consumption is around 1GB. witch could be a problem, since the RPi3 only has 1GB RAM available...
I used zram and dphys-swapfile to add another extra 1,25GB "mem", a overclock to 1,4GHz and it worked!

[image]

But until now I'm unable to get a VM or CT running, also I cannot connect to the shell of the RPi via the WebGUI, but NoVNC works fine as you can see from the screenshot.

I know the requirement is a RPi4.
So my questions are:
Is there any way to get involved in this what you have done, so that I can maybee adopt it for the RPi3?
Is there any way how to support you?

TuxfeatMac, 2021-08-22:
I've backed up a working VM from my RPi4, increased the zramswap to 2GB.... and with patience it runs !!! 
 [image]

only my CT's are crashing
[image]
any idea ?

TuxfeatMac, 2021-08-28:
Using Images from --> https://downloads.raspberrypi.org/raspios_lite_arm64/images/ <-- Resolved all my issues for the RPi 3B+

[image]

1x Pi as NFS Cluster Server for ISO's / CT-Templates / Backups, and the other
3x Pi's have a working ZFS with replication. VM's / CT's running!!!  Migration working!!! =D

closed 2021-08-28 by TuxfeatMac

timdonovanuk, 2023-08-09:
@TuxfeatMac so, if you use the rasposlite image and follow the normal pimox7 instructions, it all works fine? Thanks!

TuxfeatMac, 2023-08-15:
Running Pimox7 on a RaspberryPi 3B+ is not that easy due to the low avaiable Memory of 1GB... It was more than one year ago that I gave it a try. I Think many things have changed since then OS versions / Pimox7 versions... The key to get Pimox7 running on the 3B+ was **zram** and **dphys-swapfile** so that there is enough "RAM" avaiable to run everything. With the caviat that the CPU and SD Card where always a littel bussy swapping and compressing. Yes it worked but it was slow.

If you want to give it a try take a look at the installer script for the values I used to get it running (lines 43 to 50)
https://github.com/pimox/pimox7/blob/master/RPiOS64-IA-Install.sh

timdonovanuk, 2023-08-15:
Thanks! I'll give it a go :) I don't want to run much on the Pi PV node so I'm hopeful. Cheers!