pimox/pimox7 #3: apt warning debian 11 daily images

TuxfeatMac, 2021-08-22:
I've used the daily images from --> https://raspi.debian.net/daily-images/ <-- for my base install of debian 11

I get warning from apt as soon as i add the second repo like the pimox.sh does -->
Line 6:   echo "deb http://deb.debian.org/debian bullseye contrib" > /etc/apt/sources.list.d/buster-contrib.list

Do you recommend using other images as base ?

BlackMix, 2021-08-28:
Hi @TuxfeatMac 
first u need apt update and upgrade bullseye
and comment that line
```php
#! /bin/sh

apt install -y gnupg
curl https://raw.githubusercontent.com/pimox/pimox7/master/KEY.gpg | apt-key add -
echo "deb https://raw.githubusercontent.com/pimox/pimox7/master/ dev/" > /etc/apt/sources.list.d/pimox.list
# echo "deb http://deb.debian.org/debian bullseye contrib" > /etc/apt/sources.list.d/buster-contrib.list
apt update
apt install -y proxmox-ve
```
bullseye have contrib source.list as default

TuxfeatMac, 2021-08-28:
Hi @BlackMix

TLDR,
Using the Images from --> https://downloads.raspberrypi.org/raspios_lite_arm64/images/ <-- updateing them to 11 and then installing pimox7 indeed workes well =D

I've used the 11 / bullseye image from --> https://raspi.debian.net/daily-images/ <--, so no need to upgrade to 11, I think =)
I've tryed a "vanilla" debian 11 bullseye install, since I'm all so planing to give it a shot on the RockPi 4A 4GB and that would be the only option to get an Debian 11 running on this board I afaik ?

On the "vanilla" images (also tryed stable/tested) I've noticed some things not working so far:
- ZFS
- CT's
- Web Terminal xterm.js

I think it is related to some services not staring propperly like
- apparmor service
- zfs smoe kenrelmodules are unable to load

Butt I haven't had motivation jet to figure out what is going in detail, as I noticed it is running smoothly on the RPiOS Images =D

BlackMix, 2021-08-28:
hmmm, nice :)
I got same problem few time ago with apparmor.. I think kernel pve not compiled for arm64..

so RpiOS can run pimox7, let me test tho, thanks

TuxfeatMac, 2021-08-29:
what I've forgot to mention is that I've installed the raspberrypi-kernel-headers on the RPiOS64 bevore installing proxmox-ve.

Yes indeed your right....
on the "vanilla" installation I've forgot to install the linux-headers bevore installing proxmox-ve.... my fault... and it was not using the pve kernel...

Maybe I give it a nother try the next days =D

vonloschz, 2021-09-01:
Pull request made
https://github.com/pimox/pimox7/pull/4

hyacin75, 2021-09-15:
> On the "vanilla" images (also tryed stable/tested) I've noticed some things not working so far:
<snip>
>     * CT's

I think CTs are to be expected, aren't they?  The images are built with x86 binaries and run right on the host OS - they don't get access to the potential emulation layer in qemu.  I think custom built ARM64 CTs would work fine?

TuxfeatMac, 2021-10-02:
Yes, on a fully functional Pimox installation on RPiOS or DietPi  arm64 CT's are working fine! But for "vanilla" debain images from https://raspi.debian.net/daily-images/ it is not working "out of the box".