pimox/pimox7 #5: unmet dependencies

Ash3781, 2021-09-04:
```
root@raspberrypi:/home/pi# curl https://raw.githubusercontent.com/pimox/pimox7/master/pimox.sh | sh
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Dload  Upload   Total   Spent    Left  Speed
100   364  100   364    0     0   4333      0 --:--:-- --:--:-- --:--:--  4385
Reading package lists... Done
Building dependency tree
Reading state information... Done
gnupg is already the newest version (2.2.12-1+deb10u1).
The following package was automatically installed and is no longer required:
python-colorzero
Use 'apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Dload  Upload   Total   Spent    Left  Speed
100  3143  100  3143    0     0  34163      0 --:--:-- --:--:-- --:--:-- 34163
OK
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian-security buster/updates InRelease
Hit:3 http://archive.raspberrypi.org/debian buster InRelease
Hit:4 http://deb.debian.org/debian buster-updates InRelease
Hit:5 http://deb.debian.org/debian bullseye InRelease
Get:6 https://raw.githubusercontent.com/pimox/pimox7/master dev/ InRelease [2,186 B]
Fetched 2,186 B in 2s (1,314 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
proxmox-ve : Depends: pve-manager but it is not going to be installed
Depends: pve-qemu-kvm but it is not going to be installed
Depends: qemu-server but it is not going to be installed
Depends: spiceterm but it is not going to be installed
Depends: vncterm but it is not going to be installed
Depends: zfsutils-linux but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
updated the os first, did the other steps too but it seems like there are unresolved dependencies, does anyone know what i could do so it installs?

TuxfeatMac, 2021-09-04:
I had the same issue bevore. This is how I solved it:

After updating to bullseye (11) I deleted all sources (sudo rm /etc/apt/sources.list.d/*.list) und added only these to (sudo nano /etc/apt/sources.list):

# Raspberry Pi Bullseye Repo
deb http://archive.raspberrypi.org/debian/ bullseye main

# Pimox7 Repo
deb https://raw.githubusercontent.com/pimox/pimox7/master/ dev/

# Debian Reop
deb http://deb.debian.org/debian bullseye main contrib non-free

# Security Updated
deb http://security.debian.org/debian-security bullseye-security main contrib non-free

Basicaly what you have to do to get it working.
--> remove the buster Repos, remove the "http://deb.debian.org/debian buster-updates InRelease" Updates Repo
then it sould work for you.

Btw are you using the RPiOS64  or a regular Debianarm64 Image?

Regards Tux.feat.Mac
Hope I cloud help.

Ash3781, 2021-09-04:
i'm upgrading the packages at the moment, i just flashed a normal debian ish 64bit os from the official raspberry pi website, i don't know which one is which, but if you'd tell me how to get that info i'll share it

Ash3781, 2021-09-04:
```
Created symlink /etc/systemd/system/multi-user.target.wants/pve-cluster.service → /lib/systemd/system/pve-cluster.service.
Job for pve-cluster.service failed because the control process exited with error code.
See "systemctl status pve-cluster.service" and "journalctl -xe" for details.
Setting up libxml-sax-expat-perl (0.51-1) ...
update-perl-sax-parsers: Registering Perl SAX parser XML::SAX::Expat with priority 50...
update-perl-sax-parsers: Updating overall Perl SAX parser modules info file...
Replacing config file /etc/perl/XML/SAX/ParserDetails.ini with new version
Setting up libpve-cluster-perl (7.0-3) ...
Setting up libpve-http-server-perl (4.0-2) ...
Setting up libpve-access-control (7.0-4) ...
Setting up libpve-cluster-api-perl (7.0-3) ...
Setting up librados2-perl (1.2-1) ...
Setting up libpve-storage-perl (7.0-10) ...
Setting up pve-firewall (4.2-2) ...
Created symlink /etc/systemd/system/multi-user.target.wants/pvefw-logger.service → /lib/systemd/system/pvefw-logger.service.
Job for pvefw-logger.service failed because the control process exited with error code.
See "systemctl status pvefw-logger.service" and "journalctl -xe" for details.
Created symlink /etc/systemd/system/multi-user.target.wants/pve-firewall.service → /lib/systemd/system/pve-firewall.service.
Job for pve-firewall.service failed because the control process exited with error code.
See "systemctl status pve-firewall.service" and "journalctl -xe" for details.
Job for pve-firewall.service failed because the control process exited with error code.
See "systemctl status pve-firewall.service" and "journalctl -xe" for details.
Setting up libpve-guest-common-perl (4.0-2) ...
Setting up pve-container (4.0-9) ...
Setting up pve-ha-manager (3.3-1) ...
watchdog-mux.service is a disabled or a static unit, not starting it.
Created symlink /etc/systemd/system/multi-user.target.wants/pve-ha-lrm.service → /lib/systemd/system/pve-ha-lrm.service.
Job for pve-ha-lrm.service failed because the control process exited with error code.
See "systemctl status pve-ha-lrm.service" and "journalctl -xe" for details.
Created symlink /etc/systemd/system/multi-user.target.wants/pve-ha-crm.service → /lib/systemd/system/pve-ha-crm.service.
Job for pve-ha-crm.service failed because the control process exited with error code.
See "systemctl status pve-ha-crm.service" and "journalctl -xe" for details.
Setting up qemu-server (7.0-13) ...
Created symlink /etc/systemd/system/multi-user.target.wants/qmeventd.service → /lib/systemd/system/qmeventd.service.
Setting up pve-manager (7.0-11) ...
ipcc_send_rec[1] failed: Connection refused
ipcc_send_rec[2] failed: Connection refused
ipcc_send_rec[3] failed: Connection refused
Unable to load access control list: Connection refused
Created symlink /etc/systemd/system/multi-user.target.wants/pvedaemon.service → /lib/systemd/system/pvedaemon.service.
Created symlink /etc/systemd/system/multi-user.target.wants/pveproxy.service → /lib/systemd/system/pveproxy.service.
Created symlink /etc/systemd/system/multi-user.target.wants/spiceproxy.service → /lib/systemd/system/spiceproxy.service.
Created symlink /etc/systemd/system/multi-user.target.wants/pvestatd.service → /lib/systemd/system/pvestatd.service.
Created symlink /etc/systemd/system/getty.target.wants/pvebanner.service → /lib/systemd/system/pvebanner.service.
Created symlink /etc/systemd/system/timers.target.wants/pvesr.timer → /lib/systemd/system/pvesr.timer.
Created symlink /etc/systemd/system/timers.target.wants/pve-daily-update.timer → /lib/systemd/system/pve-daily-update.timer.
Created symlink /etc/systemd/system/sysinit.target.wants/pvenetcommit.service → /lib/systemd/system/pvenetcommit.service.
Created symlink /etc/systemd/system/pve-manager.service → /lib/systemd/system/pve-guests.service.
Created symlink /etc/systemd/system/multi-user.target.wants/pve-guests.service → /lib/systemd/system/pve-guests.service.
Backing up lvm.conf before setting pve-manager specific settings..
'/etc/lvm/lvm.conf' -> '/etc/lvm/lvm.conf.bak'
Setting 'global_filter' in /etc/lvm/lvm.conf to prevent zvols from being scanned:
global_filter=["a|.*|"] => global_filter=["r|/dev/zd.*|"]
Job for pvestatd.service failed because the control process exited with error code.
See "systemctl status pvestatd.service" and "journalctl -xe" for details.
dpkg: error processing package pve-manager (--configure):
installed pve-manager package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of proxmox-ve:
proxmox-ve depends on pve-manager; however:
Package pve-manager is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
dependency problems - leaving unconfigured
Processing triggers for mailcap (3.69) ...
Processing triggers for initramfs-tools (0.140) ...
Processing triggers for libglib2.0-0:arm64 (2.66.8-1) ...
Processing triggers for libc-bin (2.31-13) ...
Processing triggers for rsyslog (8.1901.0-1) ...
Processing triggers for systemd (241-7~deb10u8) ...
Processing triggers for man-db (2.9.4-2) ...
Setting up zfs-zed (2.0.3-9) ...
Created symlink /etc/systemd/system/zed.service → /lib/systemd/system/zfs-zed.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-zed.service → /lib/systemd/system/zfs-zed.service.
Setting up glib-networking:arm64 (2.66.0-2) ...
Setting up libsoup2.4-1:arm64 (2.72.0-2) ...
Setting up gstreamer1.0-plugins-good:arm64 (1.18.4-2) ...
Processing triggers for libgdk-pixbuf-2.0-0:arm64 (2.42.2+dfsg-1) ...
Processing triggers for pve-ha-manager (3.3-1) ...
Processing triggers for systemd (241-7~deb10u8) ...
Processing triggers for libc-bin (2.31-13) ...
Errors were encountered while processing:
pve-manager
proxmox-ve
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@raspberrypi:/home/pi#
```
welp next error

Ash3781, 2021-09-04:
if it'd be easier to do this over discord, name's Ash_#3065

aanon4, 2021-09-04:
Please confirm you have the /etc/hosts and /etc/hostname setup before doing the install because some things won't start/install without it.

Ash3781, 2021-09-04:
should be correctly set up
```
root@raspberrypi:/home/pi# cat /etc/hosts
127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

127.0.1.1               raspberrypi
root@raspberrypi:/home/pi# cat /etc/hostname
raspberrypi
```

aanon4, 2021-09-04:
I'm afraid not. The IP address assigned to the PI needs to be the one you set in /etc/network/interfaces as part of the pre-checks. This will be the one the PI uses to talk to the rest of your network.

Ash3781, 2021-09-04:
> I'm afraid not. The IP address assigned to the PI needs to be the one you set in /etc/network/interfaces as part of the pre-checks. This will be the one the PI uses to talk to the rest of your network.

it'll be connected via lan, address is 192.168.178.47, should it be the localhost or next to the hostname?

vonloschz, 2021-09-04:
https://man7.org/linux/man-pages/man5/hosts.5.html#EXAMPLES

<IP> <FQDN / hostname.domain> <hostname>

Ash3781, 2021-09-04:
> https://man7.org/linux/man-pages/man5/hosts.5.html#EXAMPLES
> ` <IP> <FQDN / hostname.domain> <hostname>`

you mean `192.168.178.47 localhost raspberrypi`?

aanon4, 2021-09-04:
So it should all look something like this:

/etc/network/interfaces should contain at least the following lines (I'm guessing at the netmask and gateway here):

auto eth0
iface eth0 net static
  address 192.168.178.47
  netmask 255.255.255.0
  gateway 192.168.178.1

/etc/hosts should contain at least the following line:

192.168.178.47  raspberrypi

Hope this helps.

Ash3781, 2021-09-04:
>Hope this helps.

it did, i saved the install script, commented out the stuff with the apt list, it says it's done but no webserver is running
```
root@raspberrypi:/home/pi# sh install.sh
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gnupg is already the newest version (2.2.27-2).
The following packages were automatically installed and are no longer required:
bsdmainutils cpp-8 geoip-database libasan5 libbind9-161 libdns1104 libdns1110 libevent-2.1-6 libgeoip1 libisc1100 libisc1105 libisccc161 libisccfg163 libisl19 libjim0.77 liblwres161 libperl5.28
libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib libusb-0.1-4 perl-modules-5.28 python-pkg-resources python2 python2-minimal python2.7 python2.7-minimal python3.7-minimal usb.ids
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 52 not upgraded.
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Dload  Upload   Total   Spent    Left  Speed
100  3143  100  3143    0     0  42472      0 --:--:-- --:--:-- --:--:-- 42472
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
OK
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:3 http://deb.debian.org/debian bullseye InRelease
Get:4 https://raw.githubusercontent.com/pimox/pimox7/master dev/ InRelease [2,186 B]
Fetched 2,186 B in 1s (1,917 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
52 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
proxmox-ve is already the newest version (7.0-2).
The following packages were automatically installed and are no longer required:
bsdmainutils cpp-8 geoip-database libasan5 libbind9-161 libdns1104 libdns1110 libevent-2.1-6 libgeoip1 libisc1100 libisc1105 libisccc161 libisccfg163 libisl19 libjim0.77 liblwres161 libperl5.28
libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib libusb-0.1-4 perl-modules-5.28 python-pkg-resources python2 python2-minimal python2.7 python2.7-minimal python3.7-minimal usb.ids
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 52 not upgraded.
root@raspberrypi:/home/pi#
```

Ash3781, 2021-09-04:
also what's that "operation not permitted" error? i'm root?
```
pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# sudo apt install proxmos-ve
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package proxmos-ve
root@raspberrypi:/home/pi# curl https://raw.githubusercontent.com/pimox/pimox7/master/pimox.sh | sh
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Dload  Upload   Total   Spent    Left  Speed
100   364  100   364    0     0   1405      0 --:--:-- --:--:-- --:--:--  1405
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gnupg is already the newest version (2.2.27-2).
The following packages were automatically installed and are no longer required:
bsdmainutils cpp-8 geoip-database libasan5 libbind9-161 libdns1104 libdns1110 libevent-2.1-6 libgeoip1 libisc1100
libisc1105 libisccc161 libisccfg163 libisl19 libjim0.77 liblwres161 libperl5.28 libpython2-stdlib libpython2.7-minimal
libpython2.7-stdlib libusb-0.1-4 perl-modules-5.28 python-pkg-resources python2 python2-minimal python2.7
python2.7-minimal python3.7-minimal usb.ids
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 52 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up pve-manager (7.0-11) ...
Setting up proxmox-ve (7.0-2) ...
ln: failed to create symbolic link '/boot/pve/vmlinuz': Operation not permitted           <-------
ln: failed to create symbolic link '/boot/pve/initrd.img': Operation not permitted           <-------
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Dload  Upload   Total   Spent    Left  Speed
0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
100  3143  100  3143    0     0  12724      0 --:--:-- --:--:-- --:--:-- 12673
OK
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease                                                            
Get:4 https://raw.githubusercontent.com/pimox/pimox7/master dev/ InRelease [2,186 B]                                      
Fetched 2,186 B in 1s (1,755 B/s)                            
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
52 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: Target Packages (contrib/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Packages (contrib/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Packages (contrib/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Translations (contrib/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Translations (contrib/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Packages (Packages) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list.d/pimox.list:1
W: Target Translations (en_GB) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list.d/pimox.list:1
W: Target Translations (en) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list.d/pimox.list:1
W: Target Packages (Packages) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list.d/pimox.list:1
W: Target Translations (en_GB) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list.d/pimox.list:1
W: Target Translations (en) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list.d/pimox.list:1
W: Target Packages (contrib/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Packages (contrib/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Packages (contrib/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Translations (contrib/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
W: Target Translations (contrib/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:15 and /etc/apt/sources.list.d/buster-contrib.list:1
```

aanon4, 2021-09-04:
The links fail because it's attempting to link things that dont exists on the ARM version (just the official Proxmox stuff).

I am a little confused at this point about the state of your install after the various attempts. If you are in the position to wipe and start again I'd do that, making sure to setup the IP and hostnames before you start the install as well as following the changes suggested by @TuxfeatMac.

Ash3781, 2021-09-04:
@aanon4 so change the hostname first, then comment out the apt source stuff in the install script, then add the other ones and then install it?

TuxfeatMac, 2021-09-04:
@aanon4 Am I allowed to post a simple Guide / StepbyStepHowTo in the ProxmoxForum or somewhere? To avoid errors because of the missing prechecks ?

@Ash3781 Yes I've done all the network setup first and then the installation of proxmox-ve

aanon4, 2021-09-04:
@TuxfeatMac Personally I think that'd be fantastic.

TuxfeatMac, 2021-09-04:
OK thank you for your permission im on it =)

@aanon4 ther is one more unmet dependecie i forgot to mention... bevore you install proxmox-ve make sure to install the kernel headers (sudo apt install -y raspberrypi-kernel-headers) since they are allso needed

aanon4, 2021-09-04:
I with they'd get a native bullseye raspian image up so I'd have a better base to fix these dependencies.

vonloschz, 2021-09-04:
DietPi use Debian bullseye by default now.

Ash3781, 2021-09-04:
just to be sure i'm not completely dumb, if everything works i get no errors? like `dpkg-statoverride: warning: --update given but /var/log/chrony does not exist`seems not right for me, but it's still running

Ash3781, 2021-09-04:
bollocks.
```
dpkg: dependency problems prevent configuration of pve-manager:
pve-manager depends on ifupdown2 (>= 2.0.1-1+pve8) | ifenslave (>= 2.6); however:
Package ifupdown2 is not configured yet.
Package ifenslave is not installed.

dpkg: error processing package pve-manager (--configure):
dependency problems - leaving unconfigured
Setting up zfs-dkms (2.0.3-9) ...
Loading new zfs-2.0.3 DKMS files...
```
still running but i guess it will crash or error out at some point

Ash3781, 2021-09-04:
next error
```
DKMS: install completed.
Module build for kernel 5.10.17-v8+ was skipped since the
kernel headers for this kernel does not seem to be installed.
dpkg: dependency problems prevent configuration of proxmox-ve:
proxmox-ve depends on pve-manager; however:
Package pve-manager is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
dependency problems - leaving unconfigured
```
and it failed with
```
Errors were encountered while processing:
ifupdown2
pve-manager
proxmox-ve
E: Sub-process /usr/bin/dpkg returned an error code (1)

aanon4, 2021-09-04:
> @aanon4 there is one more unmet dependecie I forgot to mention... bevore you install proxmox-ve make sure to install the kernel headers (sudo apt install -y raspberrypi-kernel-headers), since they are allso needed.

Well the package linux-headers-arm64 should have been brought in as a dependency from the dkms package, which I would have hoped was sufficient here. Not sure how raspberrypi-kernel-headers is different or whether its in default debian.

TuxfeatMac, 2021-09-04:
The RPiOS64 uses different kernel headers afaik... I needed to install the raspberrypi-kernel-headers in order to get it running and to get zfs working.

TuxfeatMac, 2021-09-04:
https://forum.proxmox.com/threads/how-to-run-pve-7-on-a-raspberry-pi.95658/
I will add more details tomorrow =)

@Ash3781  here you go =) step by step simply copy paste but don't forget to adjust IP's and hostname.

Ash3781, 2021-09-05:
@TuxfeatMac do I need to re-flash the pi or can I just use the pi in it's "condition"?

TuxfeatMac, 2021-09-05:
I think at this point, start with a fresh install, read the instructions first, then follow them. If asked something by apt, going with the default and just hintting enter is okay.

TuxfeatMac, 2021-09-05:
> 
> 
> DietPi use Debian bullseye by default now.
@vonloschz have you tried to install pimox7 on top of it ? Just curious... =)

> 
> 
> I with they'd get a native bullseye raspian image up so I'd have a better base to fix these dependencies.
@aanon4 Is there somewhere a Dokumentation / a brief  how to or guide how you got up your native bullseye raspian image? All so just curious =)

I am working on a fully automatic install script for the RPi 4B based on the 2021-05-07-raspios-buster-arm64-lite.zip image, as soon as I'm finished I will make a pull request.

Ash3781, 2021-09-05:
it rebooted but no webserver running, also errors with proxmos-ve and pve-manager
```
Job for pvestatd.service failed because the control process exited with error code.
See "systemctl status pvestatd.service" and "journalctl -xe" for details.
dpkg: error processing package pve-manager (--configure):
installed pve-manager package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of proxmox-ve:
proxmox-ve depends on pve-manager; however:
Package pve-manager is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
dependency problems - leaving unconfigured
Processing triggers for dbus (1.12.20-2) ...
Processing triggers for mailcap (3.69) ...
Processing triggers for initramfs-tools (0.140) ...
Processing triggers for libglib2.0-0:arm64 (2.66.8-1) ...
Processing triggers for libc-bin (2.31-13) ...
Processing triggers for rsyslog (8.2102.0-2) ...
Processing triggers for man-db (2.9.4-2) ...
Setting up zfs-zed (2.0.3-9) ...
Created symlink /etc/systemd/system/zed.service → /lib/systemd/system/zfs-zed.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-zed.service → /lib/systemd/system/zfs-zed.service.
Setting up glib-networking:arm64 (2.66.0-2) ...
Setting up libsoup2.4-1:arm64 (2.72.0-2) ...
Setting up gstreamer1.0-plugins-good:arm64 (1.18.4-2) ...
Processing triggers for libgdk-pixbuf-2.0-0:arm64 (2.42.2+dfsg-1) ...
Processing triggers for pve-ha-manager (3.3-1) ...
Processing triggers for libc-bin (2.31-13) ...
Errors were encountered while processing:
pve-manager
proxmox-ve
E: Sub-process /usr/bin/dpkg returned an error code (1)

[screen is terminating]
Connection to 192.168.178.47 closed by remote host.
Connection to 192.168.178.47 closed.
[ash@ash-pc ~]$ ssh pi@192.168.178.47
pi@192.168.178.47's password:
Linux RPi4-PVE-01 5.10.60-v8+ #1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Sep  5 05:40:14 2021 from 192.168.178.48
upi@RPi4-PVE-01:~ $ uptime#
-bash: uptime#: command not found
pi@RPi4-PVE-01:~ $ uptime
06:16:06 up 0 min,  1 user,  load average: 1.28, 0.44, 0.16
pi@RPi4-PVE-01:~ $

Ash3781, 2021-09-05:
those are my edited files, that "lo" part was generated after/during the install
```
pi@RPi4-PVE-01:~ $ cat /etc/network/interfaces
# network interface settings; autogenerated
# Please do NOT modify this file directly, unless you know what
# you're doing.
#
# If you want to manage parts of the network configuration manually,
# please utilize the 'source' or 'source-directory' directives to do
# so.
# PVE will preserve these directives, but will NOT read its network
# configuration from sourced files, so do not attempt to move any of
# the PVE managed interfaces into external files!

auto lo
iface lo inet loopback

auto eth0
iface eth0 inet static
address 192.168.178.47/24
gateway 192.168.178.1

pi@RPi4-PVE-01:~ $ cat /etc/hosts
#127.0.0.1      localhost
#::1            localhost ip6-localhost ip6-loopback
#ff02::1                ip6-allnodes
#ff02::2                ip6-allrouters

192.168.178.47  RPi4-PVE-01
127.0.0.1       localhost
pi@RPi4-PVE-01:~ $

TuxfeatMac, 2021-09-05:
Strange.. The errors in your first post should be okay and correct itself on the next reboot. That lo part is okay and should be there. Have you tied running apt-install proxmox-pve again ? Maybe try to give yor RPi a littel time to settel down e.g. 2  to 3 min of idle time so that the services can become ready before trying to access the WebGUI?

Ash3781, 2021-09-05:
>Maybe try to give yor RPi a littel time to settel down e.g. 2 to 3 min of idle time so that the services can become ready before trying to access the WebGUI?

welp it's idle for about 2 hours and 17 minutes now, i don't mind waiting a bit but almost 2 and a half hours... eh shouldn't take that long, rebooted it a couple of times too, i'll reboot it again and try installing that again, have you got discord so i could stream my terminal?

TuxfeatMac, 2021-09-05:
okay fine =) https://discord.gg/JgSrAnKv

closed 2021-09-05 by Ash3781