pimox/pimox7 #7: can't start a vm

Ash3781, 2021-09-06:
i'm trying to set up a arch linux arm vm machine, but every time i try starting it i get this error:
```
kvm: -device ide-cd,bus=ide.1,unit=0,drive=drive-ide2,id=ide2,bootindex=101: Bus 'ide.1' not found
TASK ERROR: start failed: QEMU exited with code 1
```
i tried creating a disk and changed it's "port" to ide 1 but same error message, any ideas on how to start it?

TuxfeatMac, 2021-09-07:
ide kernel drivers are not avaiable..... use scsi  instead!

aanon4, 2021-09-23:
Please re-open if this didn't fix your problem.

closed 2021-09-23 by aanon4

jeffbelmiro, 2022-03-26:
@Ash3781 You have to remove CD/DVD drive because it is IDE type. Just use the Storage as SCSI. Below is how I created my last VM.

[image]

rventura520, 2022-09-21:
[image]

Hey, i'm tring to creat a VM ubuntu server on the pimox server and no matter the config I set, I always get this error. Anyboy know how can i fix it?

TheBUllG, 2023-01-18:
> @Ash3781 You have to remove CD/DVD drive because it is IDE type. Just use the Storage as SCSI. Below is how I created my last VM.
> 
> [image]

how can i remove it?