pimox/pimox7 #10: USB Passwthough not working

lug-gh, 2021-09-17:
Hi, not sure if usb passthrough wont work by design, or if it's just a bug.

steps to reproduce:
- Fresh PI Os 64 Bit upgraded to bullseye
- fresh pimox7
- create a vm, in my case alpine linux aarch64

qemu config:
´´´
agent: 1,fstrim_cloned_disks=1
bios: ovmf
boot: order=scsi1;scsi0;net0
cores: 4
cpu: host
efidisk0: SSD:vm-580-disk-1,size=64M
memory: 2048
name: test
net0: virtio=06:B7:8E:29:59:6A,bridge=vmbr0
numa: 0
ostype: l26
parent: clean
scsi0: SSD:vm-580-disk-0,discard=on,size=20G,ssd=1
scsi1: none,media=cdrom
scsihw: virtio-scsi-pci
smbios1: uuid=49442ca9-9897-408b-bdd7-f1a4890aeb77
sockets: 1
usb0: host=0451:16c8
usb1: host=0a12:0001

[clean]
balloon: 1024
bios: ovmf
boot: order=scsi1;scsi0;net0
cores: 2
cpu: host
efidisk0: SSD:vm-580-disk-1,size=64M
memory: 1536
name: test
net0: virtio=06:B7:8E:29:59:6A,bridge=vmbr0
numa: 0
ostype: l26
runningcpu: host
runningmachine: virt-6.0+pve0
scsi0: SSD:vm-580-disk-0,discard=on,size=20G,ssd=1
scsi1: none,media=cdrom
scsihw: virtio-scsi-pci
smbios1: uuid=49442ca9-9897-408b-bdd7-f1a4890aeb77
snaptime: 1631870673
sockets: 1
vmstate: SSD:vm-580-state-clean
´´´

The VM boots just fine, but the usb devices are stuck on the pi, they dont get passed through.

aanon4, 2021-09-23:
I've never thought to try this. And idea if this works on PIs at all?

lug-gh, 2021-09-23:
Yes it works, but maybe pimox is not the problem here.

I think it's because I used the alpine "virt" iso - which is trimmed down, maybe without usb drivers?
On my x64 server with proxmox it doesn't work either, so i think it's all because the "virt" version.

I cannot test this now anymore, because I am running esxi on that pi now, where usb passthrough works with alpine "standard" version. The virt version won't boot for me onm esxi.

closed 2021-09-23 by lug-gh

TuxfeatMac, 2021-09-24:
> 
> 
> I've never thought to try this. And idea if this works on PIs at all?

For me it is allso not working..

VM is unable to start with passthrough USB

kvm: -device usb-ehci,id=ehci,bus=pci.0,addr=0x1: Duplicate ID 'ehci' for device
TASK ERROR: start failed: QEMU exited with code 1

Dattel, 2021-12-05:
same here.. once i added a USB-passthrough-drive, i only got to the UEFI-Bios. 
If i try to remove the USB-Port from the machine, the machine will no longer be able to start.
So i decited to reboot the pimox system.

After reboot the machine boots flawlessly and the USB-passthrouh was attached and working.