pimox/pimox7 #11: Display output is not active

noogugae, 2021-09-20:
I sucessfully installed  pimox on debian11 and tried to install debian 10 for Home assistant.
At first, debian 10 install screen was appered and I selected install
But right after, I only can see black screen with below comment.
'Display output is not active'
I tried serverl display option but only default option is work. 
The others option didn't show even the first install screen.
What should I do?

aanon4, 2021-09-20:
The only display option you can use when installing Debian 10 is "serial terminal". You can at least change this back to something more "normal" once the install is complete.

noogugae, 2021-09-21:
> 
> 
> The only display option you can use when installing Debian 10 is "serial terminal". You can at least change this back to something more "normal" once the install is complete.

I tried a serial terminal after changing .conf file like below

serial0: socket
vga: serial0

It shows a black screen also with this word, 'starting serial termianl on interface serial0'
It even doen't show debian initial install screen.
I waited serveral minutes but nothing happen.
Is there any different method to use 'serial terminal"

aanon4, 2021-09-23:
No, that was the one I was thinking of. Hmm ... I will try to replicate this, but it'll be a couple of weeks before I can look at it I'm afraid.

noogugae, 2021-09-24:
> 
> 
> No, that was the one I was thinking of. Hmm ... I will try to replicate this, but it'll be a couple of weeks before I can look at it I'm afraid.
I figured it out.
'novnc' console  isn't working but 'xterm' is working.
thanks

closed 2021-09-24 by noogugae