pimox/pimox7 #13: Can't connect to network at sub OS(debian 10)

noogugae, 2021-09-24:
I have installed debain 10 at pimox7.
But debian 10 can't connect to network.
I tried several normal solution for network problem but failed.

[image]

this is '/etc/network/interface' at debian11
```
auto lo
iface lo inet loopback

auto enabcm6e4ei0
iface enabcm6e4ei0 inet static
	address 192.168.35.163/24
	gateway 192.168.35.1

auto vmbr0
iface vmbr0 inet static
        address 192.168.35.8/24
        gateway 192.168.35.1
	bridge-ports none
	bridge-stp off
	bridge-fd 0
```

And this is '/etc/network/interface' at debian10

```
auto ens0s18
iface ens0s18 inet static
        address  192.168.35.8/24
        gateway 192.168.35.1
```

How can I connect to network? Any Idea?

TuxfeatMac, 2021-09-25:
Seems to be related to a missconfigured network settings....

1. Do not use the same IP adress for Proxmox and Guest.
2. Do not assing a IP address to your enabcm6e4ei0, since it will use the Bridge IP address.

try this ---->

/etc/network/interfaces   @ Proxmox
auto lo
iface lo inet loopback

iface eth0 inet manual

auto vmbr0
iface vmbr0 inet static
        address 192.168.35.163/24
        gateway 192.168.35.1
        bridge-ports eth0
        bridge-stp off
        bridge-fd 0

/etc/network/interfaces @ Debian Guest
auto enp0s18
iface enp0s18 inet static
        address  192.168.35.8/24
        gateway 192.168.35.1

noogugae, 2021-09-25:
> Seems to be related to a missconfigured network settings....
>
>     1. Do not use the same IP adress for Proxmox and Guest.
>
>     2. Do not assing a IP address to your enabcm6e4ei0, since it will use [...]

'enabcm6e4ei0' is for the debian 11 network.
I tried to remove enabcm6e4ei0 setting, but in that case, I couldn't connect debian 11 and proxmox gui.
So I thought that was not a solution.
This time, I changed 'interface' like below
As a result, proxmox ip is changed to 192.168.35.164 but sub os network still doesn't working.
Am I wrong?
```
auto lo
iface lo inet loopback

auto enabcm6e4ei0
iface enabcm6e4ei0 inet static
        address 192.168.35.163/24
        gateway 192.168.35.1

auto eth0
iface eth0 inet manual

auto vmbr0
iface vmbr0 inet static
        address 192.168.35.164/24
        gateway 192.168.35.1
        bridge-ports eth0
        bridge-stp off
        bridge-fd 0
```
[image]

TuxfeatMac, 2021-09-25:
I see, you have attached an extra USB NIC to your RPi, am I right ?

This won't work at the moment, since passtrough usb is not working...

The way to get it to run is to use the buildin ethernet port. Assing it to the bridge, only give your bridge an IP address and gateway.

Finaly assing your guest to the bridge as you did in your first post.

noogugae, 2021-09-26:
> I see, you have attached an extra USB NIC to your RPi, am I right ?
>
> This won't work at the moment, since passtrough usb is not working...
>
> The way to get it to run is to use the buildin ethernet [...]

No. I didn't attached an extra USB NIC, but I solved it.
I don't know what wrong before because it is basic network setting for proxmox.
Anyway This is my final 'interface' setting
Thanks for your help.

```
auto lo
iface lo inet loopback

iface enabcm6e4ei0 inet manual

auto vmbr0
iface vmbr0 inet static
        address 192.168.35.164/24
        gateway 192.168.35.1
        bridge-ports enabcm6e4ei0 
        bridge-stp off
        bridge-fd 0
```

closed 2021-09-26 by noogugae