pimox/pimox7 #14: Can you teach me how to download LXC AND KVM ARM and which version to download?

wartw, 2021-09-25:
Can you teach me how to download LXC AND KVM ARM and which version to download?
I tried for a long time but couldn't start

TuxfeatMac, 2021-09-25:
Maybe following this guide, helps ? 

https://forum.proxmox.com/threads/how-to-run-pve-7-on-a-raspberry-pi.95658/

wartw, 2021-09-25:
@TuxfeatMac 
My problem is that installing KVM and LXC is not installing PVE7 (I have already installed it, but I will not install ARM's LXC or KVM mirror

TuxfeatMac, 2021-09-25:
So you are searching for a guide on how to run a LXC Container with Proxmox ? And how to create a Virtual Machine with Proxmox ?

wartw, 2021-09-25:
Yes, I need the ARM version of teaching (preferably the complete installation process of KVM

TuxfeatMac, 2021-09-25:
LXC images can be downloaded from : https://us.images.linuxcontainers.org/images
I've wrote a littel skript witch grabbs the latest debian arm64 LXC and puts in the right place.
After that you shold be able to create and start a LXC Debian 11 Container

# ! /bin/bash
# Debian 11 / Bullseye Arm 64 - CT
DISTNAME=debian
CODENAME=bullseye
NEWESTBUILD=$(curl https://us.images.linuxcontainers.org/images/$DISTNAME/$CODENAME/$ARCHITEC/default/ | grep '<td>' | tail -n 1 | cut -d '='  -f 5 | cut -d '/' -f 2)
printf "\n DEBIAN 11 \n"
cd /var/lib/vz/template/cache
wget https://us.images.linuxcontainers.org/images/$DISTNAME/$CODENAME/$ARCHITEC/default/$NEWESTBUILD/rootfs.tar.xz -O Debian11$ARCHITEC-std-$NEWESTBUILD.tar.xz
# EOF #

FOR a KVM Debian 11 images can be optianed from: https://cdimage.debian.org/debian-cd/
Allso a littel script for donloading the ios and putting it in the right place.
create a VM through the gui be sure to re add the cd as scsi device, set the machine type to uefi, adjust the boot oder and use the advanced option host cpu.

# !/bin/bash
# Debian 11 arm64 net iso
cd /var/lib/vz/template/iso
wget https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/debian-11.0.0-arm64-netinst.iso

ryanc1256, 2021-10-09:
@TuxfeatMac even using the script you use (above) when starting the LXC container it gets an apparmor error

```
run_apparmor_parser: 919 Failed to run apparmor_parser on "/var/lib/lxc/133/apparmor/lxc-133_<-var-lib-lxc>": apparmor_parser: Unable to replace "lxc-133_</var/lib/lxc>".  Profile doesn't conform to protocol
apparmor_prepare: 1089 Failed to load generated AppArmor profile
lxc_init: 850 Failed to initialize LSM
__lxc_start: 2007 Failed to initialize container "133"
TASK ERROR: startup for container '133' failed
```

TuxfeatMac, 2021-11-07:
@kenji21 thanks for the workaround! Now im able to start lxc containaers on a vanilla debian insalation... however I'm still getting errors in the console. Maybe we can find a permanent fix for that?

TuxfeatMac, 2021-11-13:
Using the latest image form, 
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/2021-10-30-raspios-bullseye-arm64.zip
with the latest autoinstaller from,
https://github.com/TuxfeatMac/pimox7/blob/master/RPiOS64-IA-Install.sh
everyting works fine for me,
https://www.youtube.com/watch?v=LGb7fB1wK4Q