pimox/pimox7 #18: TASK ERROR: start failed: QEMU exited with code 1

supersunil, 2021-09-30:
After the successful installation of Proxmox server on Raspberry Pi 4 B (4GB RAM), navigated to the web browser and pointed the host server with port 8006 and I was successful at the login as well.

But when I tried to create the VM I was hit by the following error.

`kvm: -device ide-cd,bus=ide.1,unit=0,drive=drive-ide2,id=ide2,bootindex=101: Bus 'ide.1' not found
TASK ERROR: start failed: QEMU exited with code 1`

Help me to fix this issue.

TuxfeatMac, 2021-10-02:
This is an easy fix IDE drivers are not avaivable on the pi so delete your ide-cdrom and readd it as scsi-cd, then it should start up. Hope I could help.

supersunil, 2021-10-03:
Hey @TuxfeatMac,

Thanks for writing-up. I need to do this while creating the VM or while installing the proxmox to Pi?

I'm pretty new to this, so a decent tutorial will help me.

supersunil, 2021-10-03:
@TuxfeatMac - Good news. Managed to run the VM successfully but when I navigate to console, it always says, `guest has not initialized the display (yet)`

Am I missing something?

TuxfeatMac, 2021-10-03:
I think you have to adjust te boot order.

supersunil, 2021-10-03:
Like, 
1. The CD/DVD
2. Hard-disk
3. Net

Right?

metronidazole, 2021-10-10:
Same here, console does not work. 

@TuxfeatMac  Does the console work for you? What OS are you visualizing?

metronidazole, 2021-10-10:
I managed to get the console to work by using the OVMF bios (UEFI) instead of the default (seabios). However I couldn't get it to boot from an ISO / CD drive: failed to load boot0001 uefi qemu cd-rom

metronidazole, 2021-10-11:
> It's taking me to the interactive shell

You can type exit to get to the boot selection screen. I found this post by one of the devs who says this was a working config (in the pimox v6 build). I haven't tried it yet though: https://old.reddit.com/r/Proxmox/comments/n4d7ev/now_running_proxmox_on_a_raspberry_pi_4_cluster/gwzr4h4/

TuxfeatMac, 2021-11-13:
maybe this helps ?
https://github.com/pimox/pimox7/issues/20#issuecomment-968064376

soxrok2212, 2021-11-30:
> maybe this helps ? [#14 (comment)](https://github.com/pimox/pimox7/issues/14#issuecomment-968082718)

This worked for me. Cheers!

szuecs-andras, 2022-02-19:
> Hey @TuxfeatMac,
> 
> Thanks for writing-up. I need to do this while creating the VM or while installing the proxmox to Pi?
> 
> I'm pretty new to this, so a decent tutorial will help me.

hey, I have the same issue, what did you do in the end?

szuecs-andras, 2022-02-20:
nvm, found a YouTube video for us n00bs:

https://youtu.be/nulmOjaAnAM

Im-clutch, 2022-03-15:
> This is an easy fix IDE drivers are not avaivable on the pi so delete your ide-cdrom and readd it as scsi-cd, then it should start up. Hope I could help.

This worked for me. Thanks