pimox/pimox7 #20: Network connectivity lost after setting bridge setup

metronidazole, 2021-10-09:
I am having a strange issue after setting this up. I followed the readme and used @TuxfeatMac 's script to update to bullseye and install pimox7. That all worked initially, but I wanted to change my networking to use a bridge interface so I could hook my testing containers/VMs  into the bridge. However, after configuring this and rebooting I lost network connectivity to the pimox server (UI and via SSH, ping, etc). I have seen #13 which seems related.

After doing some testing, it seems that for some reason, the real physical network interface gains a local IP (169.254.36.32) and a second default route is created using that within a minute of networking starting (using ifdown and ifup). After resetting the network, I can use it until this second route is made. See the following debug log for further explanation. 

Any ideas on what is causing this? I assume it's probably some rogue debian script but I'm not sure how to proceed. Can I intercept calls to route?

> $ cat /etc/network/interfaces

> auto lo
> iface lo inet loopback
> 
> iface eth0 inet manual
> 
> auto vmbr0
> iface vmbr0 inet static
>         address 10.55.20.19/24
>         gateway 10.55.20.1
>         bridge-ports eth0
>         bridge-stp off
>         bridge-fd 0
> 
> $ ip addr
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
>     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
>     inet 127.0.0.1/8 scope host lo
>        valid_lft forever preferred_lft forever
>     inet6 ::1/128 scope host 
>        valid_lft forever preferred_lft forever
> 2: eth0: <BROADCAST,MULTICAST,DYNAMIC,UP,LOWER_UP> mtu 1500 qdisc mq master vmbr0 state UP group default qlen 1000
>     link/ether dc:a6:32:8b:4c:1a brd ff:ff:ff:ff:ff:ff
>     inet 169.254.36.32/16 brd 169.254.255.255 scope global eth0
>        valid_lft forever preferred_lft forever
>     inet6 fe80::dea6:168b:ef2a:9b4c/64 scope link 
>        valid_lft forever preferred_lft forever
> 3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
>     link/ether dc:a6:32:8b:3a:1d brd ff:ff:ff:ff:ff:ff
> 9: vmbr0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
>     link/ether dc:a6:32:8b:4c:1a brd ff:ff:ff:ff:ff:ff
>     inet 10.55.20.19/24 scope global vmbr0
>        valid_lft forever preferred_lft forever
>     inet6 fe80::dea6:168b:ef2a:9b4c/64 scope link 
>        valid_lft forever preferred_lft forever
> 
> $ route
> Kernel IP routing table
> Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
> default         0.0.0.0         0.0.0.0         U     0      0        0 eth0
> default         10.55.20.1       0.0.0.0         UG    0      0        0 vmbr0
> 10.55.20.0      0.0.0.0         255.255.255.0   U     0      0        0 vmbr0
> link-local      0.0.0.0         255.255.0.0     U     0      0        0 eth0
> 
> $ ping 1.1.1.1
> doesnt work
> 
> $ ping -I vmbr0 1.1.1.1
> works
> 
> * cannot access proxmox UI on other PC
> 
> -------------------------------------
> manual intervention
> 
> $ ifdown vmbr0 eth0
> 
> $ ifup vmbr0 eth0
> 
> $ ip addr
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
>     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
>     inet 127.0.0.1/8 scope host lo
>        valid_lft forever preferred_lft forever
>     inet6 ::1/128 scope host 
>        valid_lft forever preferred_lft forever
> 2: eth0: <BROADCAST,MULTICAST,DYNAMIC,UP,LOWER_UP> mtu 1500 qdisc mq master vmbr0 state UP group default qlen 1000
>     link/ether dc:a6:32:8b:4c:1a brd ff:ff:ff:ff:ff:ff
>     inet6 fe80::dea6:168b:ef2a:9b4c/64 scope link 
>        valid_lft forever preferred_lft forever
> 3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
>     link/ether dc:a6:32:8b:3a:1d brd ff:ff:ff:ff:ff:ff
> 9: vmbr0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
>     link/ether dc:a6:32:8b:4c:1a brd ff:ff:ff:ff:ff:ff
>     inet 10.55.20.19/24 scope global vmbr0
>        valid_lft forever preferred_lft forever
>     inet6 fe80::dea6:168b:ef2a:9b4c/64 scope link 
>        valid_lft forever preferred_lft forever
> 
> $ route
> Kernel IP routing table
> Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
> default         10.55.20.1       0.0.0.0         UG    0      0        0 vmbr0
> 10.55.20.0       0.0.0.0         255.255.255.0   U     0      0        0 vmbr0
> 
> $ ping 1.1.1.1
> works
> 
> * can access proxmox UI on other PC
> 
> 
> after a minute or so, the networking configuration reverts to the situation above the manual intervention line
> then network connectivity is lost
> for testing purposes, i tried to delete the second default route (eth0) but the route would not delete despite multiple attempts 
> 
> I know I could remove the bridge and just use masquerading but I want to have full networking

metronidazole, 2021-10-10:
Doing some further testing I have discovered that connman was the cause (i.e. fucking with my network configuration). 

It seems to be installed by default when using the suggested image in the readme (under RPiOS64 autoinstall). It should probably be removed or disabled in the auto setup script.

TuxfeatMac, 2021-11-13:
I have now updated the installer script, to work with the new latest image
https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-11-08/

Test runs appriciated, =) works for me.

0. sudo su
1. curl https://github.com/TuxfeatMac/pimox7/blob/master/RPiOS64-IA-Install.sh > RPiOS64-IA-Install.sh
2. chmod +x RPiOS64-IA-Install.sh
3. ./RPiOS64-IA-Install.sh

chadai, 2021-11-13:
Hi, I am assuming should refer to the raw file here?

`curl https://raw.githubusercontent.com/TuxfeatMac/pimox7/master/RPiOS64-IA-Install.sh > RPiOS64-IA-Install.sh`

Also, I get the following result (however, I didn't run this in a Screen session; would that make a difference?)

```shell
...
Installing new version of config file /etc/default/networking ...

network config changes have been detected for ifupdown2 compatibility.
Saved in /etc/network/interfaces.new for hot-apply or next reboot.

Reloading network config on first install
client_loop: send disconnect: Broken pipe
```

TuxfeatMac, 2021-11-13:
Tanks @chadai I've updated the comment!

this file will be delete by my script:
/etc/network/interfaces.new

TuxfeatMac, 2021-11-13:
Have you chosen the same IP address, your pi got on firt boot up via dhcp, as the static IP in the script?

TuxfeatMac, 2021-11-13:
Thanks again @chadai for time and your try!
For me there was no issue.

I've updateded the installer script.
Network changes will be applied at reboot,
preventing the ssh conection getting closed during installation.