pimox/pimox7 #23: Can't create an EFI disk anymore

SH160388, 2021-11-04:
Hi there,
I use proxmox on raspberry pi 4 (thus on arm64)

So far everything has been going well. A few days ago I performed apt update and apt upgrade on the hosts.

Now I wanted to create a VM as usual and found that the following error occurs.

"Can't use an undefined value as an ARRAY reference at /usr/share/perl5/PVE/QemuServer.pm line 3173."

I have found out so far that it appears to be a UEFI BIOS related problem.
Specifically, the error occurs when creating the EFI disk. (Without it works)

I hope you have an idea or let me know which logs / information I should leave out.

Many greetings

aanon4, 2021-11-04:
It looks like they changed the default UEFI image (TPM related I suspect). Will see if I can fix it later today.

SH160388, 2021-11-04:
Sounds great. I'm looking forward to hear from you.

aanon4, 2021-11-04:
Please update and try again.

SH160388, 2021-11-04:
It works. Thanks a lot.

Dattel, 2021-12-03:
Hi,
i just updated to 7.1-7 and now i also have the same problem with a newer PVE version:

"Can't use an undefined value as an ARRAY reference at /usr/share/perl5/PVE/QemuServer.pm line 3291. (500)"
Is there anything i forgot to change?

Thanks a lot

aanon4, 2021-12-03:
Update and give it another try if you could (they keep adding new EFI configurations)

Dattel, 2021-12-03:
brilliant... mate, big thanks
works again :-D

closed 2021-12-03 by aanon4