pimox/pimox7 #30: RPiOS64bit Interactive Automatic Installer ->  Connection failure. Network error or Proxmox VE services not running?

Dattel, 2021-11-28:
Hi there.
Thanks for the port :-D

i just used the automatic installer on the "2021-10-30-raspios-bullseye-arm64-lite.zip" image on a RPI4.
The Installation runs fine, the WebFrontend is accessible through the network but i can't log in. 

With LinuxPAM authentication module and Proxmox VE Authentication 
Connection failure. Network error or Proxmox VE services not running?

Any ideas on that? I already tried with the "curl https://raw.githubusercontent.com/pimox/pimox7/master/pimox.sh | sh" in an earlier try on a clean system but i had other issues like VMBR0 not configured or empty dashboard after login in through web.

Dattel, 2021-11-29:
turns out, that it is important to access the frontend using the DNS-name instead of IP..

so instead of using https://192.168.178.6:8006 i have to use https://pimox.fritz.box:8006
What is the reason for that?

hellresistor, 2021-12-02:
Check the "Precheck" list on README ?

Dattel, 2021-12-02:
thanks, i realized, that the pi needs a valid name and it was indead proper configured but the name was not registered in local-DNS. so i thought if the frontend came up via https://ip it could also work.... but dosn't.... 

nevertheless - it's not a problem, if you know about that :-D

hellresistor, 2021-12-02:
really, I not try yet. I am waiting for weekend ^^ 
I can give feedback with my experience ;)

Dattel, 2021-12-07:
Did you change anything on that? since the latest release i'm able to login through DDNS and IP as expected.

closed 2021-12-07 by Dattel

hellresistor, 2021-12-12:
Its ok.!!

Just thing i found. I have unable to import an HomeAssistant .qcow2 image