pimox/pimox7 #31: Use of uninitialized value $path in -c at /usr/share/perl5/PVE/QemuServer.pm line 3691.

pschonmann, 2021-12-03:
Fresh install RpiOS 64BIT LITE version.

Im getting this error

```
Use of uninitialized value $path in -c at /usr/share/perl5/PVE/QemuServer.pm line 3691.
TASK ERROR: no such serial device
```
Config of vm
```
balloon: 0
boot: order=virtio0;ide2;net0
cores: 1
ide2: local:iso/debian-11.1.0-amd64-netinst.iso,media=cdrom
memory: 2048
meta: creation-qemu=6.0.0,ctime=1638546622
name: librenms
net0: virtio=9E:E9:A6:AC:20:77,bridge=vmbr0
numa: 0
ostype: l26
scsihw: virtio-scsi-pci
smbios1: uuid=8b40d464-a7b1-45f5-bfb0-8b02fb5e1026
sockets: 1
virtio0: local:100/vm-100-disk-0.qcow2,backup=0,cache=writeback,size=10G
```

rtoenniges, 2021-12-03:
I got the same error after update from 7.0.1 to 7.1.7

aanon4, 2021-12-03:
Update and give it a try if you could

rtoenniges, 2021-12-03:
Working fine now, thank you very much!

closed 2021-12-03 by aanon4