pimox/pimox7 #32: Guest has not initialized display yet

pschonmann, 2021-12-03:
After update with #31 i started VM but getting info - guest has not initialized display yet. VM load is still @100% and nothing happens

Config is without disk, even bios not loading :(
```
agent: 0
balloon: 0
boot: order=net0
cores: 1
memory: 2048
meta: creation-qemu=6.0.0,ctime=1638565898
name: librenms
net0: virtio=8E:B8:7E:E9:00:1C,bridge=vmbr0
numa: 0
ostype: l26
scsihw: virtio-scsi-pci
smbios1: uuid=e504fca4-15ae-49b6-9253-645c1c4e6fc5
sockets: 1

```
When changed config to. VM start loading and seems netinst iso is loading. But i dont want EFI boot :/
```
agent: 0
balloon: 0
bios: ovmf
boot: order=net0;scsi2;virtio0
cores: 1
efidisk0: local:100/vm-100-disk-0.qcow2,efitype=4m,pre-enrolled-keys=1,size=64M
memory: 2048
meta: creation-qemu=6.0.0,ctime=1638565898
name: librenms
net0: virtio=8E:B8:7E:E9:00:1C,bridge=vmbr0
numa: 0
ostype: l26
scsi2: local:iso/debian-11.1.0-arm64-netinst.iso,media=cdrom,size=327980K
scsihw: virtio-scsi-pci
smbios1: uuid=e504fca4-15ae-49b6-9253-645c1c4e6fc5
sockets: 1
virtio0: local:100/vm-100-disk-1.qcow2,size=10G
```

aanon4, 2021-12-03:
I wasn't aware this has every worked without EFI boot. Has this worked for you before?

pschonmann, 2021-12-04:
No. Never working.This is my first proxmox installation. I'm pretty familiar with proxmox from work :)
I downloaded bullseye lite arm64. Burned onto card, started apt update & upgrade on fresh deb install. Then i run script from GitHub page called RPiOS64-IA-Install.sh.
Everything worked fine. Then issue #31, which is resolved now. But facing problem with this.
When i start VM i see in syslog

```
Dec  4 07:26:05 pimox01 ksmtuned[488]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Dec  4 07:26:07 pimox01 pvedaemon[81197]: start VM 101: UPID:pimox01:00013D2D:00394C43:61AB180F:qmstart:101:root@pam:
Dec  4 07:26:07 pimox01 pvedaemon[58642]: <root@pam> starting task UPID:pimox01:00013D2D:00394C43:61AB180F:qmstart:101:root@pam:
Dec  4 07:26:07 pimox01 systemd[1]: Started 101.scope.
Dec  4 07:26:07 pimox01 systemd-udevd[81216]: Using default interface naming scheme 'v247'.
Dec  4 07:26:07 pimox01 systemd-udevd[81216]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Dec  4 07:26:09 pimox01 kernel: [37553.480104] device tap101i0 entered promiscuous mode
Dec  4 07:26:09 pimox01 kernel: [37553.539624] vmbr0: port 3(tap101i0) entered blocking state
Dec  4 07:26:09 pimox01 kernel: [37553.539636] vmbr0: port 3(tap101i0) entered disabled state
Dec  4 07:26:09 pimox01 kernel: [37553.539833] vmbr0: port 3(tap101i0) entered blocking state
Dec  4 07:26:09 pimox01 kernel: [37553.539838] vmbr0: port 3(tap101i0) entered forwarding state
Dec  4 07:26:09 pimox01 pvedaemon[58642]: <root@pam> end task UPID:pimox01:00013D2D:00394C43:61AB180F:qmstart:101:root@pam: OK
Dec  4 07:26:31 pimox01 pvedaemon[81292]: starting vnc proxy UPID:pimox01:00013D8C:003955B7:61AB1827:vncproxy:101:root@pam:
Dec  4 07:26:31 pimox01 pvedaemon[58642]: <root@pam> starting task UPID:pimox01:00013D8C:003955B7:61AB1827:vncproxy:101:root@pam:
Dec  4 07:27:01 pimox01 pvestatd[1211]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)
Dec  4 07:27:01 pimox01 pvestatd[1211]: CPU flag detection failed, will try again after delay
Dec  4 07:27:05 pimox01 ksmtuned[488]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Dec  4 07:28:05 pimox01 ksmtuned[488]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Dec  4 07:29:01 pimox01 pvestatd[1211]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)
Dec  4 07:29:01 pimox01 pvestatd[1211]: CPU flag detection failed, will try again after delay
Dec  4 07:29:05 pimox01 ksmtuned[488]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
```

aanon4, 2021-12-04:
Okay. I don't think I ever made non EFI booting work (there doesn't seem to be a standard for booting ARM stuff except for EFI). As for lack of console output, try setting it up do use a serial console as that usually works for everything.

You're trying to boot over the network? Not something I've every tried, but if you don't have an EFI disk for EFI boot, the bios  does at least appear to try to do that.

pschonmann, 2021-12-04:
Ok, I installed proxmox again to make sure there was no error in the base installation. The same result. So I'll give UEFI a chance :)

But when upgraded after sh script thats recommend me to run in fresh PVE install apt upgrade i get some permission errors.

root @ pimox01: ~ # apt upgrade -y
Reading package lists ... Done
Building dependency tree ... Done
Reading state information ... Done
Calculating upgrade ... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up pve-manager (7.1-7) ...
Setting up proxmox (7.1-1) ...
ln: failed to create symbolic link '/ boot / pve / vmlinuz': Operation not permitted
ln: failed to create symbolic link '/boot/pve/initrd.img': Operation not permitted

aanon4, 2021-12-04:
Entirely normal as I don't ship a custom kernel.

Wings22Actual, 2021-12-06:
In case you haven't resolved this yet, I was having the same problems (guest has not initialized display yet or unable to boot past the uefi shell); following this configuration guide fixed it for me using an aarch64 ISO: https://youtu.be/LGb7fB1wK4Q.