pimox/pimox7 #33: Error with packages

hellresistor, 2021-12-04:
Cannot run successfully this command. Some tips?

`curl https://raw.githubusercontent.com/pimox/pimox7/master/pimox.sh | sh
`
[image]

aanon4, 2021-12-06:
I just removed using this command from the instructions as it's not been updated in a while, and the alternative provided is must better.

TuxfeatMac, 2021-12-06:
I've allso tried to use images from https://raspi.debian.net/tested-images/  (official debian) but allways run into some issues, or some thing not working propperly after installation... So I used the official 64 beta release from the Raspberry Pi Foundation (https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-11-08/) and it works like a charm. Ether via manual installation via apt after configuring the repo and network settings, or with the RPiOS64-IA-Install.sh autoinstaller.

The dietPi images (all so debian) are working without any errors or caviats so far, look here https://github.com/TuxfeatMac/pimox7-installer/tree/main/dietpi

hellresistor, 2021-12-07:
> Raspberry Pi Foundation (https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-11-08/)

I not test with 'lite' version. I will try. Thank you.

hellresistor, 2021-12-07:
YES!! DONE!

closed 2021-12-07 by hellresistor