pimox/pimox7 #41: RaspiOS arm64 as a VM on pimox7 to run on RPI4

mr12obot, 2021-12-07:
Hi

I am trying to make this work to run on a physical arm64 RPI4 with memory 8GB and it doesnt work.

Is there any guide to make RaspiOS arm64 work as a VM on pimox7 to run on RPI4 ?

Wings22Actual, 2021-12-07:
Are you trying to install pimox using this script onto your RPi, or start a VM in Proxmox-VE after install?
What steps have you tried so far, and are you getting any errors? 

### Installing pimox7:
There are instructions on the [README.md](https://raw.githubusercontent.com/pimox/pimox7/master/README.md);
```
1. Flash and startup the latest image from https://downloads.raspberrypi.org/raspios_arm64/ .
2. sudo -s
3. curl https://raw.githubusercontent.com/pimox/pimox7/master/RPiOS64-IA-Install.sh > RPiOS64-IA-Install.sh
4. chmod +x RPiOS64-IA-Install.sh
5. ./RPiOS64-IA-Install.sh
6. Follow the prompts
```

### Starting a CT / VM
Video showing CT and VM configuration: [PiMox7 - RPi4 - arm64 CT & VM Basic Configuration](https://youtu.be/LGb7fB1wK4Q).
For a VM in Web-UI:
```
1. Click 'Create VM' in upper right
Under 'General'
2. Set name
Under 'OS'
3.  'Do not use any media' (Will be added later)
Under 'System' 
4. BIOS: 'OVMF (UEFI)'
5. Add 'EFI Storage'
6.  'Format: raw'
7.  'Qemu Agent:' Yes
Under Disks
8. 'Format: raw'
Under 'CPU'
9. 'Cores: 2'
10. 'Type: host'
Under 'Memory'
11. 'Memory (MiB): 1024' (can allocate more if your model has the capacity)
Under 'Confirm'
12. 'Finish'

Newly created VM sidebar
Under 'Hardware'
13. 'Remove "CD/DVD Drive (ide2)"'
14. 'Add 'CD/DVD Drive', 'Bus/Device: SCSI, 2', Choose ISO'
Under 'Options'
15. 'Boot Order: scsi2, scsi0'

Start VM

Confirmed to be working using an aarch64/arm64 ISO
```

mr12obot, 2021-12-07:
Yes, I already install pimox using this script onto RPi and confirmed VM working using a Debian arm64 ISO

But when I need running another OS arm 64 as a vm in pimox, it can not boot and error messages is "start pxe over ipv4"

For the example Iam using 2 OS below with not ISO but IMG format. 

1. RaspiOS arm64
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-05-28/

2. HaOS RPI4 arm64
https://github.com/home-assistant/operating-system/releases/download/6.6/haos_rpi4-64-6.6.img.xz

Is there any advice to make this work ?

Sysadminfromhell, 2021-12-07:
> Yes, I already install pimox using this script onto RPi and confirmed VM using a Debian arm64 ISO works like a charm
> 
> But when I need to run another OS arm 64 as a vm in pimox, it can not boot and error messages is "start pxe over ipv4"
> 
> For the example Iam using 2 OS below with not ISO but IMG format.
> 
> 1. RaspiOS arm64
>    https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/2021-10-30-raspios-bullseye-arm64.zip
> 2. HaOS RPI4 arm64
>    https://github.com/home-assistant/operating-system/releases/download/6.6/haos_rpi4-64-6.6.img.xz
> 
> Is there any advice to make this work ?

You have to change the bootorder in the settings:

[image]

change the order to boot to your ISO.
You have to change the CD/DVD to SCSI in order to boot, then you can change the bootorder to scsi of the CD/DVD-ROM.

mr12obot, 2021-12-08:
Already change the order to boot to image and change the cd/dvd to scsi but still can not boot.

Please find below capture for 2 images IMG format and error messages. 

Is there anyone can try successfully this 2 images can run as VM in pimox7 ?

1. RaspiOS arm64
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/2021-10-30-raspios-bullseye-arm64.zip

2. HaOS RPI4 arm64
https://github.com/home-assistant/operating-system/releases/download/6.6/haos_rpi4-64-6.6.img.xz

[image]

[image]

[image]

[image]

Sysadminfromhell, 2021-12-08:
> Basically confirmed can run VM using Ubuntu and Debian arm64 ISO works like a charm but not with this 2 images below
> 
> Already makesure the order to boot to image and the cd/dvd to scsi but still can not boot normally.
> 
> Please find below capture error boot messages and VM config images IMG format
> 
> @Wings22Actual @Sysadminfromhell Any issue or special step if we use images with IMG format ?
> 
> Could you try successfully this 2 images can run as VM in pimox7 ?
> 
> 1. RaspiOS arm64
>    https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/2021-10-30-raspios-bullseye-arm64.zip
> 2. HaOS RPI4 arm64
>    https://github.com/home-assistant/operating-system/releases/download/6.6/haos_rpi4-64-6.6.img.xz
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]

Ah now I see your problem. You’re using an IMG. I guess this doesn’t work out of the box like this because the IMG Files include the “bios” or /boot/ partition. Try a normal arm64 or armfh iso file and install it directly. You could try to import the IMG file as an disk to the vm with qm importdisk. Maybe this will work.

I found this for you (after 5 min googling):
https://forum.proxmox.com/threads/how-to-work-with-img.52986/

mr12obot, 2021-12-08:
@Sysadminfromhell Yes, already try to import with qm importdisk recently. Before import, need to convert it first based on this blog below. But still can not boot normally.

https://williamlam.com/2020/10/how-to-run-raspberry-pi-os-as-a-vm-on-esxi-arm.html

@Wings22Actual @aanon4 @TuxfeatMac Any advice and support help to make this work?

Sysadminfromhell, 2021-12-08:
> @Sysadminfromhell Yes, already try to import with qm importdisk recently. Before import, need to convert it first based on this blog below. But still can not boot normally.
> 
> https://williamlam.com/2020/10/how-to-run-raspberry-pi-os-as-a-vm-on-esxi-arm.html
> 
> @Wings22Actual @aanon4 @TuxfeatMac Any other ideas, how to make it work?

did you try for HAOSS:

https://www.home-assistant.io/installation/alternative

There is a installation way for KVM/Proxmox.

Sysadminfromhell, 2021-12-08:
For your VM Installations you should use the ISO Files for ARM64, debian, ubuntu and other distros provide them so you can install a VM normally with the ARM64 Kernel and functions without any problems. If you need a custom Distro like HAOSS then HAOSS needs to provide a way to install a ARM64 VM.
The Images wont work, because theyre suppossed to be flashed to an SD, the Raspberry doesnt have a BIOS or UEFI in a "normal" way so this wont work afaik.
I try serval ways on my PiMox but it dosnt work with images, but with ISOs and a "normal" installation it works fine.

Rasbpian OS => Debian based, so you can use Debian AM64.
Debian: https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/
Ubuntu Server: https://cdimage.ubuntu.com/releases/20.04/release/ubuntu-20.04.3-live-server-arm64.iso
Ubuntu Server: https://cdimage.ubuntu.com/releases/21.10/release/ubuntu-21.10-live-server-arm64.iso
ArchLinux: http://de4.mirror.archlinuxarm.org/os/ArchLinuxARM-aarch64-latest.tar.gz (https://archlinuxarm.org/platforms/armv8/generic)

But you dont need to use ARM64 Architectur, but in my experiense so far with VMware Esxi and Proxmox if youre using ARM64 the performance is much better then using AMD64 or x86_32/x86_64 because the Pi doesnt have to emulate it.

Wings22Actual, 2021-12-09:
@mr12obot, I haven't managed to get the img to work either, if you're trying to get that RPi img in a VM to run Home Assistant, it looks like they offer a Docker container; so you could install a distro that supports ARM ISOs and docker (like those listed by @Sysadminfromhell) then just set up the Home Assistant Docker container. If you go the Docker route, you could also go with a LXC to save resources compared the full VM route, since we are operating with limited resources on Pi's.

mr12obot, 2021-12-09:
@Sysadminfromhell @Wings22Actual Thanks for the advice. will try to install it as container based

Then how about the image raspios desktop ISO file below. Did you successfully run as a VM as well?

https://downloads.raspberrypi.org/rpd_x86/images/rpd_x86-2021-01-12/2021-01-11-raspios-buster-i386.iso

Sysadminfromhell, 2021-12-09:
> @Sysadminfromhell @Wings22Actual Thanks for the advice. will try to install it as container based
> 
> Then how about the image raspios desktop ISO file below. Did you successfully run as a VM as well? and is there any arm64 version for raspios desktop ISO ?
> 
> https://downloads.raspberrypi.org/rpd_x86/images/rpd_x86-2021-01-12/2021-01-11-raspios-buster-i386.iso

I dont use the x86 architecture, I´m using the ARM64, but to be honest: just use the normal Debian ISO as I mentioned.
Raspbian OS or Raspios is Debian but includes the RPi Kernel and some Packages just for the Rasperry which you do not need in a VM, trust me.

hellresistor, 2021-12-12:
I am getting same problem with .qcow2 images.

guys. Are you getting the 8GB RAM on pimox dashboard?

Sysadminfromhell, 2021-12-12:
> I am getting same problem with .qcow2 images.
> 
> 
> 
> [image]
> 
> 
> 
> 128 K on EFI ?
> 
> 
> 
> guys. Are you getting the 8GB RAM on pimox dashboard?

Again: don't use images, use an ISO like you would install windows. Mount this as a SCSI device to install the OS on the VM Hard Disk.

Kind regards,

MakiseKurisu, 2022-01-09:
You can use images but they have to specifically build with EFI boot. For example I'm using [muVirt](https://gitlab.com/traversetech/muvirt) as an alternative for OpenWrt for now. Most system image intended for single board computer are not built with EBBR boot though.

luqman-pn, 2022-08-31:
Hi, anyone have tried to install windows 10 on pi mox vm? If yes please let me know how to do it
I've tried so many times. nothing works lol