pimox/pimox7 #44: VLAN support on pimox7

bronkoo, 2021-12-18:


Sysadminfromhell, 2021-12-18:
Did you try to activate it via GUI ?

bronkoo, 2021-12-18:
Yes, at first. But it is grayed out in ->Network ->Create ->Linyx Vlan

Sysadminfromhell, 2021-12-19:
Did you activate a Linux bridge first and activate it then or otherwise? Because I can activate it.

bronkoo, 2021-12-19:
Ok, thanks.
Just stumbled over: Default Bridge 'vmbr0' works without any changes - 'VLAN aware' is unmarked - and simple configure the 
'VLAN Tag' in network device for my CT's. Like KISS...

Not an issue currently, but 'VLAN Tag" is still grayed out in ->Network ->Create ->Linux Vlan.

Sysadminfromhell, 2021-12-22:
> Ok, thanks. Just stumbled over: Default Bridge 'vmbr0' works without any changes - 'VLAN aware' is unmarked - and simple configure the 'VLAN Tag' in network device for my CT's. Like KISS...
> 
> Not an issue currently, but 'VLAN Tag" is still grayed out in ->Network ->Create ->Linux Vlan.

afaik you need to create a linux bond before you can create a linux vlan. Read the manual, I guess its described like this.

bronkoo, 2021-12-22:
> afaik you need to create a linux bond before you can create a linux vlan. Read the manual, I guess its described like this.

sorry in the hell ;-), but that doesn't make sense... RTFM for sure.

salman79uk, 2021-12-31:
> Did you try to 

> Did you try to activate it via GUI ?

Hi. For me activating the vlan aware feature on the bridge in the gui and then rebooting causes me to lose access to the pi and requiring me to start again from scratch with a reinstall. Tried a few times.

I don't get this issue on my other non pi nodes full proxmox clusters.

Any ideas.

aanon4, 2021-12-31:
The vlan aware feature is a kernel compile option which the standard pi kernel doesn't enable. You can try this custom kernel which does have it enabled (see https://github.com/pimox/pimox7/issues/28)

MakiseKurisu, 2022-01-09:
Also you may be able to do what you want with `openvswitch-switch` package.

AlwinAntreich, 2022-01-18:
Just to add our trickery to get vlan-aware working, might not apply to your case. We needed to set our nics in `promisc` mode. Sounds like a bug. :)

https://community.mellanox.com/s/question/0D51T00008ansfP/vlan-aware-linux-bridging-is-not-functional-on-connectx4lx-card-unless-manually-put-in-promiscuous-mode