pimox/pimox7 #46: Using PiMox on other SBCs/Server?

stevefan1999-personal, 2022-01-02:
I was wondering if I can use this on a Rock64 I own, or be deployed on a ThunderX2 so this could be a very good alternative for ESXi

MakiseKurisu, 2022-01-09:
Yes. I'm using Pimox on Rock Pi 4B+ and (after a long break) I just managed to get OpenWrt (actually muVirt) and VLAN running.

You will need pimox/qemu-server#1 to run KVM on big.LITTLE ARM processors though.

benep, 2022-01-09:
I'm also using Pimox on a Rock Pi 4B for some weeks. I used Armbian as base and the [manual installation](https://github.com/pimox/pimox7#manual-installation) of Pimox. But I only got LXC containers running (Adguard, Nginx PM, Speedtest). VMs did never start.

@MakiseKurisu can you point what to do on a running system to get VMs run on RockPi 4B?

MakiseKurisu, 2022-01-10:
> I'm also using Pimox on a Rock Pi 4B for some weeks. I used Armbian as base and the [manual installation](https://github.com/pimox/pimox7#manual-installation) of Pimox. But I only got LXC containers [...]

You need to apply the patch like I said, set VM threads to be <=4, then specify `cpu_taskset` to have the same amount of cores in your vm.conf. They have to come from the same cluster, so `1-4` and `5-6` are legal but you cannot specify `4-5` since they are from different cluster.

Then you need to have machine type default, BIOS OVMF, virtio-scsi instead of single for storage controller, and virtgpu for graphic. You should at least see BIOS screen and get waiting on PXE boot with this config. You can then add either EFI enabled disk image (normal single board computer image like Armbian won't work), or use Debian's ARM64 iso to install a system.

benep, 2022-01-10:
So I added the changes from your [pr](https://github.com/pimox/qemu-server/pull/1/files) to my /usr/share/perl5/PVE/QemuServer.pm

Also I added the cpu_taskset info to my /etc/pve/qemu-server/106.conf
```
cores: 4
cpu: host
cpu_taskset: 0-3
```

But without luck:
```
kvm: kvm_init_vcpu: kvm_arch_init_vcpu failed (0): Invalid argument
TASK ERROR: start failed: QEMU exited with code 1
```

I used a debain netinst arm64 iso which is running on my raspberry with pimox.

Kisty, 2022-01-20:
How would one go about building the kernel for another SBC, like an Odroid N2? I have put on Debian 11 bullseye arm64 and have tried to look for kernel headers for 5.10.0-83 (current Pimox kernel version) but can only find -84 and -90 and running the script doesn't install. Having issues with `ceph-dkms` amongst others.

Kisty, 2023-08-23:
Could you document or outline how you build the source so that I can run it on my ODroid N2?