pimox/pimox7 #48: USB Device Passthrough not working

daleksec197, 2022-01-12:
I installed Pimox on raspios_lite_arm64-2021-11-08-Image using the installer-script,
created a new machine configured like the example in docs/VM-configuration.md 
and installed debian 11.2 using the current arm64-net.iso. 

Everything works fine.
It was even possible to setup a working homeassistant-supervised-installation in this VM! 

I tried to use the ConBee II USB Stick as well as some other USB Devices to pass through to the guest os but it is not recognized. In syslog it only tells me a speed mismatch but is trying to connect anyway. 
I had to uncheck "use USB 3" otherwise it tries to connect to a pci device

![Bildschirmfoto 2022-01-12 um 17 43 41](https://user-images.githubusercontent.com/3270463
[image]
4/149183759-fd2bfe4a-2cec-49c2-b083-0806e177dfbf.png)
.

Dattel, 2022-01-15:
i think i got it working after digging through the whole internet :-)

you have to pass the hosts-serial manually to the VM to make it working... 

- stop the vm
- ssh to the pimox
- edit the .conf of your vm using your desired editor

`sudo pico /etc/pve/qemu-server/100.conf`

-add the following line at the end

`args: -chardev tty,path=/dev/ttyACM0,id=conbee -device pci-serial,chardev=conbee`

that passes the /dev/ttyACM0 to the VM - but it will be mapped to /dev/ttyS0 inside the VM

- start the VM

hellresistor, 2022-01-26:
same issue here, thing but with a USB Camera

varoudis, 2022-02-01:
USB passthrough worked for me "out of the box" but has random "resets" that kill the USB device inside the VM.

```
lsusb		
Bus 001 Device 011: ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n
	
pimox dmesg	
		[ 1201.176990] usb 1-1.1: ath9k_htc: USB layer deinitialized
		[ 1211.438991] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
		[26465.440460] vmbr0: port 4(tap105i0) entered disabled state
		[26465.575549] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
		[26470.891366] usb 1-1.1: device descriptor read/64, error -110
		[26486.507421] usb 1-1.1: device descriptor read/64, error -110
		[26486.695427] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
		[26491.883378] usb 1-1.1: device descriptor read/64, error -110
		[26507.499569] usb 1-1.1: device descriptor read/64, error -110
		[26507.687559] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
		[26509.311786] usb 1-1.1: Device not responding to setup address.
		[26511.147381] usb 1-1.1: Device not responding to setup address.
		[26511.355328] usb 1-1.1: device not accepting address 6, error -71
		[26511.435341] usb 1-1.1: reset high-speed USB device number 6 using xhci_hcd
		[26513.059355] usb 1-1.1: Device not responding to setup address.
		[26514.891381] usb 1-1.1: Device not responding to setup address.
		[26515.099417] usb 1-1.1: device not accepting address 6, error -71
		[26515.106829] usb 1-1.1: USB disconnect, device number 6
		[26515.187404] usb 1-1.1: new high-speed USB device number 7 using xhci_hcd
		[26520.299629] usb 1-1.1: device descriptor read/64, error -110
		[26535.915757] usb 1-1.1: device descriptor read/64, error -110
		[26536.103541] usb 1-1.1: new high-speed USB device number 8 using xhci_hcd
		[26541.291731] usb 1-1.1: device descriptor read/64, error -110
		
		
inside the VM, all working perfect until...
		[  101.433151] wlan0: authenticate with 00:2a:10:cf
		[  103.328896] wlan0: send auth to 00:2a:10:cf (try 1/3)
		[  103.335886] wlan0: authenticated
		[  103.340161] wlan0: associate with 00:2a:10:cf (try 1/3)
		[  103.347913] wlan0: RX AssocResp from 00:2a:10:cf (capab=0x431 status=0 aid=12)
		[  103.421820] wlan0: associated
		[  103.539850] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
		[18087.696891] ath: phy0: Unable to remove station entry for: 00:2a:10:cf
		[22266.666353] ath: phy0: Chip reset failed
		[22266.667579] ath: phy0: Unable to reset channel (2447 Mhz) reset status -22
		[22266.669590] ath: phy0: Unable to set channel
```

Sysadminfromhell, 2022-02-01:
I had this also with my USB Flash Drive, the problem was that the drvie became to hot. Is this the case?

varoudis, 2022-02-02:
@Sysadminfromhell I did a bit more testing and I believe its mainly power delivery. Im waiting for a power hub from amazon today to test again.

My wifi usb is running on low tx power (I dont need to blast myself when Im next to it) so Im not sure if its really a power problem but there no excess heat for me.

Check your USB power delivery.

Sysadminfromhell, 2022-02-02:
> @Sysadminfromhell I did a bit more testing and I believe its mainly power delivery. Im waiting for a power hub from amazon today to test again.
>
>
>
> My wifi usb is running on low tx power (I dont need [...]

Yea I thought about it too, but I'm using a different (higher) power cord so I checked and saw that actually the pi used less then it could from the power adapter. So no worry's there. But when I tried to touch the USB it was super hot and refused even to work on my windows PC so I figured that it was simply hot for some reason. But sure thing on you it can be the power problem. Use a active usb hub and you should be fine.

varoudis, 2022-02-02:
I no advance Elec Eng but high heat = lots of Amps right? I think there is something wrong with the stick and requires more power.

Measuring rPi power supply delivery might not be the right way. I read somewhere that rPi have a hard limit per USB port (and overall USB power limit). 
My USB SSD reports around 900mA and the wifi needed 500mA, more than the 1.2A rPi foundation says its ok.

(Its my speculation at this point… Ill report back with the power hub testing)

Dattel, 2022-02-02:
there seems to be different types of USB-Passthrough-problems..
i'm driving a SSD-USB-Drive without any problems on pimix (USB-Powered) with active USB-Passthrough

But the problem described by this issue is, that a special type of USB-Adapters are not presented inside the VM (CONBEE is a usb to serial adapter).... 

So i would suggest not to mix these issues

varoudis, 2022-02-03:
Yes I think pUSB passthrough has some issues as I did the same test on bare metal PI and the USB was ok (I have a test running for hours now (lets not jinx it!)

What would be the best way to approach this and debug it?

bronkoo, 2022-02-20:
> I installed Pimox on raspios_lite_arm64-2021-11-08-Image using the installer-script, created a new machine configured like the example in docs/VM-configuration.md and installed debian 11.2 using the [...]

@daleksec197
Hi, I'm very interested in how do you managed your homeassistant-supervised-installation. I tried it several times inside VM by using the provided [QCOW2 File](https://www.home-assistant.io/hassio/installation/), but it ends up with:

```
kvm: -device usb-ehci,id=ehci,bus=pci.0,addr=0x1: Duplicate ID 'ehci' for device
TASK ERROR: start failed: QEMU exited with code 1
```
Did you installed based on your described arm64-net.iso installation? But how do you managed the supervised part, not only the core?
Best!

Dattel, 2022-02-20:
> @daleksec197 Hi, I'm very interested in how do you managed your homeassistant-supervised-installation. I tried it several times inside VM by using the provided [QCOW2 [...]

some month ago, hard to remember :-D---
i installed a base debian11 arm64 installation and followed these -> https://github.com/home-assistant/supervised-installer

bronkoo, 2022-02-20:
> some month ago, hard to remember :-D--- i installed a base debian11 arm64 installation and followed these -> https://github.com/home-assistant/supervised-installer

Thanks @Dattel !
Did it in this way from your link above inside CT, not VM. Installation was successfully, but the docker container inside didn't came up; think LXC restrictions related, like cgroups...

Will try it now inside VM from scratch by arm64-net.iso installation.

bronkoo, 2022-02-20:
> Will try it now inside VM from scratch by arm64-net.iso installation.

Ok, in this way it works...! Thanks.

Dattel, 2022-02-20:
yeah... i remembered, that installing HA in a CT is not supported, you definitely need a VM..

darekxan, 2022-03-12:
It appears that qemu can actually handle usb passthrough perfectly, but proxmox is unable to configure it properly.

Instead of adding the device by proxmox's hardware page, just edit the vm conf file and do the passthrough through args
```args: -device qemu-xhci -device usb-host,vendorid=0x10c4,productid=0xea60```

Dattel, 2022-03-12:
Brilliant... 
Much better than only passing the serial port, as i suggested earlier..

mr12obot, 2022-03-31:
> i think i got it working after digging through the whole internet :-) I'm not sure, what causes these trouble for us, maybe the Raspberry-revision? i've got a RPi4 with 8G.
> 
> you have to pass the hosts-serial manually to the VM to make it working... (Inspired by [this thread](https://serverfault.com/questions/872238/qemu-and-serial-ports-on-the-guest-os))
> 
> * ssh to the pimox
> * stop the desired vm
> * edit the .conf of your vm using your desired editor
>   `sudo pico /etc/pve/qemu-server/100.conf`
> * add the following line at the end
>   `args: -chardev serial,path=/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2466526-if00,id=conbee -device pci-serial,chardev=conbee`
> * start the desired vm
> 
> that passes the Conbee2 to the VM - but it will be mapped to /dev/ttyS0 inside the VM
> 
> * go to HA and activate the "Zigbee Home Automation" implementation - your conbee should be instant available. :-D
> 
> If there is an easier way, let me know - i'm very interessted in...

Hi @Dattel ,

Already try your suggestion and it works like charm for activate the "Zigbee Home Automation" integration. 

But can not use for "Zigbee2Mqtt". Already change to /dev/ttyS0 for configuration.yaml in Zigbe2Mqtt addon but failed to start.

 Could you advise how to make this work for Zigbee2Mqtt ?

Dattel, 2022-03-31:
Sadly not... 
i switched over to the suggested fix from @darekxan because that seems to me a much cleaner solution by passing a device instead of just tty port to the vm. 
Maybe that helps you.

noogugae, 2022-04-02:
> It appears that qemu can actually handle usb passthrough perfectly, but proxmox is unable to configure it properly.
> 
> Instead of adding the device by proxmox's hardware page, just edit the vm conf file and do the passthrough through args `args: -device qemu-xhci -device usb-host,vendorid=<0xVendorid>,productid=<0xProductid>` filling the vendor id and product id with the values obtained either from gui (you can see it when adding the usb passhtrough) or from`lsusb` on host machine. Remember to prefix it with `0x`.

Thanks. This method work very well.
But I have a problem. I have 2 zigbee stick and both have same vender&product ID. So I want to pass through them by using serial path such as '/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-port0'.  How do I pass through them?

darekxan, 2022-04-03:
> I have 2 zigbee stick and both have same vender&product ID.

According to [QEMU USB docs](https://qemu-project.gitlab.io/qemu/system/devices/usb.html)

> **usb-host,hostbus=bus,hostaddr=addr**
Pass through the host device identified by bus and addr
> **usb-host,vendorid=vendor,productid=product**
Pass through the host device identified by vendor and product ID

You should use the first option, adding it twice. Bus and addres could be determined by `lsusb`

noogugae, 2022-04-03:
> > I have 2 zigbee stick and both have same vender&product ID.
> 
> According to [QEMU USB docs](https://qemu-project.gitlab.io/qemu/system/devices/usb.html)
> 
> > **usb-host,hostbus=bus,hostaddr=addr**
> > Pass through the host device identified by bus and addr
> > **usb-host,vendorid=vendor,productid=product**
> > Pass through the host device identified by vendor and product ID
> 
> You should use the first option, adding it twice, one per device instance. Bus and address could be determined by `lsusb`

Hi. I tried it.
I tried each one seperately and  It was succesfully passed through. but If I tried both at same time, only one is successful.
Do you have any idea?

They have same ID, so I can't use ID for passthrough
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 001 Device 004: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 006: ID 1a86:7523 QinHeng Electronics CH340 serial converter**
Bus 001 Device 003: ID 13fe:6000 Kingston Technology Company Inc. UX200P
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=ch341, 12M
        |__ Port 4: Dev 4, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M

I added below line at 100.conf
args: -device qemu-xhci -device usb-host,hostbus=1,hostport=1.3
args: -device qemu-xhci -device usb-host,hostbus=1,hostport=1.4

but there is only one zigbee stack at sub os.
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 002 Device 002: ID 1a86:7523 QinHeng Electronics CH340 serial converter**
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB Keyboard
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

cloubier, 2022-04-08:
> i think i got it working after digging through the whole internet :-) I'm not sure, what causes these trouble for us, maybe the Raspberry-revision? i've got a RPi4 with 8G.
>
> you have to pass the [...]

> i think i got it working after digging through the whole internet :-) I'm not sure, what causes these trouble for us, maybe the Raspberry-revision? i've got a RPi4 with 8G.
>
> you have to pass the [...]

@Dattel Hi, but it's impossible for me to launch the VM after adding the line "args:..."... any ideas ?

Dattel, 2022-04-08:
hi @cloubier,
the easiest way would be to the solution from @darekxan 
i think it's the much cleaner way and you do not need to dig in error logs to resolve the issue.

cloubier, 2022-04-09:
> hi @cloubier, the easiest way would be to the solution from @darekxan i think it's the much cleaner way and you do not need to dig in error logs to resolve the issue.

Ohhh thank you very much guys !!!! @darekxan @Dattel 2weeks i try to resolve this !!!!

darekxan, 2022-09-28:
> Since the usb device passthrough is working well, is it possible the bluetooth built-in RPi module passthrough as well?
According to [these docs (🔎bluetooth)](https://students.mimuw.edu.pl/~zbyszek/asm/QEMU_Emulator_User_Documentation.html) 
it  _should_ be possible, however I haven't tested it. I've seen multiple QEMU docs around, and the official ones don't seem to include relevant options.

I would try adding:
`-bt hci,host`
to the invocation.

mr12obot, 2022-09-29:
> > Since the usb device passthrough is working well, is it possible the bluetooth built-in RPi module passthrough as well?
> 
> @mr12obot According to [these docs (🔎bluetooth)](https://students.mimuw.edu.pl/~zbyszek/asm/QEMU_Emulator_User_Documentation.html) it _should_ be possible, however I haven't tested it. I've seen multiple QEMU docs around, and the official ones don't seem to include relevant options.
> 
> I would try adding: `-bt hci,host` to the invocation.

@darekxan already try adding `args: -bt hci,host` to the vm conf file but cannot work. 

any complete command or could you help to tested it as well?

mr12obot, 2022-09-30:
> I would try adding: `-bt hci,host` to the invocation.

@darekxan already try adding `args: -device qemu-xhci -device -bt hci,host` to the vm conf file but cannot work.

the guest machine have error below:
`kvm: hci,host: Could not open 'hci,host': No such file or directory
TASK ERROR: start failed: QEMU exited with code 1`

any complete command or could you help to tested it as well?

El-mou33, 2022-09-30:
Same problem. I read a [possible solution in standard proxmox.](https://forum.proxmox.com/threads/guest-internal-error-when-passing-through-pcie.99239/) changing pcie_aspm to off `pcie_aspm=off` in the proxmox kernel. But I cannot access to the kernel to change that. Any ideas? When I use the command: `sudo nano /etc/default/grub` a blank pafge appears so I cannot edit nothing, becouse there is nothig to eddit like in normal proxmox or [in this example.](https://incero.com/pcie_aspmoff.php)

El-mou33, 2022-10-01:
I'm a bit new i pimox, I have a RPI4, and Home Assistant OS running in my pimox. The problem is that when I attach the usb ports the VM doesn't starts. I added this lines but I think I'm missing something, could anyone help?

`args: -device qemu-xhci -device usb-host,vendorid=<0xITead>,productid=<0xusb-Sonoff_Zigbee_3.0_USB_Dongle_Plus>

args: -chardev serial,path=/dev/serial/by-id/usb-Sonoff_Zigbee_3.0_USB_Dongle_Plus,id=sonoff -device pci-serial,chardev=ITead`

El-mou33, 2022-10-09:
> It appears that qemu can actually handle usb passthrough perfectly, but proxmox is unable to configure it properly.
>
> Instead of adding the device by proxmox's hardware page, just edit the vm conf [...]

Where do you find the vendor id?

This is what I'm trying to use, but the error persist. I don't know what I'm doing wrong:
`args: -device qemu-xhci -device usb-host,vendorid=<0xITead>,productid=<0xusb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_22ff7c23b919ec11b1dd7fe5f01c6278-if00-port0>`

bronkoo, 2023-04-23:
> Already try your suggestion and it works like charm for activate the "Zigbee Home Automation" integration.
> 
> But can not use for "Zigbee2Mqtt". Already change to /dev/ttyS0 for configuration.yaml in Zigbe2Mqtt addon but failed to start.
> 
> Could you advise how to make this work for Zigbee2Mqtt ?

[Maybe](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html) "Your adapter requires additional configuration parameters. Check [supported Adapters](https://www.zigbee2mqtt.io/guide/adapters/) section to find out if your adapter requires extra parameters (eg. ConBee II / RaspBee II)."

```
serial:
  adapter: deconz
  port: /dev/ttyACM0
```

FliegerKrieger, 2023-06-08:
I cannot seem to get my ConBee II working on HA.
I´m able to start the vm and reconfigure the implementation. But it cannot see the addon.
Can someone help me find the vendorid and productid needed? Also the path /dev/serial/by-id is missing.

Using the same configuration with a pi3 and direct install of HAos works fine.

bronkoo, 2023-06-08:
> Can someone help me find the vendorid and productid needed? Also the path /dev/serial/by-id is missing.

Checkout the `lsusb` command in posts above for the *IDs and `ls -la /dev/serial/by-path/` for the serial port inside VM.

FliegerKrieger, 2023-06-09:
> > Can someone help me find the vendorid and productid needed? Also the path /dev/serial/by-id is missing.
> 
> Checkout the `lsusb` command in posts above for the *IDs and `ls -la /dev/serial/by-path/` for the serial port inside VM.

Hi, thank you it worked now.

For anyone having problems, looking at how the lsusb command works helped me.
It outputs something like this:
`Bus 001 Device 003: ID vendorid:productid Dresden Elektronik ZigBee gateway [ConBee II]`.

So you add this line at the end of `/etc/pve/qemu-server/vmid.conf` and it just gets recognized for me:
 `args: -device qemu-xhci -device usb-host,vendorid=<0xVendorid>,productid=<0xProductid>`

mccxiv, 2023-07-21:
> args: -device qemu-xhci -device usb-host,vendorid=0x<Vendorid>,productid=0x<Productid>

This worked for me BUT only when plugging my USB device into the Pi's USB 2.0 ports. No luck with USB 3.0 so far...

Anyone have luck passing through usb 3?

KeviinCosmos, 2023-08-10:
> i think i got it working after digging through the whole internet :-) I'm not sure, what causes these trouble for us, maybe the Raspberry-revision? i've got a RPi4 with 8G.
>
> you have to pass the [...]

I get this error after adding my device.
[image]
`args: -chardev serial,path=/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_c60824c20d45ec11b42a98fd6f14af06-if00-port0`

Vovanchik, 2023-12-17:
> @mr12obot According to [these docs (🔎bluetooth)](https://students.mimuw.edu.pl/~zbyszek/asm/QEMU_Emulator_User_Documentation.html) it _should_ be possible, however I haven't tested it. I've seen [...]

@mr12obot have you resolved an issue with passing built-in Bluetooth on Raspi 4 connected to UART?

Vovanchik, 2023-12-17:
> But

@mr12obot have you resolved an issue with built-in Bluetooth device?

LeDuble, 2024-07-31:
> It appears that qemu can actually handle usb passthrough perfectly, but proxmox is unable to configure it properly.
>
> Instead of adding the device by proxmox's hardware page, just edit the vm conf [...]

ty this worked!

tsubus, 2025-06-27:
> It appears that qemu can actually handle usb passthrough perfectly, but proxmox is unable to configure it properly.
>
> Instead of adding the device by proxmox's hardware page, just edit the vm conf [...]

this worked! i just wish i found your comment 2 hours ago :D