pimox/pimox7 #49: QEMU/KVM cannot detect CPU flags on ARM (aarch64) ?

varoudis, 2022-01-13:
I see these over and over. Are these ok considering the "hacky" nature of pimox?

`Jan 13 22:42:45 RPi4-PVE pvestatd[1081]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)
Jan 13 22:42:45 RPi4-PVE pvestatd[1081]: CPU flag detection failed, will try again after delay
Jan 13 22:42:46 RPi4-PVE ksmtuned[503]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Jan 13 22:43:47 RPi4-PVE ksmtuned[503]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Jan 13 22:44:45 RPi4-PVE pvestatd[1081]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)
Jan 13 22:44:45 RPi4-PVE pvestatd[1081]: CPU flag detection failed, will try again after delay
Jan 13 22:44:47 RPi4-PVE ksmtuned[503]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Jan 13 22:45:47 RPi4-PVE ksmtuned[503]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Jan 13 22:46:45 RPi4-PVE pvestatd[1081]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)`

At the time NO vm or ct was running for quite a while.

Thanks for the hard work! Its amazing.

Sysadminfromhell, 2022-01-14:
> I see these over and over. Are these ok considering the "hacky" nature of pimox?
> 
> ```
> Jan 13 22:42:45 RPi4-PVE pvestatd[1081]: CPU flag detection failed, will try again after delay
> Jan 13 22:42:46 RPi4-PVE ksmtuned[503]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
> Jan 13 22:43:47 RPi4-PVE ksmtuned[503]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
> Jan 13 22:44:45 RPi4-PVE pvestatd[1081]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)
> Jan 13 22:44:45 RPi4-PVE pvestatd[1081]: CPU flag detection failed, will try again after delay
> Jan 13 22:44:47 RPi4-PVE ksmtuned[503]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
> Jan 13 22:45:47 RPi4-PVE ksmtuned[503]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
> Jan 13 22:46:45 RPi4-PVE pvestatd[1081]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)
> ```
> 
> At the time NO vm or ct was running for quite a while.
> 
> Thanks for the hard work! Its amazing.

Hi there and welcome here.

the pvestad is kind of nature in terms that the Raspberry Pi uses a ARM. The Proxmox tries to find the CPU Flags, this isnt supported by Proxmox for ARM CPUs. You can't get rid of it.
The KSMTuned error you can get rid off if youre using a custom kernel, as mentioned in https://github.com/pimox/pimox7/issues/28

kind regards

varoudis, 2022-01-14:
thanks. I was thinking of playing around with a kernel after seeing #28 yesterday. 

last time I really spent time building kernels from scratch was probably back in 2002-2004?!? Now I dont even know if cross compiling is mainstream or I need to do all the building on pi4. Ill investigate.

Is a kernel config file offered with the PiOS64 we are running for pimox7? (Probably yes I assume)

puccaso, 2022-07-30:
would so love that