pimox/pimox7 #54: Update to the next PVE-7.1-10

Sysadminfromhell, 2022-02-24:
Hey there,

long time, no see :)

Short question:

Im using the PVE 7.1-10 on my Intel Nuc as a x64 Maschine for VMs.
Now i setup a HA Cluster with the PiMox (7.1-8).

I have the problem, that the migration between both nodes fails, the support of Proxmox said that it is likely because of the different versions of the qemu-server and pve-qemu-kvm.

Version on the 7.1-10: qemu-server (7.1-4), pve-qemu-kvm (6.1.1-2)
Version on the 7.1-8:  qemu-server (7.1-4rpve1), pve-qemu-kvm (6.0.0-4)

Also the support recommend to upgrade the kernel of the PiMox to the same as in my Intel Nuc: 
proxmox-ve: 7.1-1 (running kernel: **5.15.19-2-pve**)

On the Pi its still:
proxmox-ve: 7.1-1 (running kernel: **5.10.83-1-pve**)

Is there any chance u could upgrade the packages and also provide the new 5.15. kernel?

Kind regards,

profblackjack, 2022-02-24:
An update to 7.1-10 would be good for my OCD, but It's not likely that migrating an x86 architecture VM to an arm host would work even with matching versions, at least not without a lot of work.

@Sysadminfromhell  You should stick to migrating between nodes of the same architecture. If you only have the 2 nodes, then I wouldn't expect migration features to work for you.

Sysadminfromhell, 2022-02-24:
> versions,

As far as I know it should be possible if you remove cpu: and add arch:aarch64.
The VMs are booting if I migrate them offline (powered off) but not Live Migration (powered on)

aanon4, 2022-02-25:
Just updated the repo to 7.1-10.
There's also pve-kernel-5.10.95-1 (synced with raspberry's official kernel release)

Sysadminfromhell, 2022-02-25:
> Just updated the repo to 7.1-10. There's also pve-kernel-5.10.95-1 (synced with raspberry's official kernel release)

Thansks for your great work!

But i cant install the new Kernel:

`PiMox01:~# apt install pve-kernel-5.10.95-1-pve
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 pve-kernel-5.10.95-1-pve : Conflicts: linux-image
E: Unable to correct problems, you have held broken packages.`

aanon4, 2022-02-25:
At some point in the past you will have held back updates to the kernel to avoid things breaking (a good move!). You now need to undo this:

  apt-mark unhold raspberrypi-kernel

Sysadminfromhell, 2022-02-28:
> At some point in the past you will have held back updates to the kernel to avoid things breaking (a good move!). You now need to undo this:
> 
> apt-mark unhold raspberrypi-kernel

I did that, now it works. Thanks!

varoudis, 2022-03-01:
@aanon4 
So is it safe to apt upgrade? (Im afraid to be honest) my upgrade list is:

```
root@RPi4-PVE:~# apt list --upgradable
Listing... Done
arandr/stable,stable 0.1.10-1.1+rpt19 all [upgradable from: 0.1.10-1.1+rpt17]
bsdextrautils/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
bsdutils/stable-security 1:2.36.1-8+deb11u1 arm64 [upgradable from: 1:2.36.1-8]
chromium-browser-l10n/stable,stable 98.0.4758.106-rpt1 all [upgradable from: 95.0.4638.78-rpt5]
chromium-browser/stable 98.0.4758.106-rpt1 arm64 [upgradable from: 95.0.4638.78-rpt5]
chromium-codecs-ffmpeg-extra/stable 98.0.4758.106-rpt1 arm64 [upgradable from: 95.0.4638.78-rpt5]
eject/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
fdisk/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
firmware-atheros/stable,stable 1:20210315-3+rpt5 all [upgradable from: 1:20210315-3+rpt4]
firmware-brcm80211/stable,stable 1:20210315-3+rpt5 all [upgradable from: 1:20210315-3+rpt4]
firmware-libertas/stable,stable 1:20210315-3+rpt5 all [upgradable from: 1:20210315-3+rpt4]
firmware-misc-nonfree/stable,stable 1:20210315-3+rpt5 all [upgradable from: 1:20210315-3+rpt4]
firmware-realtek/stable,stable 1:20210315-3+rpt5 all [upgradable from: 1:20210315-3+rpt4]
gir1.2-polkit-1.0/stable 0.105-31+rpt1+deb11u1 arm64 [upgradable from: 0.105-31+rpt1]
gui-pkinst/stable 0.6 arm64 [upgradable from: 0.5]
libblkid1/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
libcamera-apps/stable 0~git20220131+7ac5197-1 arm64 [upgradable from: 0~git20211021+2a38ae9-1]
libcamera-tools/stable 0~git20220201+e96d0201-1 arm64 [upgradable from: 0~git20211020+f4070274-3]
libcamera0/stable 0~git20220201+e96d0201-1 arm64 [upgradable from: 0~git20211020+f4070274-3]
libcryptsetup12/stable-security 2:2.3.7-1+deb11u1 arm64 [upgradable from: 2:2.3.5-1]
libexpat1-dev/stable-security 2.2.10-2+deb11u2 arm64 [upgradable from: 2.2.10-2]
libexpat1/stable-security 2.2.10-2+deb11u2 arm64 [upgradable from: 2.2.10-2]
libfdisk1/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
libjavascriptcoregtk-4.0-18/stable-security 2.34.6-1~deb11u1 arm64 [upgradable from: 2.34.3-1~deb11u1]
libmount1/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
libmutter-7-0/stable 1:3.38.6-2~deb11u1+rpt3 arm64 [upgradable from: 1:3.38.6-2~deb11u1+rpt1]
libnss3/stable-security 2:3.61-1+deb11u2 arm64 [upgradable from: 2:3.61-1+deb11u1]
libpixman-1-0/stable 0.40.0-1+rpt1 arm64 [upgradable from: 0.40.0-1]
libpolkit-agent-1-0/stable 0.105-31+rpt1+deb11u1 arm64 [upgradable from: 0.105-31+rpt1]
libpolkit-gobject-1-0/stable 0.105-31+rpt1+deb11u1 arm64 [upgradable from: 0.105-31+rpt1]
libproxmox-acme-perl/unknown 1.4.1 all [upgradable from: 1.4.0]
libproxmox-acme-plugins/unknown 1.4.1 all [upgradable from: 1.4.0]
libpve-access-control/unknown 7.1-6 all [upgradable from: 7.1-5]
libpve-common-perl/unknown 7.1-3 all [upgradable from: 7.0-14]
libpve-guest-common-perl/unknown 4.1-1 all [upgradable from: 4.0-3]
libpve-http-server-perl/unknown 4.1-1 all [upgradable from: 4.0-4]
libpve-rs-perl/unknown 0.5.1 arm64 [upgradable from: 0.4.4]
libpve-storage-perl/unknown 7.1-1 all [upgradable from: 7.0-15]
libraspberrypi-bin/stable 1:2+git20211125~155417+14b90ff-3 arm64 [upgradable from: 1:2+git20211125~155417+14b90ff-2]
libraspberrypi-dev/stable 1:2+git20211125~155417+14b90ff-3 arm64 [upgradable from: 1:2+git20211125~155417+14b90ff-2]
libraspberrypi-doc/stable,stable 1:2+git20211125~155417+14b90ff-3 all [upgradable from: 1:2+git20211125~155417+14b90ff-2]
libraspberrypi0/stable 1:2+git20211125~155417+14b90ff-3 arm64 [upgradable from: 1:2+git20211125~155417+14b90ff-2]
libsasl2-2/stable-security 2.1.27+dfsg-2.1+deb11u1 arm64 [upgradable from: 2.1.27+dfsg-2.1]
libsasl2-modules-db/stable-security 2.1.27+dfsg-2.1+deb11u1 arm64 [upgradable from: 2.1.27+dfsg-2.1]
libsmartcols1/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
libsmbclient/stable-security 2:4.13.13+dfsg-1~deb11u3 arm64 [upgradable from: 2:4.13.13+dfsg-1~deb11u2]
libssl1.1/stable 1.1.1k-1+deb11u1+rpt1 arm64 [upgradable from: 1.1.1k-1+deb11u1]
libuuid1/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
libwbclient0/stable-security 2:4.13.13+dfsg-1~deb11u3 arm64 [upgradable from: 2:4.13.13+dfsg-1~deb11u2]
libwebkit2gtk-4.0-37/stable-security 2.34.6-1~deb11u1 arm64 [upgradable from: 2.34.3-1~deb11u1]
lxinput/stable 0.3.5-1+rpi13 arm64 [upgradable from: 0.3.5-1+rpi12]
lxpanel-data/stable,stable 0.10.1-2+rpt12 all [upgradable from: 0.10.1-2+rpt11]
lxpanel/stable 0.10.1-2+rpt12 arm64 [upgradable from: 0.10.1-2+rpt11]
lxplug-volumepulse/stable 0.12 arm64 [upgradable from: 0.10]
mount/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
mutter-common/stable,stable 1:3.38.6-2~deb11u1+rpt3 all [upgradable from: 1:3.38.6-2~deb11u1+rpt1]
mutter/stable 1:3.38.6-2~deb11u1+rpt3 arm64 [upgradable from: 1:3.38.6-2~deb11u1+rpt1]
openssl/stable 1.1.1k-1+deb11u1+rpt1 arm64 [upgradable from: 1.1.1k-1+deb11u1]
pixflat-icons/stable,stable 0.9 all [upgradable from: 0.8]
policykit-1/stable 0.105-31+rpt1+deb11u1 arm64 [upgradable from: 0.105-31+rpt1]
proxmox-backup-client/unknown 2.1.5-1 arm64 [upgradable from: 2.1.2-1]
proxmox-backup-file-restore/unknown 2.1.5-1 arm64 [upgradable from: 2.1.2-1]
proxmox-widget-toolkit/unknown 3.4-7 all [upgradable from: 3.4-4]
pve-container/unknown 4.1-4 all [upgradable from: 4.1-3]
[... 3 lines cut ...]
[... cut ...]
samba-dsdb-modules/stable-security 2:4.13.13+dfsg-1~deb11u3 arm64 [upgradable from: 2:4.13.13+dfsg-1~deb11u2]
samba-libs/stable-security 2:4.13.13+dfsg-1~deb11u3 arm64 [upgradable from: 2:4.13.13+dfsg-1~deb11u2]
smbclient/stable-security 2:4.13.13+dfsg-1~deb11u3 arm64 [upgradable from: 2:4.13.13+dfsg-1~deb11u2]
util-linux/stable-security 2.36.1-8+deb11u1 arm64 [upgradable from: 2.36.1-8]
xserver-common/stable,stable 2:1.20.11-1+rpt1+deb11u1 all [upgradable from: 2:1.20.11-1+rpt1]
xserver-xorg-core/stable 2:1.20.11-1+rpt1+deb11u1 arm64 [upgradable from: 2:1.20.11-1+rpt1]
root@RPi4-PVE:~#
```

Sysadminfromhell, 2022-03-01:
> @aanon4 So is it safe to apt upgrade? (Im afraid to be honest) my upgrade list is:
>
> ```
> root@RPi4-PVE:~# apt list --upgradable
> Listing... Done
> arandr/stable,stable 0.1.10-1.1+rpt19 all [upgradable [...]

its 100% safe, i did it the same way.

aanon4, 2022-03-02:
@varoudis I absolutely understand the anxiety, but it's safe. If you want to do it bit-by-by you can, by upgrading packages one at a time by hand at the comment line. They will pull in their dependencies and you can check them before you confirm. The only thing to watch is if these bit-by-bit upgrades say they'll REMOVE (it says it in big letters) packages ... that's usually very bad.

varoudis, 2022-03-02:
just to confirm, all good here. I did a normal apt update and upgrade.

thank you both!