pimox/pimox7 #55: Qemu

jhmc93, 2022-02-24:
I have qemu error if I run a windows vm
Tried a Linux one and I got “screen needs to be configured” I tried q35 machine but I get an ehci error can someone please help

Sysadminfromhell, 2022-02-24:
> I have qemu error if I run a windows vm Tried a Linux one and I got “screen needs to be configured” I tried q35 machine but I get an ehci error can someone please help

Windows is not supported to ARM64m if you try to use ARM64 you have to download a fixed Image from Microsoft.
In our community you can talk to @the-wondersmith he had a working VM for W11 i guess it was with ARM64, but it was a pain to makde it work. I couldnt figure it out and gave up at some point, I also lost his shared files because my Windows PC broke one day.

jhmc93, 2022-02-24:
what about the Linux error I'm getting