pimox/pimox7 #57: Operation not permitted while setup, cannot run lxc/vm

darekxan, 2022-03-10:
I download latest bullseye 64bit and follow the instructions. After restart, th

`
pi@pimox:~ $ sudo -s
root@pimox:/home/pi# apt upgrade -y
[...]
ln: failed to create symbolic link '/boot/pve/vmlinuz': Operation not permitted
ln: failed to create symbolic link '/boot/pve/initrd.img': Operation not permitted
[...]
`
I can login into gui, he summary looks like that:
[image]

running contaienrs fails with:

sync_wait: 36 An error occurred in another process (expected sequence number 7)
__lxc_start: 2073 Failed to spawn container "101"
TASK ERROR: startup for container '101' failed

These are basic debian11 and alpine templates downloaded from CT Templates
[image]

I cannot make it to work, tried googling but nothing relevant pops out.

darekxan, 2022-03-10:
Ok maybe this is related to these templates being amd64?

varoudis, 2022-03-17:
also images for apple M1 works most of the times.

darekxan, 2022-03-18:
That would be right. Ive successfully started debian arm64 cloud image

closed 2022-03-18 by darekxan