pimox/pimox7 #60: Install script

bouldinnathan, 2022-03-27:
Install script does not work on fresh install of the 64 bit Raspbian OS from Raspberry pi imager.

Reproduction steps:
Burn latest 64 bit Raspbian image from Raspberry pi imager.
Use instructions.

Error:
Package proxmox-ve does not install correctly because proxmox-manager is not configure correctly 

Work around:
follow instructions about remove localhost and other extra entries...
https://forum.proxmox.com/threads/problem-with-installing-proxmox-ve-on-debian.44316/
apt install proxmox-ve 
then run the script

TuxfeatMac, 2022-03-27:
Thanks for the feedback bouldinnathan! I will take a look at it.

Be patient, it will take some time, since im working on other projects (non github relarted) at the moment.

bouldinnathan, 2022-03-28:
Thanks. I hope if some one else has the same problem this might help.

BadIdeasExecutedPoorly, 2022-04-01:
@bouldinnathan  Thank-you for posting this. I have the exact same issue. Could you please post your hosts file? I'm not sure exactly which steps to take in the forum thread you posted. 

Cheers!

bouldinnathan, 2022-04-01:
@BadIdeasExecutedPoorly 

I think I removed "127.0.0.1 localhost". I think removed everything but "192.168.86.136  uhura". (uhura was the name of my RPI)

Then...
apt install proxmox-ve
then run the script (Setting up the host name to be "uhura" and used "192.168.86.136"  as the ip)

After everything is installed correctly this is my resulting hosts file.
`root@uhura:~# cat /etc/hosts
127.0.0.1       localhost
192.168.86.136  uhura`