pimox/pimox7 #63: Home Assistant OS VM not booting

edoardo-conti, 2022-03-28:
I’m trying to create a Home Assistant OS virtual machine on PimoxV7 running on my RPi 4 with 4GB of RAM (I don’t know if it’s even possibile what I’m trying to do)

At the moment I tried 2 different attempts:

- Create a VM with a SCSI cd/dvd drive with the [HA OS image for RPi 4](https://www.home-assistant.io/installation/raspberrypi) (haos_rpi4-64-7.5.img) to be installed on a disk;
- Create a VM and import the qcow2 image provided by the [alternative installations page](https://www.home-assistant.io/installation/alternative) (haos_ova-7.5.qcow2)

Neither method allows me to boot the image, ending on the UEFI shell. I even tried to change the boot order as described by [this guide](https://pve.proxmox.com/wiki/OVMF/UEFI_Boot_Entries) without success.

I leave some screenshots of the vm’s hardware and configuration:
[image]
[image]
[image]
[image]
[image]

Could someone help me? Even just letting me know if it’s theoretically possible?

darekxan, 2022-03-28:
> * Create a VM and import the qcow2 image provided by the [alternative installations page](https://www.home-assistant.io/installation/alternative) (haos_ova-7.5.qcow2)

This appears to be x86_64 image, so is a no-go. 

Not sure about the rpi image, you might have some luck setting the emulated machine as rpi3 (in config file, GUI is not properly configured to show arm machines)

What I did to run HA on pimox is to download a debian cloud image, add user password via CloudInit, then follow the 'supervised' installation method.

edoardo-conti, 2022-03-29:
> Not sure about the rpi image, you might have some luck setting the emulated machine as rpi3 (in config file, GUI is not properly configured to show arm machines)
Can you show me how to config the vm as rpi3? I'll give a try!

> What I did to run HA on pimox myself is to download a debian cloud image, add user password via CloudInit, then follow the 'supervised' installation method. You will have issues with USB passthrough, eg. ZigBee adapter.
What is Debian cloud image and why choose it over standard debian 11.3.0 netinst image?
USB Issues are fixable or not atm?

darekxan, 2022-03-29:
Debian cloud image is preinstalled qcow2 that runs in seconds. No need to
install. Its just faster to do it this way.
https://cloud.debian.org/images/cloud/bullseye/latest/

hastmu, 2022-04-10:
hi it works with arm64 uefi image native starting with haos 8.0, haos-8.0.dev20220325 works at my setup fine.
my source haos_generic-aarch64-8.0.dev20220325.img

edoardo-conti, 2022-04-10:
Great news! It's still a release candidate or I am wrong? 

the latest version available is that of April 4th, I'll try that next week!

hastmu, 2022-04-11:
hi - yes it is a RC, but i found no issues for my setup, which was a restore of a full backup from a x86_64 installation.
So for me all good - no change planned until auto-update will proceed to official release version 8.0+

edoardo-conti, 2022-04-16:
@hastmu how did you load the image? it keeps to not work for me even with the latest 8.0 image... 

can you even show the "Options" view of the HA's vm, please?

hastmu, 2022-04-19:
hi - sure i used "qm importdisk" to import the img, the settings of the vm are visible in the screen shot above.

edoardo-conti, 2022-04-20:
Ok, I managed to get a working vm of HA OS! 

now i was trying to do the same thing with a [Raspberry Pi OS Lite image](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit), has anyone succeeded with it? 

I loaded the img like a SCSI cd/dvd drive with a OVMF (uefi) bios but I can't get it booting...

luckydonald, 2022-05-10:
```
wget https://github.com/home-assistant/operating-system/releases/download/8.0.rc4/haos_generic-aarch64-8.0.rc4.img.xz
xz --decompress haos_generic-aarch64-*.img.xz
mv haos_generic-aarch64-8.0.rc4.img /var/lib/vz/template/iso/haos_generic-aarch64-8.0.rc4.img

# not sure if this needs sudo. It did work in the shell on the website.
qm importdisk 100 /var/lib/vz/template/iso/haos_generic-aarch64-8.0.rc4.img local
```
Then make sure that's the actually "HDD", remove any other volumes including any CD drive installer iso, and make sure you have the disk imported with `importdisk` set to be a boot drive.

hastmu, 2022-05-29:
Auto update - kicked in yesterday, now stable on 8.1 official release. no issues so far.

pbix, 2022-05-30:
Thanks to the input from all the folks on this thread I too was able to get HA running as a VM under Proxmox on my RPi4.   Once properly installed it built from my backup without issue and has been running for days.  Performance seems unchanged to me from when HA was solo installed on the RPI4.  For the novice Proxmox users I below give the details on how to get this working:

1) Get Proxmox installed on your RPi4 with at least 64GB flash SDI card.  Many guides and videos out there on this topic.

From the Proxmox web interface select "Shell" at the top level and execute the following (v8.1 as an example)
2) wget https://github.com/home-assistant/operating-system/releases/download/8.1/haos_generic-aarch64-8.1.img.xz
3) xz --decompress haos_generic-aarch64-*.img.xz
4) mv haos_generic-aarch64-*.img /var/lib/vz/template/iso
5) qm create 101 --bios ovmf --cores 2  --memory 4096 --scsihw virtio-scsi-pci -net0 virtio,bridge=vmbr0
6) qm importdisk 101 /var/lib/vz/template/iso/haos_generic-aarch64-*.img local
7) qm resize 101 scsi0 +50G

On the Proxmox web interface highlight your new VM.
Hardware->Attach the created disk (use SCSI)
Hadware->Remove the unused disk
Options->BootOrder->Set the boot disk to the newly added one.
Options->Start at Boot->Check yes.

Press the Start button and your VM should come up.  You can watch it boot using the Console of your new VM.
Your new VM will have a DHCP assigned IP address which you must discover by examining your router.  You use that to access HA in the traditional way.

El-mou33, 2022-10-09:
Now we have a home assistant OS, as a VM, you only need to enter to the proxmox shell and paste this:
`bash -c "$(wget -qLO - https://github.com/tteck/Proxmox/raw/main/vm/pimox-haos-vm-v4.sh)"`

This script is posted [here](https://tteck.github.io/Proxmox/)

> @hastmu how did you load the image? it keeps to not work for me even with the latest 8.0 image...
> 
> can you even show the "Options" view of the HA's vm, please?

Now we have a home assistant OS, as a VM, you only need to enter to the proxmox shell and paste this:
`bash -c "$(wget -qLO - https://github.com/tteck/Proxmox/raw/main/vm/pimox-haos-vm-v4.sh)"`

This script is posted [here](https://tteck.github.io/Proxmox/)

pbix, 2022-10-09:
[image]

pbix, 2022-10-09:
[image]