pimox/pimox7 #64: Script Failure 'Syntax error: "(" unexpected (expecting "do")'

jonstump, 2022-03-28:
I'm getting a syntax error on my raspberry pi cm4 with 4gb of ram. I followed the instructions from the readme and I don't immediately see an additional ( on the line it's reporting. Full error below:

`./RPiOS64-IA-Install.sh: 57: Syntax error: "(" unexpected (expecting "do")`

This happens as soon as I attempt to enter a hostname. Is there something I may have done with setting up my pi that might cause some issues for this?

rossmaclean, 2022-03-28:
Hi,

Do you get the same error when running `sh ./RPiOS64-IA-Install.sh` and `bash ./RPiOS64-IA-Install.sh` (instead of just `./RPiOS64-IA-Install.sh`)?

I can replicate the issue with a simplified version of the script (see below) when using `sh`.
```
# !/bin/bash

#### GET USER INPUTS #### HOSTNAME ######################################################################################################
read -p "Enter new hostname e.g. RPi4-01-PVE : " HOSTNAME
while [[ ! "$HOSTNAME" =~ ^(([a-zA-Z0-9]|[a-zA-Z0-9][a-zA-Z0-9\-]*[a-zA-Z0-9])\.)*([A-Za-z0-9]|[A-Za-z0-9][A-Za-z0-9\-]*[A-Za-z0-9])$  ]]
 do
  printf " --->$RED $HOSTNAME $NORMAL<--- Is NOT an valid HOSTNAME, try again...\n"
  read -p "Enter new hostname e.g.: RPi4-01-PVE  : " HOSTNAME
done
```
AFAIK the script should be using `bash` as the shebang specifies `/bin/bash`, however something to note is that there is a space which I don't think should be there (a space between the `#` and `!` i.e. `# !/bin/bash`). Does removing this space change anything for you?

Cheers,
Ross

TuxfeatMac, 2022-03-28:
Funny, yes it should be using bash!
I added a space betwen the # and the /bin/bash , as Github code gudielines sugested it to be better code... feel free to change it back if it helps!

I will check the script again. I will take some time since i do not have time for private scripts at the moment... maybe next weekend. No promises...

jonstump, 2022-03-29:
I don't think this was my issue. Before testing this I went ahead and refreshed my cm4 to a fresh install of raspberry pi os. My guess is that I had something on my other install that I setup not realizing it would interfere with the script.

After reflashing I was able to install using the script and I'm even connected to the cluster. Thanks again for the hard work!

closed 2022-03-29 by jonstump