pimox/pimox7 #66: Installation fails on raspios 64bit

gabrielecaruso, 2022-04-02:
Hello,
i have tried like 20 times to install pimox but the error is always the same:

Done.
Loading new ceph-dkms-0.0.2 DKMS files...
It is likely that 5.15.30-v8+ belongs to a chroot's host
Building for 5.15.30-v8+
Building initial module for 5.15.30-v8+
Error! Bad return status for module build on kernel: 5.15.30-v8+ (aarch64)
Consult /var/lib/dkms/ceph-dkms/0.0.2/build/make.log for more information.
dpkg: error processing package ceph-dkms (--configure):
 installed ceph-dkms package post-installation script subprocess returned error exit status 10
Setting up pve-manager (7.1-10) ...
Job for pvestatd.service failed because the control process exited with error code.
See "systemctl status pvestatd.service" and "journalctl -xe" for details.
dpkg: error processing package pve-manager (--configure):
 installed pve-manager package post-installation script subprocess returned error exit status 1
Setting up zfs-dkms (2.0.3-9) ...
Removing old zfs-2.0.3 DKMS files...

------------------------------
Deleting module version: 2.0.3
completely from the DKMS tree.
------------------------------
Done.
Loading new zfs-2.0.3 DKMS files...
It is likely that 5.15.30-v8+ belongs to a chroot's host
Building for 5.15.30-v8+
Building initial module for 5.15.30-v8+

configure: error:
        *** None of the expected "capability" interfaces were detected.
        *** This may be because your kernel version is newer than what is
        *** supported, or you are using a patched custom kernel with
        *** incompatible modifications.
        ***
        *** ZFS Version: zfs-2.0.3-9
        *** Compatible Kernels: 3.10 - 5.10

Error! Bad return status for module build on kernel: 5.15.30-v8+ (aarch64)
Consult /var/lib/dkms/zfs/2.0.3/build/make.log for more information.
dpkg: error processing package zfs-dkms (--configure):
 installed zfs-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of proxmox-ve:
 proxmox-ve depends on pve-manager; however:
  Package pve-manager is not configured yet.
 proxmox-ve depends on ceph-dkms; however:
  Package ceph-dkms is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of zfs-zed:
 zfs-zed depends on zfs-modules | zfs-dkms; however:
  Package zfs-modules is not installed.
  Package zfs-dkms which provides zfs-modules is not configured yet.
  Package zfs-dkms is not configured yet.

dpkg: error processing package zfs-zed (--configure):
 dependency problems - leaving unconfigured
Processing triggers for initramfs-tools (0.140) ...
Errors were encountered while processing:
 ceph-dkms
 pve-manager
 zfs-dkms
 proxmox-ve
 zfs-zed
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@rasp1-prox:~#

I followeb every single step in this guide (and i tried even other guides) but the installation in never "successful".

My hardware:

Raspberri Pi 4 model B 8GB RAM
32GB Samsung SD Card

Tried even the os image described in the installation script but nothing changed, the error is still the same.
I really don't know what to do next or what to try :/

topa-LE, 2022-04-02:
Have exactly the same errors and can not complete the installation on the Pi4.

topa-LE, 2022-04-02:
None of the expected "capability" interfaces were detected.
*** This may be because your kernel version is newer than what is
*** supported, or you are using a patched custom kernel with
*** incompatible modifications.
***
*** ZFS Version: zfs-2.0.3-9
*** Compatible Kernels: 3.10 - 5.10

Error! Bad return status for module build on kernel: 5.15.30-v8+ (aarch64)

**Installed kernel is too new for installation.**

Everything else is a consequential error, because the dependencies are missing.
Here one should times nachbesser or use kernel 5.10!

gabrielecaruso, 2022-04-04:
> Have exactly the same errors and can not complete the installation on the Pi4.

Hi, after trying like 25 times, i finally found out the problem (maybe, i think).
You need to expand the root partition, after flashing the os ssh into the Pi, while you are in "pi" user, du "sudo raspi-config" > advanced > expand root partition, then reboot.
After reboot, as user "pi" di "sudo apt update -y && sudo apt upgrade -y", after all of this, you can run the install script.

There are many errors but proxmox works (somehow), but...is really difficult to boot an effective vm inside it, is note realy stable.
To upload the .iso file i needed to try 4 times, everytime failed.
Anyway, i recommend you to erad this ENTIRE post BEFORE you try https://raspberrytips.com/proxmox-on-raspberry-pi/

topa-LE, 2022-04-04:
This has less to do with the Expanded Filesytem under raspi-config.

The installation should already go through without errors, because after each apt update dmks ceph will try to build again. Remedy is simply to use a new kernel -> **5.10.95**

https://github.com/raspberrypi/linux/tree/rpi-5.10.y
https://www.raspberrypi.com/documentation/computers/linux_kernel.html

**Build:**
sudo apt install git bc bison flex libssl-dev make
git clone --depth=1 --branch rpi-5.10.y https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel8
make bcm2711_defconfig -j4

**Config:**
sudo apt install libncurses5-dev
make menuconfig -j4

KSM Support active

**Kernel 64Bit Build:**
make -j4 Image modules dtbs
sudo make modules_install
sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/
sudo cp arch/arm64/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm64/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm64/boot/Image /boot/$KERNEL.img

Reboot

Done ;-)

gabrielecaruso, 2022-04-08:
Well thanks!
I tought it was this just because the panel won't load up before.

jimsemara, 2022-04-08:
I also ran into problems with zfs-dkms when setting up Pimox7 using [RPiOS64-IA-Install.sh](https://raw.githubusercontent.com/pimox/pimox7/master/RPiOS64-IA-Install.sh) on a freshly installed 2021-10-30-raspios-bullseye-arm64-lite, RPI 4B 8GB.

So I decided to manually setup Pimox7 essentially doing the same thing the script above does but skipping the installation of the package raspberrypi-kernel-headers and I got through the whole process withouth errors.
Could also successfully install ceph and everything was looking real fine.

Wanted to give the install script another run, this time removing `&& apt install -y raspberrypi-kernel-headers` from line 148.
It worked perfectly!

Could not spend much time on real debugging but wanted to share my trial&error results here :)

gabrielecaruso, 2022-04-09:
Hello,
somebody has been able to start a vm in a rpi4 ? I followed every guide but the vm just won't start up, it throws some error like:

BdsDxe: failed to load Boot0001 "UEFI QEMU QEMU CD-ROM " from PciRoot(0x0)/Pci(0x5,0x0)/Scsi(0x0,0x2): Not Found
BdsDxe: failed to load Boot0003 "UEFI QEMU QEMU HARDDISK " from PciRoot(0x0)/Pci(0x5,0x0)/Scsi(0x0,0x0): Not Found

>>Start PXE over IPv4.

I'm trying to virtualize home assistant image right now

jebotz, 2022-04-09:
With regard to @topa-LE kernel build instructions...
> sudo cp arch/arm64/boot/dts/broadcom/.dtb /boot/
> sudo cp arch/arm64/boot/dts/overlays/.dtb* /boot/overlays/

This should be...
```
sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/
sudo cp arch/arm64/boot/dts/overlays/*.dtbo /boot/overlays/
```
And then you still need to do...
`make headers_install`

Also, before installing the proxmox-ve package, remove at least the raspberrypi-kernel-headers package (leaving only the headers from the just-compiled kernel) or dpkg will still try to build the modules for the packaged kernel as well.

hastmu, 2022-04-10:
> Hello, somebody has been able to start a vm in a rpi4 ? I followed every guide but the vm just won't start up, it throws some error like:
>
> BdsDxe: failed to load Boot0001 "UEFI QEMU QEMU CD-ROM " [...]

hi - ha works works with generic aarch64 starting with 8+, see the other issue.
haos_generic-aarch64-8.0.dev20220325.img works fine.
do not use the rp4 native images, you need a uefi arm64/aarch64 image/iso.

ToSMaverick, 2022-04-14:
Instead of compiling the kernel from scratch I was able to install Pimox7 with the last 5.10 kernel from the official archive without issues.
The downside of this is that you won't get any kernel updates anymore. Let's see when Proxmox and Pimox moves on to Kernel 5.15 and beyond 😄 

```bash
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20220120-1_arm64.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20220120-1_arm64.deb

sudo dpkg -i raspberrypi-kernel_1.20220120-1_arm64.deb raspberrypi-kernel-headers_1.20220120-1_arm64.deb

sudo apt-mark hold raspberrypi-kernel raspberrypi-kernel-headers

sudo -s
curl https://raw.githubusercontent.com/pimox/pimox7/master/RPiOS64-IA-Install.sh > RPiOS64-IA-Install.sh
chmod +x RPiOS64-IA-Install.sh
./RPiOS64-IA-Install.sh
```

alanivey, 2022-04-22:
Thanks, @ToSMaverick!

To make things a little easier, you can start with `2022-01-28-raspios-bullseye-arm64-lite` which includes the 20220120 kernel. You would still need to download and install the 20220120 version of the kernel-headers package.

Another small change that may not really make a difference: I noticed that the `raspberrypi-bootloader` package had the same datestamp version as kernel and kernel-headers, so I used `apt-mark hold` on that as well.

18iteration, 2022-04-25:
I had the same issue. 
It would be great to have an uninstaller script to revert all the changes. Are we supposed to reflash rpios?

luckydonald, 2022-05-08:
it says it can't find the kernel `Building for 5.15.32-v8+` as supported, only supporting `Compatible Kernels: 3.10 - 5.10`.

According to https://github.com/openzfs/zfs/issues/12786, `5.15` should be supported though?

luckydonald, 2022-05-08:
The great thing with [ToSMaverick's header install](#issuecomment-1099110466) is that it can fix your automated install script install after the fact.

18iteration, 2022-05-09:
> it says it can't find the kernel `Building for 5.15.32-v8+` as supported, only supporting `Compatible Kernels: 3.10 - 5.10`.
> 
> According to https://github.com/openzfs/zfs/issues/12786, `5.15` should be supported though?

I think I was able to fix it with a recommendation from another issue (like removing a line or something like that).

Even then, pixmox was not really runnable nor was it properly documented.

Switching to VMware ESXi was the best decision. Literally no errors, super easy and it actually works at the end...

rgsilva, 2022-05-14:
Another solution is to download the ceph-dkms repository and replace with the current kernel code. The procedure is the following:

1. First you need to clone RPi's Kernel source code.

  ```
  git clone https://github.com/raspberrypi/linux.git --depth=1
  ```

2. Then copy all ceph files into a new default-src directory.

  ```
  mkdir -p default-src/drivers/block
  cp -r linux/drivers/block/{rbd.c,rbd_types.h} sdefault-srcrc/drivers/block/
  grep -E "^#|^ccflags|rbd.o" linux/drivers/block/Makefile > default-src/drivers/block/Makefile
  mkdir -p default-src/fs/ceph
  cp linux/fs/ceph/* default-src/fs/ceph/
  mkdir -p default-src/net/ceph
  cp -r linux/net/ceph/* default-src/net/ceph/
  ```

3. Now you need to clone the original repository

  ```
  git clone https://github.com/pimox/ceph-dkms.git
  ```

4. Then you need to replace the default-src directory with the one you created in the step before

  ```
  rm -rf ceph-dkms/src/default-src
  cp -r default-src ceph-dkms/src/default-src
  ```

5. To make sure the deb won't get (easily) replaced, we can increase its minor version.
  This step is optional and can cause issues with future updates, but I recommend it.

  ```
  (
    cd ceph-dkms/
    sed -i "s/0\\.0\\.2/0.0.2-1/g" debian/changelog
    sed -i "s/0\\.0\\.2/0.0.2-1/g" debian/dkms
  )
  ```

6. Now you need to regenerate the .deb

  ```
  (cd ceph-dkms && make)
  ```

7. You can finally install it using dpkg.
  Note: if you didn't change the version before, you need to update the package version below.

  ```
  (
    cd ceph-dkms
    sudo dpkg -i ceph-dkms_0.0.2-1_all.deb
  )
  ```

For the zfs-dkms, you can just follow [OpenZFS's instructions for installing on Debian Bullseye](https://openzfs.github.io/openzfs-docs/Getting%20Started/Debian/index.html#installation).

As of today, at least, this seems to be fully working. Just tested on both nodes on my cluster and they all seem pretty happy with the new packages:

```
$ apt list --installed | grep -E "(ceph|zfs)-dkms"

ceph-dkms/now 0.0.2-1 all [installed,local]
zfs-dkms/bullseye-backports,bullseye-backports,now 2.1.4-1~bpo11+1 all [installed,automatic]
```

Have fun!

luckydonald, 2022-05-17:
Well, I just ordered another pi to get into the clustering fun, and will try out that new way,
while this time also documenting everything step for step.

I'll write it down at [gist.github.com/luckydonald/1849291fb5e19c87df8c8a1618e29eaa](https://gist.github.com/luckydonald/1849291fb5e19c87df8c8a1618e29eaa),
and I will keep it up to date there.

deB4SH, 2022-05-19:
Hi all,
I'm stumbling into an issue with the automated install. 
My rpi4 get stuck (blank screen, no other tty possible, network not available) on `DEBIAN_FRONTEND=noninteractive apt install -y -o Dpkg::Options::="--force-confdef" proxmox-ve`

I tried pinning the headers as mentioned in https://gist.github.com/luckydonald/1849291fb5e19c87df8c8a1618e29eaa#1-install-pimox-proxmox-on-the-raspberry-pi but it didn't help et all. 

Also tried: https://github.com/pimox/pimox7/issues/66#issuecomment-1126826946
But didn't work either - install wasn't successful :/

I'm using an external ssd via usb boot with this pi. 
Could it be possible that my issue is related to this?

rgsilva, 2022-05-30:
@deB4SH It shouldn't be an issue, as I'm using external SSDs to boot both my Pis as well. Are you sure you're using Raspbian OS 64bit (https://github.com/pimox/pimox7#install-from-scratch-rpios64bit-interactive-automatic-installer)?

You can also try to install manually as per the README - it isn't much more complicated.

deB4SH, 2022-05-31:
@rgsilva yes the latest image flashed by the rpi-imager. also latest bios

rgsilva, 2022-05-31:
@deB4SH that's unusual. Try installing over a local tty instead of SSH, so you can switch between consoles and check what is going on. It could be also power issues (`dmesg` should warn you about that, plus the power LED usually indicates issues as well), as I had some trouble with that in the past that caused exactly those issues: blank screen, no tty, no network.

luckydonald, 2022-06-04:
@ rgsilva [rgsilva](https://github.com/rgsilva) do I have to do [those steps](#issuecomment-1126826946) before or after the installer?

rgsilva, 2022-06-04:
@luckydonald after. You can check if it's broken by checking the result of the installation to see if the other packages failed to install.

Ooops, sorry about the typo! I've fixed it!

rmadrona, 2022-07-16:
To install pimox on top of 5.15 kernel, you need to install zfs 2.1
To achieve that, you need 
- first to install linux-headers matching your kernel
- then to configure apt to use bullseye-backports repo as documented here https://openzfs.github.io/openzfs-docs/Getting%20Started/Debian/index.html#installation
# apt install zfs-dkms/bullseye-backport
# apt install zfsutils-linux/bullseye-backport
# apt install zfs-zed/bullseye-backport

Result can be checked 
# apt list | grep zfs | grep installed
libzfs4linux/bullseye-backports,now 2.1.5-1~bpo11+1 arm64 [installed,automatic]
zfs-dkms/bullseye-backports,now 2.1.5-1~bpo11+1 all [installed]
zfs-zed/bullseye-backports,now 2.1.5-1~bpo11+1 arm64 [installed]
zfsutils-linux/bullseye-backports,now 2.1.5-1~bpo11+1 arm64 [installed]

Then there is another similar trick to solve with ceph-dkms dependecy.
this package needs to be compiled as root on your system https://github.com/pimox/ceph-dkms

GneeChee, 2022-07-19:
Pulling my hair out over this.

Confused as to when this should be done in relation to the actual PiMox install.

It seems I've tried every combination, and I'm failing miserably.  

RPi 4, 8G, Raspberry Pi OS 64 Lite

Thanks mucho!

alanivey, 2022-07-20:
@GneeChee - install the Raspberry Pi OS, follow the steps in https://github.com/pimox/pimox7/issues/66#issuecomment-1099110466, then run this repository's installer script.

luckydonald, 2022-07-25:
You can try to follow my guide for that, if you like: 
https://gist.github.com/luckydonald/1849291fb5e19c87df8c8a1618e29eaa#file-1-pimox-md ([permalink](https://gist.github.com/luckydonald/1849291fb5e19c87df8c8a1618e29eaa/511381e19753ef2d7b57fe9655e1fcfd0ff4f56d#file-1-pimox-md))

GneeChee, 2022-08-07:
Sorry for the delay in responding, and thanks for the link.

I tried option A ... no bueno, batman :(

I also should mention that I tried the new install script that fixes the very issue that I'm having.

I'm still having the same issue.

I guess I'm off to start digging into esxi.

Thanks to all who tried to help me :)

wwalker0307, 2022-08-18:
Reporting in to say I get the same errors.  However, Pimox continues to function for me, though maybe less than optimally.  I have a cluster of six.  They work 99% of the time, but I've had to rebuild one node multiple times because it stops communicating after a reboot.  May or may not be related to the error.

c4lliope, 2022-09-12:
Hello; I obeyed @ToSMaverick's procedure, and upon my pi's relaunch I see:

```bash
grace@chesapeake:~/src$ nmap -p- 10.0.0.128/26
Starting Nmap 7.80 ( https://nmap.org ) at 2022-09-12 00:37 EDT
Nmap scan report for 10.0.0.128
Host is up (0.019s latency).
Not shown: 65531 closed ports
PORT     STATE SERVICE
22/tcp   open  ssh
25/tcp   open  smtp
3128/tcp open  squid-http
8006/tcp open  wpl-analytics
```

My `/var/log/syslog` includes many error lines regarding `ksm`, like:

```
Sep 11 23:37:40 program postfix/smtpd[5154]: connect from unknown[unknown]
Sep 11 23:37:40 program postfix/smtpd[5154]: lost connection after CONNECT from unknown[unknown]
Sep 11 23:37:40 program postfix/smtpd[5154]: disconnect from unknown[unknown] commands=0/0
Sep 11 23:37:40 program ksmtuned[522]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Sep 11 23:38:41 program ksmtuned[522]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Sep 11 23:39:02 program pvestatd[1180]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)
Sep 11 23:39:02 program pvestatd[1180]: CPU flag detection failed, will try again after delay
Sep 11 23:39:41 program ksmtuned[522]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Sep 11 23:40:41 program ksmtuned[522]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
Sep 11 23:41:00 program postfix/anvil[5157]: statistics: max connection rate 1/60s for (smtp:unknown) at Sep 11 23:37:40
Sep 11 23:41:00 program postfix/anvil[5157]: statistics: max connection count 1 for (smtp:unknown) at Sep 11 23:37:40
Sep 11 23:41:00 program postfix/anvil[5157]: statistics: max cache size 1 at Sep 11 23:37:40
```

c4lliope, 2022-09-12:
During my `ceph-dkms` build as @rgsilva recommends, I see:

```bash
access@program:~/src $ (
  cd ceph-dkms
  sudo dpkg -i ceph-dkms_0.0.2-1_all.deb
)
(Reading database ... 125408 files and directories currently installed.)
Preparing to unpack ceph-dkms_0.0.2-1_all.deb ...

-------- Uninstall Beginning --------
Module:  ceph-dkms
Version: 0.0.2
Kernel:  5.10.92-v8+ (aarch64)
-------------------------------------

# ...

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.0.2
completely from the DKMS tree.
------------------------------
Done.
Unpacking ceph-dkms (0.0.2-1) over (0.0.2) ...
Setting up ceph-dkms (0.0.2-1) ...
Loading new ceph-dkms-0.0.2 DKMS files...
It is likely that 5.10.92-v8+ belongs to a chroot's host
Building for 5.10.92-v8+
Building initial module for 5.10.92-v8+
Error! Bad return status for module build on kernel: 5.10.92-v8+ (aarch64)
Consult /var/lib/dkms/ceph-dkms/0.0.2/build/make.log for more information.
dpkg: error processing package ceph-dkms (--install):
 installed ceph-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 ceph-dkms
```

Kornelius777, 2022-11-24:
@rgsilva: I'm also running into this error message:

```
depmod.....

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.0.2
completely from the DKMS tree.
------------------------------
Done.
Unpacking ceph-dkms (0.0.2-1) over (0.0.2) ...
Setting up ceph-dkms (0.0.2-1) ...
Loading new ceph-dkms-0.0.2 DKMS files...
It is likely that 5.15.76-v8+ belongs to a chroot's host
Building for 5.10.0-19-arm64 and 5.15.76-v8+
Building initial module for 5.10.0-19-arm64
Error! Bad return status for module build on kernel: 5.10.0-19-arm64 (aarch64)
Consult /var/lib/dkms/ceph-dkms/0.0.2/build/make.log for more information.
dpkg: error processing package ceph-dkms (--install):
 installed ceph-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 ceph-dkms
```

While /var/lib/dkms/ceph-dkms/0.0.2/build/make.log starts:
```
make: Entering directory '/usr/src/linux-headers-5.10.0-19-arm64'
  CC [M]  /var/lib/dkms/ceph-dkms/0.0.2/build/src/drivers/block/rbd.o
  CC [M]  /var/lib/dkms/ceph-dkms/0.0.2/build/src/fs/ceph/super.o
  CC [M]  /var/lib/dkms/ceph-dkms/0.0.2/build/src/net/ceph/ceph_common.o
  CC [M]  /var/lib/dkms/ceph-dkms/0.0.2/build/src/fs/ceph/inode.o
/var/lib/dkms/ceph-dkms/0.0.2/build/src/net/ceph/ceph_common.c: In function ‘ceph_alloc_options’:
/var/lib/dkms/ceph-dkms/0.0.2/build/src/net/ceph/ceph_common.c:354:5: error: ‘struct ceph_options’ has no member named ‘con_modes’
  354 |  opt->con_modes[0] = CEPH_CON_MODE_UNKNOWN;
      |     ^~
In file included from /var/lib/dkms/ceph-dkms/0.0.2/build/src/fs/ceph/super.c:22:
/var/lib/dkms/ceph-dkms/0.0.2/build/src/fs/ceph/cache.h:12:10: fatal error: linux/netfs.h: No such file or directory

```
How can I get this compiled?

Kind regards!

diginfo, 2022-12-04:
spent last couple of days on this, does anyone have a proven instal method where they have succeeded in the last few weeks using the latest 64Bit Lite Version installed with the official installer ?

Kornelius777, 2022-12-04:
It works when you choose the older version of the two ("pacific").
"Quincy" just doesn't work.

diginfo, 2022-12-04:
@Kornelius777 - thanks, but dont understand the versions you are referring to ?

Kornelius777, 2022-12-04:
Didn't you get here because of trouble installing ceph?

If you "just" want to install pimox - that works fine if you follow the README. No trouble, here.

Kornelius777, 2022-12-04:
Just one important thing:
Make sure that your /etc/hosts doesn't contain a line with "127.0.1.1"
The rest is a no brainer

diginfo, 2022-12-04:
spent last 2 days, trying different suggestions / versions, so far no success.

Do you mean "127.0.0.1" (localhost) or "127.0.1.1" ?

Kornelius777, 2022-12-04:
127.0.1.1

diginfo, 2022-12-04:
Thanks, managed to get it installed, but unable to startup any containers:

```
sync_wait: 34 An error occurred in another process (expected sequence number 7)
__lxc_start: 2074 Failed to spawn container "101"
TASK ERROR: startup for container '101' failed
```
Tried alpine & Debian CT Templates.

[image]

```
Dec 04 18:32:11 rpi4-pve systemd[1]: pve-container@101.service: Main process exited, code=exited, status=1/FAILURE
Dec 04 18:32:11 rpi4-pve systemd[1]: pve-container@101.service: Failed with result 'exit-code'.
Dec 04 18:32:11 rpi4-pve systemd[1]: pve-container@101.service: Consumed 1.929s CPU time.
Dec 04 18:32:34 rpi4-pve ksmtuned[478]: /usr/sbin/ksmtuned: line 61: /sys/kernel/mm/ksm/run: No such file or directory
```

Kornelius777, 2022-12-04:
You cannot run amd64 containers
You need arm64

Use one of these:
https://us.lxd.images.canonical.com/images/

diginfo, 2022-12-04:
@Kornelius777 - thanks, but what is the final URL to query, each time I go further down the path, it just keeps asking me to re-query:

[image]

What is the filename that I should be looking for ?

Kornelius777, 2022-12-04:
Try this one for bullseye
https://us.lxd.images.canonical.com/images/debian/bullseye/arm64/default/20221204_05:25/rootfs.tar.xz

El-Vim55, 2023-01-09:
@topa-LE 
I downgraded to kernel 5.10, but was having the same issue, even tried kernel 5.9 and still the same

TuLiSTua, 2023-04-27:
I followed the Tutorial of @rgsilva but i got stuck at this point:
`root@node1:~# (cd ceph-dkms && make)
rm -rf build
mkdir build
cp -a src build
cp -R debian build
cd build ; dpkg-buildpackage -b -uc -us
dpkg-buildpackage: info: source package ceph-dkms
dpkg-buildpackage: info: source version 0.0.2-1
dpkg-buildpackage: info: source distribution buster
dpkg-buildpackage: info: source changed by Tim <tim.j.wilkinson@gmail.com>
dpkg-buildpackage: info: host architecture arm64
 dpkg-source --before-build .
dpkg-checkbuilddeps: error: Unmet build dependencies: debhelper (>= 9)
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
make: *** [Makefile:4: deb] Error 3`

Any idea?

TheBossME, 2023-11-30:
1. load Install image for pi 4 from here 

https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2023-10-10/2023-10-10-raspios-bookworm-armhf-lite.img.xz

2. install the image to an usb stick or AD cars.

3. boot the Debian bookworm and do prepare for root login.

4. follow the install steps on that web page

https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md

curl https://mirrors.apqa.cn/proxmox/debian/pveport.gpg -o /etc/apt/trusted.gpg.d/pveport.gpg

nano /etc/apt/sources.list.d/pveport.list

Put in the file:

deb [arch=arm64] https://mirrors.apqa.cn/proxmox/debian/pve bookworm port

5. do apt update & apt dist-upgrade

6. Nano /etc/hosts
   put your actual ip address you've got from dhcp into the file like:
192.168.x.x pve.domain.local pve

7. apt install ifupdown2 open-iscsi ksmtuned postfix chrony proxmox-pve

ATTENTION!! NO REBOOT BEGORE VMBR0 is created on web page 

8. login to https://192.168.x.x:8006 with your created root credentials

9. add vmbr0 bridge with the proper informations.

That's it. 

Thanks to JiangCuo for preparing the mirror in Germany too 👍

sp33dyS, 2023-12-01:
> curl https://mirrors.apqa.cn/proxmox/debian/pveport.gpg -o /etc/apt/trusted.gpg.d/pveport.gpg
> 
> nano /etc/apt/sources.list.d/pveport.list
> 
> Put in the file:
> 
> deb [arch=arm64] https://mirrors.apqa.cn/proxmox/debian/pve bookworm port
> 
> 5. do apt update & apt dist-upgrade

Many thanks for this. I had problems installing Pimox on RaspberryPi OS today, because of missing packages. You solution helped me a lot!

vs4vijay, 2024-10-18:
It worked for me after following these steps.

> Instead of compiling the kernel from scratch I was able to install Pimox7 with the last 5.10 kernel from the official archive without issues. The downside of this is that you won't get any kernel updates anymore. Let's see when Proxmox and Pimox moves on to Kernel 5.15 and beyond 😄
> 
> ```shell
> wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20220120-1_arm64.deb
> wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20220120-1_arm64.deb
> 
> sudo dpkg -i raspberrypi-kernel_1.20220120-1_arm64.deb raspberrypi-kernel-headers_1.20220120-1_arm64.deb
> 
> sudo apt-mark hold raspberrypi-kernel raspberrypi-kernel-headers
> 
> sudo -s
> curl https://raw.githubusercontent.com/pimox/pimox7/master/RPiOS64-IA-Install.sh > RPiOS64-IA-Install.sh
> chmod +x RPiOS64-IA-Install.sh
> ./RPiOS64-IA-Install.sh
> ```