pimox/pimox7 #67: Take snapshot vm failed

mr12obot, 2022-04-09:
Hi all,

Anyone can help why I can not take a snapshot vm. The error messages below.

TASK ERROR: VM 100 qmp command 'query-savevm' failed - got timeout 

[image]

mr12obot, 2022-05-18:
@aanon4 @TuxfeatMac @Wings22Actual 

Let you know the current kernel that I used:

Host pimox: 5.15.32-v8+ aarch64 GNU/Linux
VM 100: 5.10.0-13-arm64 aarch64 GNU/Linux

I think it's a kernel version compatibility another issue #66 and #71 

Appreciate it when you up pimox version and moves on to kernel 5.15 beyond ?

hastmu, 2022-05-29:
same here, with Linux 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
and pve-manager/7.1-10/821f136a

mr12obot, 2022-07-19:
@aanon4  Could you help how to solve this issue to take snapshot vm failed?

Already update pimox to latest version below but still can not take snapshot.

Kernel Version Linux 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
PVE Manager Version pve-manager/7.2-7/f1b67e2b

aanon4, 2022-07-19:
What kind of disk and filesystem are you using? I did a very quick test of a VM I run with disk on Ceph and the snapshot was fine.

mr12obot, 2022-07-19:
@aanon4 Let you know this type disk and filesystem below. Any more check?

Filesystem     Type      Size  Used Avail Use% Mounted on
/dev/root      ext4       59G   55G  1.9G  97% /
devtmpfs       devtmpfs  3.7G     0  3.7G   0% /dev
tmpfs          tmpfs     3.9G   46M  3.8G   2% /dev/shm
tmpfs          tmpfs     1.6G  916K  1.6G   1% /run
tmpfs          tmpfs     5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1 vfat      253M   31M  222M  12% /boot
/dev/fuse      fuse      128M   20K  128M   1% /etc/pve
tmpfs          tmpfs     782M     0  782M   0% /run/user/0

aanon4, 2022-07-19:
Specifically, the VM you're trying to snapshot must have a virtual disk attached. What kind of disk is that virtual disk, and what underlying disk and filesystem is that on?

mr12obot, 2022-07-19:
@aanon4 Let you know this type disk and filesystem below. Any more check?

This one is pimox disk
[image]

This one is VM disk trying to snapshot.
[image]

mr12obot, 2022-07-20:
@aanon4 Let you know this type disk and filesystem below.:

This one is pimox disk
[image]

This one is VM disk trying to snapshot.
[image]

Last year trying to snapshot successfully without virtual disk attached.
[image]

Any more checked and suggested? Is snapshot must have a virtual disk attached for latest pimox update?