pimox/pimox7 #69: Nobody interested in Asahi Linux? (Apple M1 ARM64)

almeyras, 2022-04-29:
I think about the potential in Asahi Linux. I wonder if anyone involved in the Pimox project is taking any steps to port Pimox to Asahi Linux.

I tried reddit and not many people look interested: <https://www.reddit.com/r/AsahiLinux/comments/txl8be/if_i_had_a_mac_mini_debian_pimox/>

aanon4, 2022-04-29:
The fundamental problem with porting Proxmox/Pimox to Asahi Linux is that it's not Debian based, and Proxmox very much is. I'm hoping the Asahi Linux work migrates into Debian at some point.

almeyras, 2022-04-29:
But I see there is an option to install Debian (or some kind of flavour) on top of Asahi Linux: <https://git.zerfleddert.de/cgi-bin/gitweb.cgi/m1-debian/>

almeyras, 2022-05-12:
> @almeyras in this case it's must work without extra porting. I'm using pimox7 on top of regular Debian arm64 on HoneyComb LX2

Wow that's some good news for me, thank you. I'll give this a try in summer, as soon as I can get an Apple Silicon Mac.