pimox/pimox7 #71: Install PiMOX on Raspberry PI

nexusguy59, 2022-05-06:
I ran the script to install this on RPiOS Bullseye and I got an error at the end:

dpkg: error processing package pve-manager (--configure):
 installed pve-manager package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for dbus (1.12.20-2) ...
Processing triggers for initramfs-tools (0.140) ...
Processing triggers for libglib2.0-0:arm64 (2.66.8-1) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...
Processing triggers for rsyslog (8.2102.0-2) ...
Setting up glib-networking:arm64 (2.66.0-2) ...
Setting up libsoup2.4-1:arm64 (2.72.0-2) ...
Setting up gstreamer1.0-plugins-good:arm64 (1.18.4-2+~rpt1) ...
Processing triggers for libgdk-pixbuf-2.0-0:arm64 (2.42.2+dfsg-1) ...
Processing triggers for pve-ha-manager (3.3-3) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...
Errors were encountered while processing:
 zfs-dkms
 ceph-dkms
 zfs-zed
 proxmox-ve
 pve-manager
E: Sub-process /usr/bin/dpkg returned an error code (1)

=========================================================================================
 ! FIXING NETWORK CONFIGURATION.... ERRORS ARE NOMALAY FINE AND RESOLVED AFTER REBOOT !
=========================================================================================

=========================================================================================
                        ! INSTALATION COMPLETED ! WAIT ! REBOOT !
=========================================================================================
    after reboot the PVE web interface will be reachable here :
      --->   https://xxx.xxx.xxx.xxx:8006/  <---

         run --->  apt upgrade -y  <---
           in a root shell to complete the installation.
So I am now running apt upgrade -y 

[image]

dpkg errors all through this, what am I doing wrong?

[image]

Not sure why I am getting these can some help please?
root@RPi-01-PVE:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Thanks guys

luckydonald, 2022-05-08:
getting the same errors. Therefore, here's a text version of that png:

```
root@pimox:/home/luckydonald#  apt upgrade -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
4 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up ceph-dkms (0.0.2) ...
Removing old ceph-dkms-0.0.2 DKMS files...

------------------------------
Deleting module version: 0.0.2
completely from the DKMS tree.
------------------------------
Done.
Loading new ceph-dkms-0.0.2 DKMS files...
It is likely that 5.15.32-v8+ belongs to a chroot's host
Building for 5.15.32-v8+
Building initial module for 5.15.32-v8+
Error! Bad return status for module build on kernel: 5.15.32-v8+ (aarch64)
Consult /var/lib/dkms/ceph-dkms/0.0.2/build/make.log for more information.
dpkg: error processing package ceph-dkms (--configure):
 installed ceph-dkms package post-installation script subprocess returned error exit status 10
Setting up zfs-dkms (2.0.3-9) ...
Removing old zfs-2.0.3 DKMS files...

------------------------------
Deleting module version: 2.0.3
completely from the DKMS tree.
------------------------------
Done.
Loading new zfs-2.0.3 DKMS files...
It is likely that 5.15.32-v8+ belongs to a chroot's host
Building for 5.15.32-v8+
Building initial module for 5.15.32-v8+
configure: error:
	*** None of the expected "capability" interfaces were detected.
	*** This may be because your kernel version is newer than what is
	*** supported, or you are using a patched custom kernel with
	*** incompatible modifications.
	***
	*** ZFS Version: zfs-2.0.3-9
	*** Compatible Kernels: 3.10 - 5.10

Error! Bad return status for module build on kernel: 5.15.32-v8+ (aarch64)
Consult /var/lib/dkms/zfs/2.0.3/build/make.log for more information.
dpkg: error processing package zfs-dkms (--configure):
 installed zfs-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of proxmox-ve:
 proxmox-ve depends on ceph-dkms; however:
  Package ceph-dkms is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of zfs-zed:
 zfs-zed depends on zfs-modules | zfs-dkms; however:
  Package zfs-modules is not installed.
  Package zfs-dkms which provides zfs-modules is not configured yet.
  Package zfs-dkms is not configured yet.

dpkg: error processing package zfs-zed (--configure):
 dependency problems - leaving unconfigured
Processing triggers for initramfs-tools (0.140) ...
Errors were encountered while processing:
 ceph-dkms
 zfs-dkms
 proxmox-ve
 zfs-zed
 ```

luckydonald, 2022-05-08:
it says it can't find the kernel `Building for 5.15.32-v8+` as supported, only supporting `Compatible Kernels: 3.10 - 5.10`.

According to https://github.com/openzfs/zfs/issues/12786, `5.15` should be supported though.

luckydonald, 2022-05-08:
Seems to be a duplicate of #66.

rgsilva, 2022-05-14:
It's a kernel version compatibility issue, and it is indeed a duplicate of #66. I've found a way that avoids downgrading the kernel, but you have to build `ceph-dkms` yourself. You can get `zfs-dkms` from a bullseye-backports. See https://github.com/pimox/pimox7/issues/66#issuecomment-1126826946 for instructions.