pimox/pimox7 #74: Network lost after pimox7 install

thezoomlesss, 2022-05-15:
I've tried multiple installs on a RPi4 and every single time, upon reboot, no ip address is assigned to the pi anymore.

I wonder if it's something I am doing wrong, as I don't have too much knowledge when it comes to `networking`.
I ran the install script and gave it `192.168.0.118/24` and `192.168.0.1` as the `local IP + Gateway`.

The install runs fine and I get prompted with the screen telling me to restart and upgrade. 
Upon restart, no ip is assigned and connection to the internet is lost, obviously. 
This is on LAN, if it makes any difference.

Is there anything that I am obviously doing wrong?

kaliavarun, 2022-05-22:
I had a similar issue. Try giving the ip as 192.168.1.100/24. Make sure you are connected via ethernet. It doesn't work with wireless network.

maietta, 2022-05-22:
I'm having the same problem. I've tried using 192.168.1.100/24 with my router specified to start addresses from 192.168.1.1 but to no avail.

After install, there's no web UI, even after reboot. It doesn't actually appear the a service is installed.

kaliavarun, 2022-05-22:
Do you see any new device in your router's connected device page?

maietta, 2022-05-22:
@kaliavarun Yes I do.

puccaso, 2022-07-31:
Hi, 
i would try something like
`sudo ufw allow from [ip of machine you are using to connect]`
I had a similar issue.

maietta, 2022-07-31:
That wasn't my problem, @puccaso but that's good for others to check.

I found that by switching my OS allowed me to install without issues. No idea why Ubuntu 20.04 and 20.10 didn't work.

maietta, 2023-01-29:
Debian 11