pimox/pimox7 #76: Can't create VM with a .img image.

kaliavarun, 2022-05-22:
Hello.
I have successfully installed pimox on my raspberry pi 4 on rpi OS 64 bit. It works fine and I was able to create a new VM with the [debian](https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/debian-11.3.0-arm64-netinst.iso) iso image. However. I can't seem to make it work with rasp pi os or diet pi or any .img files. I followed the exact same procedure. It doesn't not detect the CD rom containing the img file and If i create a local drive by importing the img file using the command 
`qm importdisk 100 2022-04-04-raspios-bullseye-arm64-lite.img local`
it creates the local drive and still can't find it. I get the following error.
[image]

Following is my VM configuration:

[image]
[image]

Any idea , what could be the issue here?

parroyo-arroyo, 2022-05-28:
It happens the same to me

kaliavarun, 2022-05-28:
It turns out that `.img` images are not supported at the moment. The only option is `.iso` images.

jiangcuo, 2022-08-01:
qemu requires the guest to use efi to boot