pimox/pimox7 #77: Any way to remove the static IP requirement?

Snuupy, 2022-05-24:
Sometimes if I'm traveling with my rpi I don't always have a network set up with me.

- Is the requirement for a static IP still there after the initial install?
- Is there any workaround for the static IP requirement?
- What needs to be changed to have this working, if possible at all?

hastmu, 2022-05-29:
Hi, i did a once a proxmox install on a notebook, here you have the same issue with eth0 + wlan0 running with dhcp and vmbr0 with the static ip. The only issue i saw/see is that you never know what ip config you get from outside, so you may have a conflict.
e.g. eth0+wlan0 on dhcp, vmbr0 with 192.168.101.1/24 fine as long as you do not get a 192.168.101.0/24 from your dhcpd.

Snuupy, 2022-06-25:
Thanks for the write-up, I just haven't had a chance yet to play with this but thank you for taking the time. Will report back if/when I get to play with pimox again.

kolonuk, 2022-08-05:
I would second the point about knowing what IP you are connecting to.  Also, be aware that if using some hotel/maccy d's wifis, you may not be able to talk to your PI from your laptop/tablet - ie the APs may practice client isolation.

There are projects that convert a Pi into a personal hotel hotspot, can't remember any off hand right now.