pimox/pimox7 #78: arm64 openwrt not running

Babyvox550, 2022-05-29:
CPU full working.
[image]
VNC notworking(or VM not running)
[image]
VM settings。
[image]

cvocvo, 2022-06-18:
I have similar issues when trying to use the arm64 container build of OpenWRT, but mine fails when trying to create the container.
(from here: https://us.lxd.images.canonical.com/images/openwrt/21.02/arm64/default/) 

```
Formatting '/var/lib/vz/images/100/vm-100-disk-0.raw', fmt=raw size=8589934592 preallocation=off
Creating filesystem with 2097152 4k blocks and 524288 inodes
Filesystem UUID: 51c3ccb7-edaa-4f6c-add6-c0a280a791ca
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632
extracting archive '/var/lib/vz/template/cache/OpenWRT-21.02-arm64.tar.xz'
Total bytes read: 12103680 (12MiB, 21MiB/s)
Detected container architecture: arm64
unknown ID 'openwrt' in /etc/os-release file, trying fallback detection
TASK ERROR: unable to create CT 100 - unable to detect OS distribution
```

pbix, 2022-06-22:
> **UPDATE** I found this repo and some useful screenshots here -- and I got it running: [varoudis/openwrt_arm64_uefi#1 (comment)](https://github.com/varoudis/openwrt_arm64_uefi/issues/1#issuecomment-1159571443)
> 
The referenced github contains binaries that do work which is great but they are not the current release of OpenWRT.  How to get the most recent v22.03 version of OpenWRT to work?

I saw this page [https://openwrt.org/docs/guide-user/virtualization/qemu](https://github.com/varoudis/openwrt_arm64_uefi/issues/url) were the below command it listed.
"qemu-system-arm -nographic -M virt -m 64 -kernel openwrt-armvirt-zImage-initramfs"

I tried this from my PiMox shell using the armvirt-64 target and it amazingly works. I used the below armvirt-64 binary.
[https://downloads.openwrt.org/releases/22.03.0-rc4/targets/armvirt/64/openwrt-22.03.0-rc4-armvirt-64-Image-initramfs](https://github.com/varoudis/openwrt_arm64_uefi/issues/url)

Since this command works is it really a long way to getting a VM to boot properly under PiMox?  How is this qemu command different than what PiMox does in a VM?

jiangcuo, 2022-08-01:
Hi, guest need acpi and efi support.
see:
https://nekodaemon.com/2022/02/26/Building-OpenWrt-from-Scratch-for-ARM64-UEFI-ACPI-VM/

or you can try my img.
https://foxi.buduanwang.vip/pan/foxi/Virtualization/armvirt/openwrt/openwrt-argon/openwrt-21.02.3-armvirt-64-zh-argon.qcow2.tar.gz