pimox/pimox7 #82: Impossible to install raspbian on pimox

El-mou33, 2022-06-11:
I tried to install raspbian on pimox but never worked. I dont know if there is any way to get raspbian as a VM. The system only shows the bios. But not raspbian. I have debian working more or less (without sudo). But I couldn't be abble to use raspbian 32bit as virtual machine.

Any idea?

D4rk4, 2022-07-17:
Which hardware do you use?

El-mou33, 2022-07-17:
Raspberry pi 4B+ 8Gb ram

D4rk4, 2022-07-22:
@El-mou33 did you check [this](https://github.com/pimox/pimox7/issues/41)?
For me this config is working fine:
```
agent: 1
bios: ovmf
boot: order=scsi2;scsi0;net0
cores: 2
cpu: host
efidisk0: ZFS:100/base-100-disk-0.qcow2,efitype=4m,pre-enrolled-keys=1,size=64M
memory: 2048
meta: creation-qemu=6.2.0,ctime=1646778194
name: Reference-VM
net0: virtio=4E:22:83:43:B1:C7,bridge=vmbr0
numa: 0
ostype: l26
scsi0: ZFS-RAW:base-100-disk-2,discard=on,size=32G
scsi2: ZFS:iso/debian-11.2.0-arm64-netinst.iso,media=cdrom,size=327980K
scsihw: virtio-scsi-pci
smbios1: uuid=b6595e40-f4f8-4a6d-980b-8451ccafc11a
sockets: 1
template: 1
```

El-mou33, 2022-07-22:
Becouse it is more lightweight and it is specifically for raspberrys. I had another problem with debian, I can't use sudo in the terminal it say, user is not in the suddoers file. I opened the bios to add it, but I couldn't.

D4rk4, 2022-07-22:
@El-mou33 you completely do not understand what you do =(

hastmu, 2022-07-24:
hi - so rasbian will not boot as it is not native arch64.
as mentioned use debian with the right architecture.
regarding the sudo stuff, during the debian installation you specify a root password and you create a local user
to allow sudo for the local user just do the following as root:
adduser <username> sudo
this adds the <username> to the unix group sudo which is the admin group in debian.
sorry completely off topic :)

D4rk4, 2022-07-29:
@El-mou33 So, as it's not pimox related issue. May you close this issue?

El-mou33, 2022-07-29:
Of course, thanks for the answers.

closed 2022-07-29 by El-mou33