pimox/pimox7 #87: ZFS does not work

JWheelMaker, 2022-08-03:
Hey there!
For any reason ZFS does not work in my Pimox7 Instance.

When I choose ZFS in the node menu I get the following error:
`command '/sbin/zpool list -HpPLo name,size,alloc,free,frag,dedup,health' failed: exit code 1 (500)`

I tried executing `zfs list` in the Shell but then I receive this:
`Try running '/sbin/modprobe zfs' as root to load them.`

Running the recommended command brings me to this message:
`modprobe: FATAL: Module zfs not found in directory /lib/modules/5.15.56-v8+`

I am at a loss what to do now and not sure if this is a general error or just in my installation.

Best regards 
Jonas

hastmu, 2022-08-04:
Hi Jonas,

so i assume you have kernel 5.15 something?

[image]

so if you have zfs-2.0.3 this dkms will not compile, so my workaround was to download 2.0.7 and compile the kernel module and if was good to go.

JWheelMaker, 2022-08-05:
> hastmu

Thanks for the idea! I tried your solution the whole day but I didn't manage to build the deb package. I downloaded the 2.0.7 sources and followed the build docs but it didn't work. Could you explain how you built it?

JWheelMaker, 2022-09-04:
@hastmu thank you very much for your help and effort :)
In the last weeks I honestly have lost sigth of this project but now I'm back in and I was actually successfull!
After a few tries your solution worked fine but I found another way to run the latest zfs version.
After the usual installation of Pimox you need to perform the following steps:

```
apt update
apt install dpkg-dev raspberrypi-kernel-headers linux-image-generic
apt install zfs-dkms zfsutils-linux
```

Then after a reboot everything should work as it should :)
(In advance it is important to update your Pi to the latest EEPROM-Version by running rpi-update)

closed 2022-09-04 by JWheelMaker

thiscantbeserious, 2022-12-19:
I can confirm it works - but its not stable.

For example doing a simple docker image pull from within an LXC container on ZFS Storage will actually crash the Raspberry Pi after slowing it down to a pulp (RPi4B - 8GB). It can be a 100% reproduced with this script here (at least with mine):

https://github.com/tteck/Proxmox/blob/main/setup/homeassistant-install.sh

As such I will switch to BTRFS, which is much more lightweight and less demanding on all my other machines too. I really don't understand how Raspian doesn't ship it by default tought, will require me reinstalling everything on a proper debian or ubuntu image ... I had troubles with ZOL on Ubuntu on an 16 Core Intel too whenever I used Docker, so bye bye ZOL for me.

i-am-gizm0, 2023-08-01:
I was having a similar problem on regular Raspbian... installing the kernel headers helped: https://gizm0.dev/troubleshooting-pi-zfs-modprobe/