pimox/pimox7 #91: Failed Start (Raspberry Pi 4)

InikoMatthewPro, 2022-08-17:
Hi, can you help me please?

How to fix this?
```
kvm: -no-hpet: Option not supported for this target
TASK ERROR: start failed: QEMU exited with code 1
```

jiangcuo, 2022-10-17:
CPU:host
bios: ovmf
machine: i440fx
ostype: l26
diskandcdrom: scsi
net: virtio

dr03m, 2023-01-29:
I also have the same error. Using a Rock Pi 4b. Trying to start a VM with Windows 11 ARM.

InikoMatthewPro, 2023-01-29:
> I also have the same error. Using a Rock Pi 4b. Trying to start a VM with Windows 11 ARM.

if you need install Windows 11 ARM on Raspberry Pi 4. You can use [WoR](https://worproject.com/)

dr03m, 2023-02-06:
> > I also have the same error. Using a Rock Pi 4b. Trying to start a VM with Windows 11 ARM.
> 
> if you need install Windows 11 ARM on Raspberry Pi 4. You can use [WoR](https://worproject.com/)

This does nothing for me.

InikoMatthewPro, 2023-02-16:
> > > I also have the same error. Using a Rock Pi 4b. Trying to start a VM with Windows 11 ARM.
> > 
> > 
> > if you need install Windows 11 ARM on Raspberry Pi 4. You can use [WoR](https://worproject.com/)
> 
> This does nothing for me.

why not? just youtube tutorial.

dr03m, 2023-02-16:
> why not? just youtube tutorial.

Are you trolling? I have a Rock Pi4b with Pimox. I want WoA in a VM. I have no idea what you’re trying to say to me.

InikoMatthewPro, 2023-03-23:
Oh sorry