pimox/pimox7 #95: apparmor error when starting a LXC container

Bastler96, 2022-08-27:
Hi!

I downloaded Debian Bullseye arm64 from here and installed it on a Raspberry PI 4GB.
https://raspi.debian.net/tested-images/

Then I installed Pimox following the instructions for Manual installation. However, when I want to start an LXC container, the following error message appears in the web interface:

```
run_apparmor_parser: 915 Failed to run apparmor_parser on "/var/lib/lxc/100/apparmor/lxc-100_<-var-lib-lxc>": apparmor_parser: >>lxc-100_</var/lib/lxc><< kann nicht ersetzt werden.  Das Profil entspricht nicht dem Protokoll
apparmor_prepare: 1085 Failed to load generated AppArmor profile
lxc_init: 879 Failed to initialize LSM
__lxc_start: 2008 Failed to initialize container "100"
TASK ERROR: startup for container '100' failed
```

Tried it again on the command line:

```
root@proxmox:/# lxc-start -n 100 -F
lxc-start: 100: ../src/lxc/lsm/apparmor.c: run_apparmor_parser: 915 Failed to run apparmor_parser on "/var/lib/lxc/100/apparmor/lxc-100_<-var-lib-lxc>": apparmor_parser: >>lxc-100_</var/lib/lxc><< kann nicht ersetzt werden.
lxc-start: 100: ../src/lxc/lsm/apparmor.c: apparmor_prepare: 1085 Failed to load generated AppArmor profile
lxc-start: 100: ../src/lxc/start.c: lxc_init: 879 Failed to initialize LSM
lxc-start: 100: ../src/lxc/start.c: __lxc_start: 2008 Failed to initialize container "100"
lxc-start: 100: ../src/lxc/tools/lxc_start.c: main: 306 The container failed to start
lxc-start: 100: ../src/lxc/tools/lxc_start.c: main: 311 Additional information can be obtained by setting the --logfile and --logpriority options
root@proxmox:/# apparmor_status
apparmor module is loaded.
3 profiles are loaded.
3 profiles are in enforce mode.
   lsb_release
   nvidia_modprobe
   nvidia_modprobe//kmod
0 profiles are in complain mode.
0 processes have profiles defined.
0 processes are in enforce mode.
0 processes are in complain mode.
0 processes are unconfined but have a profile defined.
root@proxmox:/# cd /var/lib/lxc
root@proxmox:/var/lib/lxc# ls
100  101  102  103
root@proxmox:/var/lib/lxc# cd 100
root@proxmox:/var/lib/lxc/100# ls
config  rootfs  rules.seccomp
root@proxmox:/var/lib/lxc/100# cat config
lxc.cgroup.relative = 0
lxc.cgroup.dir.monitor = lxc.monitor/100
lxc.cgroup.dir.container = lxc/100
lxc.cgroup.dir.container.inner = ns
lxc.arch = arm64
lxc.include = /usr/share/lxc/config/ubuntu.common.conf
lxc.include = /usr/share/lxc/config/ubuntu.userns.conf
lxc.seccomp.profile = /var/lib/lxc/100/rules.seccomp
lxc.apparmor.profile = generated
lxc.apparmor.allow_nesting = 1
lxc.mount.auto = sys:mixed
lxc.monitor.unshare = 1
lxc.idmap = u 0 100000 65536
lxc.idmap = g 0 100000 65536
lxc.tty.max = 2
lxc.environment = TERM=linux
lxc.uts.name = test
lxc.cgroup2.memory.max = 536870912
lxc.cgroup2.memory.swap.max = 536870912
lxc.rootfs.path = /var/lib/lxc/100/rootfs
lxc.net.0.type = veth
lxc.net.0.veth.pair = veth100i0
lxc.net.0.hwaddr = 3A:23:0E:85:57:77
lxc.net.0.name = eth0
lxc.net.0.script.up = /usr/share/lxc/lxcnetaddbr
lxc.cgroup2.cpuset.cpus = 2
root@proxmox:/var/lib/lxc/100#
```

To me it looks like apparmor is installed. What is apparmor used for? Could someone help me?

Thanks for your support!

cjangrist, 2022-09-04:
can you share details of what app you're running in the lxc container. I've gotten this error when trying to run some apps in non privileged lxc containers (on x84 cloud envs) so I don't think it's related to this repo

can you try on a x86 proxmox and post results?

jiangcuo, 2022-10-17:
Try reinstalled apparmor
https://gitlab.com/apparmor/apparmor

Bastler96, 2022-10-23:
Hi!
I completely reinstalled debian arm64 and followed the instructions for manual installation.
I downloaded the images here and tested Ubuntu, Debian and Fedora. Run into the same problem. I tested the deault and cloud versions.
https://uk.lxd.images.canonical.com/images
Thanks for your help

MartinLoeper, 2022-11-27:
Reinstalling apparmor fixed the issue for me! What I did:

- Clone: https://gitlab.com/apparmor/apparmor
- Checkout branch: apparmor-3.1
- Install build deps: apt install bison flex swig autoconf automake libtool gettext 
- Build libapparmor as described in README
- Build Parser as described in README
- Install Parser and check via `apparmor_parser --version` that new version is indeed `3.1.0`