pimox/pimox7 #100: Totally lost both wlan and lan connection after pimox install

mc9625, 2022-09-15:
I’ve tried to install pimox on my raspberry pi 4 (buster 64bit )

Just after the installation, the raspberry booted without lan nor wlan working. Basically I am in a closed box, can’t manage to make the lan working again.

How can I troubleshoot this issue?

mc9625, 2022-09-16:
info: I have simply started the .sh script, chosen the same io address already used by Rpi (192.168.1.53/24) and follow the prompt correctly. After the boot the Raspberry completely lost wlan (probably due to the purge of dhcpcd5 during the pimox setup) and I haven’t found a way to restore it,  and the eth0 is missing from the ifconfig command. If I do the hostname -i command I can see that the io address is correct but it can’t reach the router nor any other ip address on the network. The router can’t see the raspberry on the LAN (the lights on the network port on the back of the raspberry are on). 
Since the raspberry now is completely offline is not very easy for me to post every relevant files info here. So if any good soul wants to help me and guide me to find a solution (the only one seems to reinstall everything from scratch but I would like to avoid it since the raspberry was already customized)…

Thank you

vm75, 2022-11-23:
I ran into the same issue. Pity.

automationxpert, 2023-03-19:
Same here too. tried two sets of installations.. First tried to install with 16GB sdcard for testing and it succeeded and everything is fine. But then i proceeded with SSD but ended up into no lan/wlan working issue. tried multiple times again and again but with no luck :(
couldnt understand how its still working with sdcard but not with ssd (since i followed same steps).