pimox/pimox7 #101: Question: Windows support

brainchild0, 2022-09-25:
What is the status of  support for running Windows  as an ARM64 guest?

Based on other discussions, it would appear that achieving a stable configuration with minimal hassle is not currently viable. What are the known obstacles and limitations, and what may be expected to change in the foreseeable future?

MarcMocker, 2022-10-05:
You can definetely use windows, but it has to be compiled for the arm processor. There is a special windows distribution for that you should use in that case. You'll find more information about this [here](https://learn.microsoft.com/en-us/windows/iot-core/tutorials/rpi)

MarcMocker, 2022-10-06:
I'm pretty much sure that if you virtualize the entire pi i.e. select the host processor, give it 2gb of ram and so on you have good chances of getting it up and running, but I'll try to build a windows vm myself later on and tell you if I faced any issues

brainchild0, 2022-10-06:
If you are running Windows 11, you would need to modify the registry settings, either manually or through modified installation media, to disable the enforcement of a mandatory minimum of 4Gb of memory.

s0mm3rb, 2023-08-11:
where any of you successfull?