pimox/pimox7 #105: is this project dead?

MartinVerges, 2022-11-22:
Hello,

does anyone know if this project is dead? The software state is quite outdated, does therefore not support PBS Namespaces for example and it looks like there is nothing going on.

Any way to help with it or to build it automatically with a newer version?

Thanks for the work so far, I hope it will continue!

Regards
Martin

aanon4, 2022-11-22:
Mostly sleeping.
I'll do a 7.3 update, hopefully before the end of the year - just need to find the time.

MartinVerges, 2022-11-23:
Is there anything that would help you? Like having a fast aarch64 build machine or so?
Please let me know if it's something besides time

HLeithner, 2022-12-19:
@aanon4 do you have a script for updating or do you update very repo by hand? do need help doing this?

I just installed pimox on a raspberry 4 and it works really well thanks

aanon4, 2022-12-19:
Unfortunately it's by hand as they tend to create new packages over time making this hard to automate. Still hoping to have time before the year is out.

HLeithner, 2022-12-20:
ok but doing this manually for 60 repos doesn't sounds sustainable for me^^

Can you layout what steps you are doing and I would like to write a script for you, which at least could update the current available repos?

MartinVerges, 2022-12-20:
I would also love to help and build an automated pipeline!

jiangcuo, 2022-12-20:
Hi i have a git  for proxmox-arm , and up to 7.3, both pve and pbs.

HLeithner, 2022-12-20:
> Hi i have a git for proxmox-arm , and up to 7.3, both pve and pbs.

you mean https://github.com/jiangcuo/Proxmox-Arm64 ?

thefiredragon, 2022-12-21:
@jiangcuo thanks for sharing your repo here, also
It looks like you repack the packages for arm64 capability instead to build it native from source ? 

Your readme also contain  info to add your repo but how to install proxmox is missing in your documentation.

A script to build native binarys in arm64 would be nice. This should be possible with a docker build script to keep this project alive.

chatziko, 2023-02-18:
First, let me say thanks for this amazing project, it has turned my little pis into a datacenter. :heart:

I was wondering whether there has been any progress in automating pimox updates. It works so well that I am really tempted to move everything I run on pis on top of pimox. But at the same time I am a bit scared that, say, a year from now something breaks and no updates are possible.

nima-1102, 2023-05-21:
Is this repo an up-to-date alternative for Pimox7?
https://github.com/jiangcuo/Proxmox-Arm64

Does anyone have experience with whether an upgrade works without problems?

TheBossME, 2023-05-21:
Of course, i've done ist with success.

anipad, 2023-05-22:
I tried the same via adding the sources from https://github.com/jiangcuo/Proxmox-Arm64 and `apt upgrade` from a working Pimox 7.2 installation. After reboot I had a booting Proxmox-Arm 7.4, but starting VMs (both, migrated and newly created) had no connection to the host network bridge despite a correct configuration.
So in the end I reverted back to Pimox 7.2.

TheBossME, 2023-05-22:
Do a Fresh Installation with Debian Image.
It will Work properly. Use the Debian Kernel 5.10-20/23.

Take the stable Debian (not raspberry) Image.

gruppferi, 2025-05-25:
Hello,
First of all thanks for the nice job done here, my Pimox runs since some year fine thanks to effort here.
now I build a cluster of Raspi 5 and would like to see if there is a hope for Pimox8 here or this repo will be not maintained anymore?
I don't want to goto to the pxvirt due ot Chinese Gov.
anyone else find something else ?