pimox/pimox7 #107: Pimox 7.2-7 can't start VM

jimoyong, 2022-11-23:
Hi guys,
I installed pimox 7.2-7 successfully.
But can't start and install VMs, the VMs stop at 
[image]

Hardwares:
[image]

Options:
[image]

ISO:
[image]

Does anyone get some ideas about my situations?
What should I do with it?
Thanks a lot!

n8lauscher, 2022-11-25:
same configuration - same Problem - can´t boot any img,iso etc...

diginfo, 2022-12-02:
me too, new install:

```
sync_wait: 34 An error occurred in another process (expected sequence number 7)
__lxc_start: 2074 Failed to spawn container "100"
```

bp0, 2022-12-04:
> Hi guys, I installed pimox 7.2-7 successfully. But can't start and install VMs, the VMs stop at [image]

Config looks right. If you press ESC at that screen, do you get the UEFI menu?

diginfo, 2022-12-04:
Just tried again, even use the image that you tested with:
```
#########################################################################################################################################
# Tested with image from:														                                                                                                 #
# https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-11-08/2021-10-30-raspios-bullseye-arm64-lite.zip	 #
#########################################################################################################################################
```
pve-manager still fails to install & configure

bp0, 2022-12-04:
@diginfo You're not having the same problem as @jimoyong . My response was to the OP. 

@diginfo, the output you've provided isn't useful. We need to see the errors when it is trying to configure pve-manager. Let's see your `dkms status` and `apt-get install --fix-broken` output?

I remember I had a similar problem at install and it was fixed this way: (https://github.com/pimox/pimox7/issues/87#issuecomment-1236403022)
```apt install dpkg-dev raspberrypi-kernel-headers linux-image-generic```
But I don't know if that is your problem or not.

diginfo, 2022-12-04:
Thanks, I am halfway thru testing a manual install, I did add the kernel headers but install also failed in a similar way.

Also notice the the O/S no longer uses /etc/network/interfaces now and saw some errors about that, I have modified /etc/dhcpcd.conf instead but it still failed.

```
# apt-get install --fix-broken
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 42 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
apt-listchanges: Can't set locale; make sure $LC_* and $LANG are correct!
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = "en_US.UTF-8",
	LC_CTYPE = "en_US.UTF-8",
	LC_TERMINAL = "iTerm2",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
Setting up ceph-dkms (0.0.2) ...
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
Removing old ceph-dkms-0.0.2 DKMS files...
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

------------------------------
Deleting module version: 0.0.2
completely from the DKMS tree.
------------------------------
Done.
Loading new ceph-dkms-0.0.2 DKMS files...
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
It is likely that 5.15.61-v8+ belongs to a chroot's host
Building for 5.15.61-v8+ and 5.15.76-v8+
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Module build for kernel 5.15.61-v8+ was skipped since the
kernel headers for this kernel does not seem to be installed.
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Building initial module for 5.15.76-v8+
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Error! Bad return status for module build on kernel: 5.15.76-v8+ (aarch64)
Consult /var/lib/dkms/ceph-dkms/0.0.2/build/make.log for more information.
dpkg: error processing package ceph-dkms (--configure):
 installed ceph-dkms package post-installation script subprocess returned error exit status 10
Setting up pve-manager (7.2-7) ...
Job for pvestatd.service failed because the control process exited with error code.
See "systemctl status pvestatd.service" and "journalctl -xe" for details.
dpkg: error processing package pve-manager (--configure):
 installed pve-manager package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of proxmox-ve:
 proxmox-ve depends on pve-manager; however:
  Package pve-manager is not configured yet.
 proxmox-ve depends on ceph-dkms; however:
  Package ceph-dkms is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 ceph-dkms
 pve-manager
 proxmox-ve
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

bp0, 2022-12-04:
@diginfo, it does still use /etc/network/interfaces, managed by proxmox. vmbr0 should have a static IP assigned. Proxmox discourages using dhcp.
It also seems that the missing kernel headers preventing dkms from building modules is your installation problem. If the dpkg-dev headers solution didn't solve it for you, then you might try starting over and editing the install script as described here: https://youtu.be/g1FXZIvVkq4?t=301

diginfo, 2022-12-04:
@bp0  - Thanks for helping but no luck I'm afraid:

```
apt update
apt -y install dpkg-dev raspberrypi-kernel-headers linux-image-generic
apt -y install zfs-dkms zfsutils-linux
```
**and at the end of zfs install:**

```
DKMS: install completed.
Setting up libzpool5linux (2.1.5-1~bpo11~rpt1) ...
Setting up zfsutils-linux (2.1.5-1~bpo11~rpt1) ...
modprobe: FATAL: Module zfs not found in directory /lib/modules/5.15.61-v8+
Created symlink /etc/systemd/system/zfs-import.target.wants/zfs-import-cache.service → /lib/systemd/system/zfs-import-cache.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-import.target → /lib/systemd/system/zfs-import.target.
Created symlink /etc/systemd/system/zfs-mount.service.wants/zfs-load-module.service → /lib/systemd/system/zfs-load-module.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-load-module.service → /lib/systemd/system/zfs-load-module.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-mount.service → /lib/systemd/system/zfs-mount.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-share.service → /lib/systemd/system/zfs-share.service.
Created symlink /etc/systemd/system/zfs-volumes.target.wants/zfs-volume-wait.service → /lib/systemd/system/zfs-volume-wait.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-volumes.target → /lib/systemd/system/zfs-volumes.target.
Created symlink /etc/systemd/system/multi-user.target.wants/zfs.target → /lib/systemd/system/zfs.target.
zfs-import-scan.service is a disabled or a static unit, not starting it.
Job for zfs-load-module.service failed because the control process exited with error code.
See "systemctl status zfs-load-module.service" and "journalctl -xe" for details.
A dependency job for zfs-import-cache.service failed. See 'journalctl -xe' for details.
Processing triggers for initramfs-tools (0.140) ...
ln: failed to create hard link '/boot/initrd.img-5.10.0-19-arm64.dpkg-bak' => '/boot/initrd.img-5.10.0-19-arm64': Operation not permitted
update-initramfs: Generating /boot/initrd.img-5.10.0-19-arm64
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ...
Processing triggers for man-db (2.9.4-2) ...
Setting up zfs-zed (2.1.5-1~bpo11~rpt1) ...
Created symlink /etc/systemd/system/zed.service → /lib/systemd/system/zfs-zed.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-zed.service → /lib/systemd/system/zfs-zed.service.
```

Maybe someone else could try out this installation using today's binaries etc etc ?

bp0, 2022-12-04:
@diginfo This morning I installed pimox7 on a Raspberry Pi 4, to try and replicate your problem, but everything worked as expected.
* I used the official Raspberry Pi Imager and the latest Lite arm64 image, 2022-09-22-raspios-bullseye-arm64-lite.img.xz. 
* I followed the "Interactive Automatic Installer" instructions in the pimox7 README exactly, without editing the script. 
* After it finished, I had a working pimox install, I could log in to the web interface and create a VM using the Debian arm64 netinst ISO, only changing hardware/options to make the virtual CD-ROM use SCSI instead of IDE.
* I did not have working ZFS or Ceph because of the DKMS missing headers problem. Running `apt install dpkg-dev raspberrypi-kernel-headers linux-image-generic` triggered those modules to build for the 5.15.76-v8+ kernel, and after `modprobe zfs`, ZFS was working.

diginfo, 2022-12-05:
Thanks very much, got it working with:

```
## 221205
## ======

sudo -s
curl https://raw.githubusercontent.com/pimox/pimox7/master/RPiOS64-IA-Install.sh > RPiOS64-IA-Install.sh;
chmod +x RPiOS64-IA-Install.sh;
./RPiOS64-IA-Install.sh;

## Post Install
###############
sudo -s
apt -y upgrade
apt install -y dpkg-dev raspberrypi-kernel-headers linux-image-generic
reboot

## Images
#########
sudo -s
cd /var/lib/vz/template/cache
curl https://us.lxd.images.canonical.com/images/debian/bullseye/arm64/default/20221204_05:25/rootfs.tar.xz > deb_arm64_20221204.tar.xz
curl https://us.lxd.images.canonical.com/images/alpine/3.17/arm64/default/20221204_13:13/rootfs.tar.xz > alpine_arm64_20221204.tar.xz
curl https://us.lxd.images.canonical.com/images/archlinux/current/arm64/default/20221204_04:18/rootfs.tar.xz > arch_arm64_20221204.tar.xz
```

jimoyong, 2022-12-06:
> 
Thank you for your reply. I tried press esc with the button VNC provided and  can't get into UEFI menu.

jiangcuo, 2022-12-20:
Hi , try ```qm set <vmid> -m virt,gic-version=host```

jimoyong, 2022-12-21:
> 
Thank you for your reply.
I tried your cmd, but it gave the answer as the following pic:
[image]
I also tried -m with -machine  with no luck.
[image]
 Do you have any idea about that? Thank you

titogrima, 2022-12-21:
Hi 

I had same problem and I solved it with this configuration for display

[image]

At least in a non-graphical environment

[image]

Regards!!

jimoyong, 2022-12-22:
> Hi
> 
> I had same problem and I solved it with this configuration for display
> 
> [image]
> 
> At least in a non-graphical environment
> 
> [image]
> 
> Regards!!
Thank you for your reply!
I tired your config, with no luck. T_T
[image]
[image]

Thank you all the same!

jiangcuo, 2022-12-24:
qm set vmid --args "-machine virt,gic-version=host"

jimoyong, 2022-12-26:
> qm set vmid --args "-machine virt,gic-version=host"
It works!!
Thank you very much! 

By the way, can you tell me the meaning of gic-version with pimox? Thank you!
[image]

jimoyong, 2023-01-12:
> [image]
> 
> [image]
> 
> [image]
> 
> Friends, I too have the same problem I installed Pimox on a Raspberry PI4 8Gb Rad and 1TB SSD and I would like to run multiple virtualized raspberry systems on them. I tried to imprint as you can see from the screenshots a raw of a disk of a physical raspberry it sees and imports without problems But then the VM won't start for me How do I fix it and how should I care for the raspberry clone VM?
> 
> Thank you

I 'm not sure about your situation, in my case, I use: [qm set vmid --args "-machine virt,gic-version=host"] in the shell, maybe you can have a try.

saviothecnic, 2023-01-12:
[image]

Thanks, I tried your command and it gave me an error
But can you do me a favor you can try putting it on your PiMox
any rapsberry physical machine by importing its raw disk
and tell me if you part? Thank you

jimoyong, 2023-01-13:
> [image]
> 
> Thanks, I tried your command and it gave me an error But can you do me a favor you can try putting it on your PiMox any rapsberry physical machine by importing its raw disk and tell me if you part? Thank you

Sorry, it's my my ambiguous words. The "vmid" in the command above is the id of your vm, in the pic it should be 100 or 185.

saviothecnic, 2023-02-22:
Ok I tried with  qm set 100 --args "-machine virt,gic-version=host"
as in the photo below now I have no error but the VM still does not start :(
[image]