pimox/pimox7 #110: why is proxmox looking for usb device when no media was  added?

Monallabs, 2022-12-22:
I created a vm with no media attached to it. When I start the VM, I get the error:
```
kvm: -device usb-ehci,id=ehci,bus=pci.0,addr=0x1: Duplicate ID 'ehci' for device
TASK ERROR: start failed: QEMU exited with code 1
```

Why is proxmox looking for usb. I had attached a passthrough usb for another VM that I had deleted. 
Is it possible that the config might be getting messed up? should I uninstall and reinstall proxmox again?

emadadel2008, 2023-08-04:
@Monallabs do you fix this, can you please give me the solution

kenkin360, 2024-01-18:
You might have tried to run a x86 vm on your pimox environment, to do that you'd need to specify `arch: x86_64` in the vm config. Have a look at: 

https://www.reddit.com/r/Proxmox/comments/ntg650/proxmox_on_the_raspberry_pi_running_x86_vms_for/?rdt=54490

zoechi, 2024-09-26:
I run into the same error when I tried to run a aarch64 VM on an Intel x86_64 system.
Changing the "Machine: q35" to "Machine: Default (i440fx)" fixed the error.