pimox/pimox7 #113: Install Pimox on a normal Debian ARM64 VM??

ymurawski, 2023-01-12:
Hi im using a mac mini m1 and im using Ubuntu ARM as a VM for my containers.
I wanted to give Proxmox a try and installed Pimox on a Debian ARM64 VM. install just seems to work finde but i cant reach the IP i gave to enter in the setup....eg 192.168.1.202/24

So the URL must be 192.168.1.202:8006 right ?

ymurawski, 2023-01-12:
Seems that the VM has no internet / networks working after install

cant reach any Homepage or curl anything 

Any ideas ???

rcarmo, 2023-01-12:
You need to be a bit more explicit about what VM manager you are using on the Mac, and check the network configuration for it - quite often you will have to make sure the setup IP fits within a predefined range, etc.

ymurawski, 2023-01-12:
> You need to be a bit more explicit about what VM manager you are using on the Mac, and check the network configuration for it - quite often you will have to make sure the setup IP fits within a predefined range, etc.

I'm using parallels desktop and set all my VMs to their own ip to optain from my router.

So for xampöe alls my devices at home are in 192.168.1.X

The Debian VM is set to static on 192.168.1.202 and in the Pimox setup I enter 192.168.1.202/24

rcarmo, 2023-01-12:
Well, with a setup like that you may not be able to connect directly from your Mac to the VM - especially if you're using Wi-Fi, since those interfaces are typically not set up with bridging.

Are you using Ethernet? Can you SSH or talk to your other VMs from the Mac? If yes, then it might just be Proxmox not setting up its network interface correctly and you need to fix that via the VM console (I use Parallels too but never tried to set up Proxmox on my laptop).

ymurawski, 2023-01-12:
> Well, with a setup like that you may not be able to connect directly from your Mac to the VM - especially if you're using Wi-Fi, since those interfaces are typically not set up with bridging.
>
> Are [...]

All other VMs just working fine. Just the Debian VM with primox has no networking at all. No Homepage like YouTube or Google can be opened in this vm. 
But my router shows its connected as ethernet. So I think it's the broken network of the vm ( I already tried a complete new vm but same result) 

Any ideas how to fix that?

ymurawski, 2023-01-12:
BTW im using the Mac m1 because it's so powerful but extremely energy saving. My old server took around 130w and this mac m1 is around 14w 😅

rcarmo, 2023-01-12:
Oh, you could have said you had a desktop environment set up. Fixing the network in Pimox will require you to use the terminal to debug and check the configuration - and that's something that is very hard to go over here. 

You should start by checking what network interfaces are up, and whether they have an IP address - and then go over the network configuration scripts (which pimox bludgeons into submission to set up a fixed IP address on a bridged interface).

I _suspect_ that the bridging and Parallels won't get along well, but I don't have the time to reproduce that in my laptop...

ymurawski, 2023-01-12:
> Oh, you could have said you had a desktop environment set up. Fixing the network in Pimox will require you to use the terminal to debug and check the configuration - and that's something that is [...]

Sorry for misunderstanding 😅
OK I will try it but Linux is not my favorite os 😅🙈
If I get this to work I would share the vm for the community

ymurawski, 2023-01-12:
> Oh, you could have said you had a desktop environment set up. Fixing the network in Pimox will require you to use the terminal to debug and check the configuration - and that's something that is [...]

I tried a little around ...when i edit the "sudo nano /etc/network/interfaces"
`auto lo
iface lo inet loopback

iface eth0 inet manual

auto vmbr0
iface vmbr0 inet static
        address 192.168.1.202/24
        gateway 192.168.1.1
        bridge-ports eth0
        bridge-stp off
        bridge-fd 0

to
`auto lo
iface lo inet loopback

iface eth0 inet manual

#auto vmbr0
#iface vmbr0 inet static
 #       address 192.168.1.202/24
 #       gateway 192.168.1.1
 #       bridge-ports eth0
 #       bridge-stp off
 #       bridge-fd 0

my internet works fine but still not reachable under 192.168.1.202:8006

more ideas are not in my brain :D

thoschl, 2023-01-12:
You are on the right way! In /etc/network/interfaces subset eth0 by enp0s5 That should work. I have the same configuration running here...

ymurawski, 2023-01-12:
> You are on the right way! In /etc/network/interfaces subset eth0 by enp0s5 That should work. I have the same configuration running here...

Sorry but what do you mean by subset? My Linux knowledge isn't that good 😅

thoschl, 2023-01-12:
Oh, sorry! I think your /etc/network/interfaces should look like this:
```
auto lo
iface lo inet loopback

iface enp0s5 inet manual

auto vmbr0
iface vmbr0 inet static
        address 192.168.1.202/24
        gateway 192.168.1.1
        bridge-ports enp0s5
        bridge-stp off
        bridge-fd 0
```

ymurawski, 2023-01-12:
> 

ok internet 

> Oh, sorry! I think your /etc/network/interfaces should look like this:
>
> ```
> auto lo
> iface lo inet loopback
>
> iface enp0s5 inet manual
>
> auto vmbr0
> iface vmbr0 inet static
>         address [...]

ok thanks :) internet in this vm is working now. But i cant reach the proxmox web UI...how can i check if this is running ? 
Are you using also Parrallels or a normal debian ARM64 VM ?

ymurawski, 2023-01-12:
I FOUND THE LAST SOLUTION....simply just use https://......:8006

my browser used http automatically. i can reach now Proxmox ....wuhuuuuuuuu

closed 2023-01-12 by ymurawski

thoschl, 2023-01-12:
CONGRATULATIONS!
I use Parallels, but at the moment I  am moving to UTM (Proxmox ARM64 runs on there too).
You can check if a prcess ist listening on port 8006 by using the following command:
ss -antl
this should return a line like this
`LISTEN              0                   4096                                         *:8006                                      *:*       `

denisgolius, 2023-03-17:
Hey @rcarmo 

Do you think there is a chance to run pimox on https://deploy.equinix.com/product/servers/c3-large-arm64/ ?