pimox/pimox7 #114: Packer boot_command not working

rossmaclean, 2023-01-13:
I'm not sure if this is an issue with Pimox specifically, but I'm not sure where else to ask this.

I'm trying to use Packer to automatically create templates from ISOs, however during the `boot_command`(https://developer.hashicorp.com/packer/plugins/builders/proxmox/iso#boot-command) phase where it performs key presses on the grub menu, it doesn't work. Nothing happens, as if the key presses are being ignored.

It works fine on regular Proxmox. I'm happy to post my definition files etc but just wanted to see if anyone else was using Packer and had experienced similar issues.

jiangcuo, 2023-01-16:
Can you share your vm configuration and vm startup screenshots?

rossmaclean, 2023-01-27:
Hey, here is the vm config.

If you are wondering why there are 2 iso files, it's a hack I'm using to get around limitations in the Packer SDK (https://github.com/hashicorp/packer-plugin-sdk/issues/119)

`# Ubuntu Server jammy
# ---
# Packer Template to create an Ubuntu Server (jammy) on Proxmox

# Variable Definitions
variable "proxmox_api_url" {
    type = string
}

variable "proxmox_api_token_id" {
    type = string
}

variable "proxmox_api_token_secret" {
    type = string
    sensitive = true
}

# Resource Definiation for the VM Template
source "proxmox" "ubuntu-server-jammy" {
 
    # Proxmox Connection Settings
    proxmox_url = "${var.proxmox_api_url}"
    username = "${var.proxmox_api_token_id}"
    token = "${var.proxmox_api_token_secret}"
    # (Optional) Skip TLS Verification
    insecure_skip_tls_verify = true
    
    # VM General Settings
    node = "RPi4-01-PVE"
    vm_id = "169"
    vm_name = "ubuntu-server-jammy"
    template_description = "Ubuntu Server jammy Image"

    # VM Bios Settings
    bios = "ovmf"
    efidisk = "USBT1"

    # VM OS Settings
    # (Option 1) Local ISO File
    iso_file = "local:iso/ubuntu-22.04.1-live-server-arm64.iso"
    # - or -
    # (Option 2) Download ISO
    # iso_url = "https://releases.ubuntu.com/22.04/ubuntu-22.04-live-server-amd64.iso"
    # iso_checksum = "84aeaf7823c8c61baa0ae862d0a06b03409394800000b3235854a6b38eb4856f"
    iso_storage_pool = "local"
    unmount_iso = true

    # VM System Settings
    qemu_agent = true

    # VM Hard Disk Settings
    scsi_controller = "virtio-scsi-pci"

    disks {
        disk_size = "20G"
        format = "raw"
        storage_pool = "USBT1"
        storage_pool_type = "lvm"
        type = "scsi"
    }

    additional_iso_files {
            device = "scsi1"
            iso_file = "local:iso/ubuntu-22.04.1-live-server-arm64.iso"
            unmount = true
    }

    boot = "order=scsi1;"

    # VM CPU Settings
    cpu_type = "host"
    cores = "1"
    
    # VM Memory Settings
    memory = "2048" 

    # VM Network Settings
    network_adapters {
        model = "virtio"
        bridge = "vmbr0"
        firewall = "false"
    } 

    # VM Cloud-Init Settings
    cloud_init = true
    cloud_init_storage_pool = "local-lvm"

    # PACKER Boot Commands
    boot_command = [
        "<wait8>",
        "e<wait2>",
        "<down>",
        "<wait>",
        "<down>"
#        "<esc><wait>",
#       "e<wait>",
#         "<down><down><down><end>",
#        "<bs><bs><bs><bs><wait>",
#        "autoinstall ds=nocloud-net\\;s=http://{{ .HTTPIP }}:{{ .HTTPPort }}/ ---<wait>",
#        "<f10><wait>"
    ]
    boot_wait = "45s"

    # PACKER Autoinstall Settings
    http_directory = "http" 
    # (Optional) Bind IP Address and Port
    # http_bind_address = "0.0.0.0"
    # http_port_min = 8802
    # http_port_max = 8802

    ssh_username = "your-user-name"

    # (Option 1) Add your Password here
    ssh_password = "your-password"
    # - or -
    # (Option 2) Add your Private SSH KEY file here
    # ssh_private_key_file = "~/.ssh/id_rsa"

    # Raise the timeout, when installation takes longer
    ssh_timeout = "20m"
}

# Build Definition to create the VM Template
build {

    name = "ubuntu-server-jammy"
    sources = ["source.proxmox.ubuntu-server-jammy"]

    # Provisioning the VM Template for Cloud-Init Integration in Proxmox #1
    provisioner "shell" {
        inline = [
            "while [ ! -f /var/lib/cloud/instance/boot-finished ]; do echo 'Waiting for cloud-init...'; sleep 1; done",
            "sudo rm /etc/ssh/ssh_host_*",
            "sudo truncate -s 0 /etc/machine-id",
            "sudo apt -y autoremove --purge",
            "sudo apt -y clean",
            "sudo apt -y autoclean",
            "sudo cloud-init clean",
            "sudo rm -f /etc/cloud/cloud.cfg.d/subiquity-disable-cloudinit-networking.cfg",
            "sudo sync"
        ]
    }

    # Provisioning the VM Template for Cloud-Init Integration in Proxmox #2
    provisioner "file" {
        source = "files/99-pve.cfg"
        destination = "/tmp/99-pve.cfg"
    }

    # Provisioning the VM Template for Cloud-Init Integration in Proxmox #3
    provisioner "shell" {
        inline = [ "sudo cp /tmp/99-pve.cfg /etc/cloud/cloud.cfg.d/99-pve.cfg" ]
    }

    # Provisioning the VM Template with Docker Installation #4
    provisioner "shell" {
        inline = [
            "sudo apt-get install -y ca-certificates curl gnupg lsb-release",
            "curl -fsSL https://download.docker.com/linux/ubuntu/gpg | sudo gpg --dearmor -o /usr/share/keyrings/docker-archive-keyring.gpg",
            "echo \"deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/docker-archive-keyring.gpg] https://download.docker.com/linux/ubuntu $(lsb_release -cs) stable\" | sudo tee /etc/apt/sources.list.d/docker.list > /dev/null",
            "sudo apt-get -y update",
            "sudo apt-get install -y docker-ce docker-ce-cli containerd.io"
        ]
    }
}`

I can post screenshots later.

jiangcuo, 2023-02-19:
I'm not good at the Packer.
Sorry, can't help you.
I've tried installing ubuntu manually with no issues.