pimox/pimox7 #115: Gpu Passthrough?

cvocvo, 2023-01-15:
Is it possible to passthrough the pi’s GPU to a VM? Or otherwise allow a VM to take advantage of hardware acceleration/hardware encoding?
I have a VM with Docker running frigate, and frigate has hardware acceleration support for ffmpeg with raspberry pi 3/4. If not it seems the only way to achieve that would be to not use pimox and just run it directly on the hardware.

Thank you!

jiangcuo, 2023-01-16:
You can try creating a linux container .
https://www.youtube.com/watch?v=-Us8KPOhOCY

cvocvo, 2023-01-17:
Ahh yea I’m familiar with passing GPUs, etc through to VMs and LXC. The knowledge gap for me is how to do that for a raspberry pi 4.

The structure is like:
- Raspberry Pi 4
  - Pimox7
    - Debian LXC
      - Docker
        - Frigate Container that wants V4L2 hardware acceleration 

I’m thinking it must be something like some combination of these:
- https://github.com/linuxserver/docker-jellyfin#v4l2-raspberry-pi
- https://forum.openmediavault.org/index.php?thread/45370-hardware-encoding-in-jellyfin-docker-portainer/&postID=330696#post330696
- https://discuss.linuxcontainers.org/t/solved-how-to-passthrough-dev-ttys0/1782

Then it’s essentially:
Somehow pass through this from Proxmox (Pi) to LXC (running Docker + Portainer):
```
/dev/video10
/dev/video11
/dev/video12
```
And then specify on the Docker container:
```
--device=/dev/video10:/dev/video10
--device=/dev/video11:/dev/video11
--device=/dev/video12:/dev/video12
```
Has anyone done this? Is this feasible?
Are those device IDs the correct ones? (I’m thinking so if it works for Jellyfin transcoding?)

cvocvo, 2023-01-17:
It looks like it does work; hardware acceleration on pi 3's and 4's is currently broken though on Frigate so I will have to find another way to test further.

I edited my container configuration:
`nano /etc/pve/lxc/500.conf`
And added the following lines to do the passthrough from the proxmox Pi 4 host to the LXC:
```
lxc.cgroup2.devices.allow: c 81:* rwm
lxc.mount.entry: /dev/video10 dev/video10 none bind,optional,create=file
lxc.mount.entry: /dev/video11 dev/video11 none bind,optional,create=file
lxc.mount.entry: /dev/video12 dev/video12 none bind,optional,create=file
```
And then just rebooted the LXC docker container.

Passthrough works to LXC:
[image]

Passthrough works to the Docker Container:
[image]

mmiszy, 2023-03-09:
`video10`…`video10` show in `/dev/`, unfortunately it seems I don't have access to them. Their mod:own is set to `nobody:nogroup` :(