pimox/pimox7 #119: Can pimox be part of proxmox ?

FilipStadler, 2023-01-21:
I know RPI can be used with two proxmox nodes but is it possible to join a proxmox closter and migrate containers/vm to the promox and back to pimox I am not really sure if pimox can do this or is it limited to pimox to pimox ?

jiangcuo, 2023-01-23:
Hi. 
Live migrate need the same cpu. 
You can use offline migrate  on different node.
I have tested unable to live migrate from Proxmox-amd64 to pimox .I think the reason is aarch64 on proxmox-amd64 using tcg to emulate,
I don't recommend doing this.

FilipStadler, 2023-01-23:
I understand that it should be possible to run arm64 on my intel proxymox like this 
https://i12bretro.github.io/tutorials/0661.html  so maybe it was possible to migrate.

Its was just cool if one can do some work and test later migrate to a raspberry.

But I understand this project and nothing todo with the https://github.com/pimox/pimox7 ?