pimox/pimox7 #120: Syslog flooding with: "pvestatd[1092]: bad key=value pair in nested keyed file" message

fakemanhk, 2023-01-29:
I just found out that in my syslog, the message mentioned in title repeated multiple times (not fixed, can be around 8-14 times), and showing up every 10 sec, trying to search around but couldn't find what's causing it, may I have some hints?

ArchemedIan, 2023-09-09:
ever find any info on this?

soakes, 2023-12-26:
I have the same problem.

```
Dec 26 02:49:22 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:22 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:31 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:31 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:31 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:31 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:31 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:31 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:31 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
Dec 26 02:49:31 pve03 pvestatd[712557]: bad key=value pair in nested keyed file
```

I have tried to remove the `/var/lib/rrdcached/db/` dir which I thought it was related to stats and was hopping that it would regenerate on next pool, but this doesnt resolve the issue.

If anyone has another idea, would love to hear from you.  In the meantime, ile see if I can track the problem/fix down and will post here if I am successful.