pimox/pimox7 #121: Can't connect

SageSphinx63920, 2023-02-06:
Hello,
I can't find any other way to get help with pimox so I just opened this issue. I have a PI 4B+ with 8GB ram. I runned the install script like the repo said but I can not connect to proxmox at all. The install script finished and restarted my pi. It just boots into normal rasped 64 bit and yes I can ping the pi with the ip I set

automationxpert, 2023-03-19:
This could be a dumb question but just confirming, did you try entering with HTTPS and 8006 port i.e
https://{your_ip_adress}:8006

366Cobra, 2023-05-27:
i have the same issue.

i have done a test install, but lost all connection to the pi.
after OS install, ssh was fine, but now, i can no longer use ssh (kex_exchange_identification: read: Connection reset by peer) and going to the site, i just get "An error occurred during a connection to 192.168.2.218:8006. PR_CONNECT_RESET_ERROR"

when installing pimox, i used the same IP as it already had.  there were a couple of errors, but watching install vids, they had the same ones and it worked for them.
i can login and then use the site locally, but no longer over the network.
proxmox does not show entries in the firewall settings and ssh status is up.

i even tried setting the IP for ssh to the one supplied at the install.
port 22 is open and ssh is listening on it.

also set iptable rules for full access.
logs are not showing anything when trying to access the box, so guess traffic is not even let in.
i am a bit rusty what linux is concerned so have not been able to work out what the issue.

Stats:
RaspPi 4b 4GB
1TB SSD

TheBossME, 2023-05-27:
Use a Debian testen 11 bullseye Image From Herr:

https://raspi.debian.net/tested/20230102_raspi_4_bullseye.img.xz

And Use the Instructuons From Here.

https://github.com/jiangcuo/Proxmox-Arm64/wiki

366Cobra, 2023-05-27:
i would call 2 months not a very long time.
at least that is what the time stamps in the rep say.

TheBossME, 2023-05-27:
Sorry. Pimox Stands at 7.2 

Thermometer i've Posten is up to Date on 7.4

366Cobra, 2023-05-28:
got it working.

i was using my mac to connect to the pi, like i always do, but after the pimox install it no longer worked.
am also running a windows vm.  i tried it there and it works.
ssh and the site come up.
cannot explain why, but i now have a workaround.

still find it odd to butcher eth0, but i guess the team must have a reason.
QEMU usually just creates additional adapters.